FormerLurker/ArcWelderPlugin #1: First impressions

KenLucke, 2020-04-26:
Installed and doing first print. 

- Reduced a 10.8Mb file to 1.9Mb.
- Can already see smoother arcs happening, but tight corners slows way down.  I'm sure it will all balance out
- Please add a Before and After Filesize (or just bytes saved) to the results dialog, for information (and dancing in the streets) purposes.
- Should be a way (am I missing it?) to process already uploaded files (and to prevent re-processing ones that have already been done), so they don't have to be deleted and re-uploaded.
- Have not checked, but does it handle multiple files uploaded at once (i.e., dragging a stac k of files to OctoPrint), or does it require one-by-one upload?
- Have not checked, but hoping this will affect larger gcode files that would crash the gcode viewer enough that that it will be able to handle them once reduced.

FormerLurker, 2020-04-26:
> Please add a Before and After Filesize (or just bytes saved) to the results dialog, for information (and dancing in the streets) purposes.

Can you add a feature request for this please?  Just create a new issue and add [feature request] before the title.  Describe exactly what you'd like to see please (even though I have a good idea, it is more for other people's benefit).

> Should be a way (am I missing it?) to process already uploaded files (and to prevent re-processing ones that have already been done), so they don't have to be deleted and re-uploaded.

Yes, I've thought about this.  This is pre-release ATM.  It's just a little more than the minimum viable plugin.  Please add a feature request for this too.  It's also something I've thought of.

> Have not checked, but does it handle multiple files uploaded at once (i.e., dragging a stac k of files to OctoPrint), or does it require one-by-one upload?

The preprocessor hook is a bit wonky for this purpose, even though it is designed for pre-processing.  It's a bit difficult to explain actually, but I'm working to utilize the event hook instead with a queue.  It's terribly complicated, and there are lots of edge cases (deleting a file while processing, etc) to work out.  If you upload multiple files right now, it will work, but the performance is bad.  Working on this.

> Have not checked, but hoping this will affect larger gcode files that would crash the gcode viewer enough that that it will be able to handle them once reduced.

The gcode viewer has problems displaying gcode with arcs, though OctoPrint v1.4 is much better.  It's still not perfect.  It should actually increase performance since there are fewer gcodes to analyze, especially since the viewer only shows one layer at a time.

KenLucke, 2020-04-26:
Will do on the separate feature requests.

As for the other issues, just expressing thoughts... I know this is pre-pre-pre-pre-release :)

| have seen better results from the processed files, but still a little blobbing, but seems to be more in specificc areas than randomly as before AW.  For some reason, github does not like my .jpg files and will not accept them.

FormerLurker, 2020-04-26:
Any blobs you see now may be related to how your specific firmware handles arc commands.  If you can find the configuration_adv.h file for your firmware, paste in the Arc section parameters and I might be able to give you some insight.

FormerLurker, 2020-04-26:
OH, and thanks for creating those feature requests!  It's awesome to have them documented.

FormerLurker, 2020-04-27:
Hey, regarding this:

> Can already see smoother arcs happening, but tight corners slows way down. I'm sure it will all balance out

Can you send me a link to your firmware, and if it's marlin based or a fork of that, can you send me a link to the configuration_adv.h file?  Based on some experimentation and tinkering, I might be able to suggest slight settings modification that can improve tight corners.  I'm working on a modification that should greatly reduce firmware slowdown on tight corners.

FormerLurker, 2020-04-27:
> Have not checked, but does it handle multiple files uploaded at once (i.e., dragging a stac k of files to OctoPrint), or does it require one-by-one upload?

This sort of worked before, but now I've created a worker thread that handles the processing.  This was necessary to get the cancel button working (as you can see in the progress bar display above.  I do need to hide the cancel buttons for non-admin users still (server side cancel would be rejected for non-admins).

The last thing on the list is enabling processing for existing files, which I will answer in your feature request issue.

FormerLurker, 2020-05-05:
Ok, so I think most of the issues mentioned here have been corrected by the new release.  If you have any other issues or feature requests, please create a new issue.  Thank you!  Closed :)

closed 2020-05-05 by FormerLurker