FormerLurker/ArcWelderPlugin #2: Works great for arcs of r≤2.5mm, stuttering with arcs of r>2.5mm

kulfuerst, 2020-04-26:
I am expecting a firmware issue here, rather than an issue with the plugin. I'm still running Marlin2 bugfix on an Anycubic i3 Mega.
Small circles are printed perfectly, but circles above 5 mm diameter are stuttering more so than before. Does somebody have an idea what could be the cause here? The i3 Mega board is 8-bit and generally not very powerful, could that be an issue here? 

video: https://photos.app.goo.gl/o4Ueo4JyRiK5tWpM6
[image]

FormerLurker, 2020-04-26:
O.1 is too low foe mm_per_arc_segment.  Try 0.5.

kulfuerst, 2020-04-26:
what about min_arc_segments and n_arc_correction? As much as I'd love to I cannot make sense of what they do.

FormerLurker, 2020-04-26:
Oh, i did not see min arc segments!  I had just gotten out of bed.  Mm_per_arc_segment is tye maximum length of a segment when converting G2/G3 to line segments (Cartesian printers dont do curces natively).  Min_arc_segments is the minimum number of segments for a full circle of any radius.  For large arcs mm_per_arc_segment is used.  For smaller arcs the minimum number of segments is used.  So, set Mm_per_arc_segment to 1 (plenty short for getting excellent large arcs), and leave the rest alone.  You will get great arcs from those settings.  Ignore the correction setting, it has almost 0 impact for curves you will print.

Check out this pic i made showing mm_per_arc_segment = 0.1:

 https://user-images.githubusercontent.com/33363128/79670433-92ffe280-8188-11ea-864f-7927bc1d07f2.png

Here is one using 1 for that setting and 32 for min_arc_segments:

https://user-images.githubusercontent.com/33363128/79670525-613b4b80-8189-11ea-9f54-a9d954eb9c8d.png

Much better.  You can probably see why you were getting some stutter now.

kulfuerst, 2020-04-26:
Running another test now :)

kulfuerst, 2020-04-26:
It is working far, far, better now, but there is a new issue. I now hat the problem that the nozzle stops after long line segments. What slicers have you tested the plugin with? I know certain script post processors have issues with gcode generated by cura for some reason.

FormerLurker, 2020-04-26:
Are you still using the same gcode you posted above?  I will try to test is as soon as I can once you verify you are still using the same code.

kulfuerst, 2020-04-26:
@FormerLurker, that's correct. I am still using the same on.
I'd like to add that the printer is now able to print large circles at speeds I could not achieve via USB printing before. It's only in certain instances mostly affecting small circles where issues occur. My changed parameters look as such:
[image]

FormerLurker, 2020-04-26:
OH, keep in mind it could be an issue with the processor still if I've messed up the feedrate change detection at all.  Will know more later.

FormerLurker, 2020-04-26:
One last thing (I keep thinking of stuff):  Link the current source of your firmware here and I'll see if I can make a quick patch to add the new configuration value.  It would be good to test the change in the wild.

kulfuerst, 2020-04-26:
I'm not very good with github, so please let me know if I need to grant permission for you to edit or something like that. An older version of my current FW can be found here, almost exactly like what I'm currently using: https://github.com/kulfuerst/Marlin_Ai3M_2.0.x_bugfix_piezo

FormerLurker, 2020-04-27:
@kulfuerst, thanks for the link.  I would probably need permission to edit.  Let me take a look and see what I can do.  It might take a bit for me to get around to it since I'm still making modifications to my firmware.

I'm also going to check all the feedrates and do some output testing of your original gcode to see if I can gain some insight.  FYI, I've tested PrusaSlicer and Cura so far.

FormerLurker, 2020-04-27:
Also, I just realized that I misread your title!  I got it backwards.

Try setting MM_PER_ARC_SEGMENT to 2.  You should get very similar results hopefully.  Let me know how that works since I've never gone above 1MM for that setting.

kulfuerst, 2020-04-27:
@FormerLurker, the title isn't really accurate anymore, since you've already solved that issue by recommending for my MM_PER_ARC_SEGMENT to be increased to 1. It is now medium sized arcs that are the problem, along with pauses after straight line segments, feedrate changes and several G0 commands in a row.
Thanks a ton for taking care of this. The idea of this plugin is absolutely brilliant! I'm happy to report my findings and help along.

FormerLurker, 2020-04-27:
@kulfuerst, thanks for your compliments!  I will definitely dig into this more.  Still try increasing MM_PER_ARC_SEGMENT.  If the arc quality still looks good, it will cut the amount of processing in half.  If I can get MIN_MM_PER_ARC_SEGMENT from my firmware fork implemented in yours, we might be able to turn both MM_PER_ARC_SEGMENT and MIN_ARC_SEGMENTS up quite a bit to further reduce processing without impacting quality.

marekorok, 2020-05-04:
Hi there,
I want to try arc's as well..
these settings are what I see defaults for SKR E3 Mini 1.2 
general Arc support was disabled by default (Marlin-bugfix-2.0.x, fresh download)
I'll keep it and try.

```
//
// G2/G3 Arc Support
//
#define ARC_SUPPORT               // Disable this feature to save ~3226 bytes
#if ENABLED(ARC_SUPPORT)
  #define MM_PER_ARC_SEGMENT      1 // (mm) Length (or minimum length) of each arc segment
  //#define ARC_SEGMENTS_PER_R    1 // Max segment length, MM_PER = Min
  #define MIN_ARC_SEGMENTS       24 // Minimum number of segments in a complete circle
  //#define ARC_SEGMENTS_PER_SEC 50 // Use feedrate to choose segment length (with MM_PER_ARC_SEGMENT as the minimum)
  #define N_ARC_CORRECTION       25 // Number of interpolated segments between corrections
  // #define ARC_P_CIRCLES           // Enable the 'P' parameter to specify complete circles
  //#define CNC_WORKSPACE_PLANES    // Allow G2/G3 to operate in XY, ZX, or YZ planes
#endif
```

Prickly-Pear, 2020-05-05:
@marekorok, you would want to reduce MM_PER_ARC_SEGMENT to 0.1 mm.
Or, you could reduce it to 0.5 mm and increase MIN_ARC_SEGMENTS to 32.

FormerLurker, 2020-05-05:
@prickley-pear, it seems to depend on the board, the exact firmware version, and the shape being printed.  Some people seem to have no problem with 0.1 (or even 0.05 for one person), but I am having trouble going under .5 with my stock firmware.  In general it seems like marlin 2.0 does pretty well though, from what I have heard.

FormerLurker, 2020-05-05:
@marekorok, make sure you have enough memory to enable arc commands.  If you get it working I would love to hear about it!

kulfuerst, 2020-09-13:
Hello @FormerLurker, my issue still persists and I just cannot seem to find a way to resolve it. the modified g-code partially prints at incredibly low speeds that must be somewhere in the 1 to 2 mm/s range. I have no explanation for this slow print speed. Please find one benchy g-code, modified and original, attached here. I would be very grateful if you could have a look.

[ArcWelder Benchy.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5213922/ArcWelder.Benchy.zip)

FormerLurker, 2020-09-17:
@kulfuerst, can you check your marlin version (in case you have upgraded), and ensure it is at 2.0.6 at a minimum?  A bunch of fixes were added for G2/G3 arc support.  I've heard that others have had great success with that version and above.

FormerLurker, 2020-10-17:
I'm going to go ahead and close this because I believe the problem is firmware related, and I think the firmware has been fixed.  Let me know if you want me to re-open this issue.  Thank you for posting!!!

closed 2020-10-17 by FormerLurker