FormerLurker/ArcWelderPlugin #3: Windows 10 Pop-Up Notifications

jneilliii, 2020-04-26:
There is a bit of a formatting issues with this latest version in the Windows 10 Pop0Up after a file is processed.

[image]

It would also be nice to have this notification a configurable option (to show/not to show),

FormerLurker, 2020-04-26:
I need to turn off desktop notification, or find a way to use a separate desktop toast for that notification.

Funny you should ask to turn it off because 3 people asked for such a pop-up (it didn't exist originally), lol.  I'm hesitant to add too many more settings, but i think i could make room for this.

Thanks for posting!

FormerLurker, 2020-04-26:
Try this new branch:  https://github.com/FormerLurker/ArcWelderPlugin/archive/devel.zip

I believe I fixed the desktop notification issue.  I just removed the desktop bit from the progress, and added a separate completed desktop notification with no fallback.  That means you will get both a desktop and browser toast for the completed message if it is enabled (If I did it right, they don't show up for me for some reason, though I allowed permissions.. happens with most notifications on most sites).

Let me know if it's fixed, and thanks for posting!

FormerLurker, 2020-04-26:
Oh, and I added notification settings for you too ;)

FormerLurker, 2020-05-05:
This has been completed and released.  Please open a new issue if you have any problems with desktop notifications.  For some reason they just don't appear (they are not disallowed on my machine) at all for me, so I cannot test :(

Closed!

closed 2020-05-05 by FormerLurker

jneilliii, 2020-05-05:
You know, I'm not sure I've seen those since installing the development package for testing either...