FormerLurker/ArcWelderPlugin #4: Absence of g2/g3 codes on created file. Small reduction of size.

toponto, 2020-04-26:
you will see there plenty of gcode error when printing, hence my need to reduce size of gcode models, even increasing buffers size to the limit in marlin, are still there. I´ve found that disconnecting the tft I reduce the number of issues at almost 100% but wondering if you solution will help reducing the number of gcode instructions to process. I´ve set the debussing level of the antistutter plugin to debug, and run it with another model... it produced the AS_xxx file again only 200kb smaller... and no log has been created:
[octoprint.log](https://github.com/FormerLurker/ArcWelderPlugin/files/4535883/octoprint.log)

As we have been trough varius iterations on private messaging, not sure what information to paste here, let me know if you need anything aditional. Waiting for you to come back with your next idea to fix the absence of G2/G3 codes in my conversion. No rush, it can wait.

FormerLurker, 2020-04-26:
Still trying to get to this.  It's such a difficult problem to debug since I can't replicate it.  I have some ideas though.  If you try installing from the github link (check the thread), post the plugin manager console log for me.  It could be your compiler/linker.

FormerLurker, 2020-04-26:
Oh, I've added verbose output flags to the compiler as well, just to help with your issue :)

toponto, 2020-04-26:
All rigth, will try as soon as current printing is done...cya!

toponto, 2020-04-26:
done as per your instructions.... file size only reduced 200k, as in the last tries. 
Attached the results of the plugin console.
Link to the arc welder log:
[https://drive.google.com/file/d/1KrhNzjhjvPLUrJczHTcq7myXup-OBCH1/view?usp=sharing](url)
Hope this helps, unfortunately the issue is still there... keep going! 

Best and thanks!

toponto, 2020-06-10:
hi man, just tried again with your last release, and updated python to V3 just in case... again... reduction of some space but no g2/g3 on the file.. weird

FormerLurker, 2020-10-17:
So, this is super old now, but I just added some compiler compatibility fixes.  Might want to try the latest release, which I'm in the process of adding (dev 5)

FormerLurker, 2021-01-23:
Going to close this out due to inactivity.  Thanks for your post, and feel free to open another issue if you are still having problems!

closed 2021-01-23 by FormerLurker