FormerLurker/ArcWelderPlugin #5: First try Not Sure what I have Done!!

colinjohnm, 2020-04-26:
Not sure if I missed something in the setup
I Sliced a file in cura and sent to Octoprint with your plgin active. It said it processed ok.

Created two files  see snip below one much smaller than the other -but it started printing the larger of the two. I stopped it and selected the file prefixed AS this started but was nothing like expected so I stopped and the opened the g-code in cura see below

I wondered if I have done something or missed something?
should it run the AS file? If so will it do this automatically from upload from cura?
please advise as keen to get this working

many thanks

 
[image]
[image]

colinjohnm, 2020-04-26:
sorry meant to add files
but dont seem to be able to add

blacksurgeon, 2020-04-26:
Maybe you need to zip the files because not all formats are accepted.

colinjohnm, 2020-04-26:
[gcodes.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/4536060/gcodes.zip)
Thanks

kulfuerst, 2020-04-26:
Cura Gcode viewer cannot display G2 and G3 commands as far as I know, so it will look messed up. Try using the octoprint gcode viewer.

FormerLurker, 2020-04-26:
Use Simplify 3d, or go to ncviewer.com.  Please note that ncviewer is not perfect (simplify3d seems to be perfect for G2/G3 rendering), and it interprets some misc gcodes as arcs sometimes, making the gcode appear not to render.  I will analyze your gcode now and will post the results.

FormerLurker, 2020-04-26:
Looks good to me !

[image]

Ignore that it hangs off the bed.  My printer is obviously smaller than yours :)

colinjohnm, 2020-04-26:
Yes I looked at Octoprint and looked ok!! Maybe I need to try again.
Would you expect Octoprint to print the AS code when uploaded from Cura??
It went straight in to print the larger file
Thanks for your quick replies

FormerLurker, 2020-04-26:
?? I'm not sure what you mean print straight from cura.  Did you upload the file, download it again, and try to upload once again through cura??  If so, why?

colinjohnm, 2020-04-26:
So just to explain after originally slicing in Cura -The Print with octoprint button appears
So pressed and it uploaded to Octoprint and Your Plugin did the processing.
But Octoprint then lined up the full fat file to print instead of the AS 

Would you not expect it to print the modified file?

Regards
Colin

[image]
[image]

FormerLurker, 2020-04-26:
Sorry, I don't control the cura upload plugin.  Whatever it is doing, it (probably correctly) assumes you want to print the file you sent.  It has no idea my plugin is modifying the gcode in the background, and it probably doesn't understand that it should wait for my preprocessing to complete before starting the print.  Is there no 'upload without printing' option?

FormerLurker, 2020-04-26:
Oh, btw, I totally understand why you expected it to work, and exactly what you expected to have happen.  I just don't know how in the world I could accomplish it without modifications to support the cura upload plugin specifically (probably both the plugin and Arc Welder would need substantial modifications).

colinjohnm, 2020-04-26:
No Option just to upload from Cura. I can save to file and then up;oad from Octoprint and deal with it there! No Problem.  I am printing at the moment so It will be tomorrow to give it another try.  Very Impressed at the file size reduction!!! Well done

colinjohnm, 2020-04-27:
NO RUSH I KNOW YOU ARE BUSY!!
Tried a smaller file to test  started ok. I only allowed it to run a few layers as a couple of things were apparent.
- When it transitioned from the long arc to the flat at the end the flat was very slow in movement and therefore seemed to over extrude
- at the centrepoint of the arc it stopped but continued extruding causing big blob
I am continuing printing without Plug In for now  - I am happy to trial further if you need or send more info
Regards 

[image]
[image]
 
[AS_CCR10_bottoms_x_5-PETG.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/4538503/AS_CCR10_bottoms_x_5-PETG.zip)

FormerLurker, 2020-04-27:
I assume you are running some fork of marlin?  If so, can you find the configuration_adv.h file for your current firmware version and send me a link?

colinjohnm, 2020-04-27:
Hi Marlin 2.0 Skr Mini E3 

Hope this helps

[Configuration_adv.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/4542140/Configuration_adv.zip)

colinjohnm, 2020-04-28:
FYI Just reviewing changing to Marlin 2.05 SKR Mini E3 C 1.2 when complete shall I re-run the test??

FormerLurker, 2020-04-28:
Yes please!

colinjohnm, 2020-04-28:
Retested with Marlin 2.05 SKR MINI E3
Same basic result -Mostly the ARCs look good except on the outer where there is a big stall !!
Also the small flat areas at the end are v.v slow
I took a vid but MP4 is 85Mb so cant email
Regards

colinjohnm, 2020-04-28:
Retested with Marlin 2.05 SKR MINI E3

Same basic result 
-Mostly the ARCs look good except on the outer where there is a big stall !!

Also the small flat areas at the end are v.v slow

I took a vid but MP4 is 85Mb so cant email

FormerLurker, 2020-04-28:
So strange.  I checked the print speeds, and they look correct.  Can you check your firmware to see if junction deviation is enabled?  There is an open marlin issue about that setting and arc stalling.  Maybe classic jerk would work?  I will research this too.

FormerLurker, 2020-04-28:
[Here is that issue, for reference.](https://github.com/MarlinFirmware/Marlin/issues/15543). It has been reported for CNC only, but i think it should apply to 3d printers too.  Not sure if this would have affected your old firmware.

colinjohnm, 2020-04-28:
So Marlin was set at

#if DISABLED(CLASSIC_JERK)
  #define JUNCTION_DEVIATION_MM 0.08 // (mm) Distance from real junction edge

The link above gives

Formula: `0.4 * sq(min(xjerk, yjerk)) / (default accel)`
and from the link inside Marlin
[image]

From my Cura
Jerk =5mm/s
Accel 500mm/s
therefore = 0.04((5 x 5)/500) = 0.02

But from the same Link in Marlin it states "The smaller the junction deviation number, the more the machine will slow down when encountering corners.  Making this number too small may slow down the print speed and cause over extrusion in the corners.  Making the number larger will increase the speed at which the machine traverses corners.  Setting this number too high may cause vibrations in the mechanism that produce ringing in the print surface or in extreme cases a stepper motor to loose position."

So by making the number smaller ie 0.02 vs 0.08 it should make the corners slower and they are very slow and overextruding.

 If I get time tomorrow (meant to be working!!) I will try at 0.02 and then comment out to get classic Jerk.

Regards

colinjohnm, 2020-04-28:
Meant to say If you have a way for me to send the MP4 (85Mb zipped) then I have a clip that shows it quite well

FormerLurker, 2020-04-28:
I think it would be best to try classic jerk before attempting a modification that will likely further slow down the corners.  If it improves the situation you may want to chime in on that marlin thread.

Also, you might try printing from SD to see if the problems still appear, just in case.  I'll give your gcode another look.  If you can produce a smaller object that  has the same issues I could try printing it on my machine (your gcode won't fit on my build plate, which is 210x250mm) to see if I have any problems.  If it prints correctly I think we can eliminate improper feedrates in the gcode file.

colinjohnm, 2020-04-29:
Marlin Set with Junction Deviation

Test 1 Single Part Sliced by Cura via Octoprint using Arc welder
Result as before v slow ends Blob in the middle stopped after 6 layers

Test 2 Single Part Sliced by Creality (V Old copy cura) via Octoprint using Arc Welder
Completed print en speeds ok. One Blob half way up the print in the inside curve near hole ~14 mins

Test 2a Single Part Sliced by Creality (V Old copy cura) via Octoprint NOT Using Arc Welder
Very Smooth not quite as fast as Test 2 but not blobs!! Interesting compared to Cura latest ver~14 mins

Test 2b Single Part Sliced by Cura via Octoprint NOT using Arc Welder
Not as clean as 2a or 2 above but smoother and better than 1 above ~15mins

Test 3 Single Part Sliced by Cura via SD Card after using Arc welder
as per 1 above

Sopped testing
I hope the results above are interesting for you
It was in particular interestind to see the Creality Slicer being better than cura in all instances especially as it is basically a very old version of cura!!! As it doesnt have all the features as cura I copied as many across to basically have the same to act as control!

I will be looking at using this on a troublesome larger print using octoprint o see how it goes.
Colin

[Test.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/4553017/Test.zip)

colinjohnm, 2020-04-29:
looking at the prints and comparing the ends the creality parts are crisp and clean in all versions the cura seems to be a bit weird. 
If I am to try the classic jerk it might have to be later in the week i didnt get round to updating the firmware.

FormerLurker, 2020-04-29:
Is is possible that Marlin does not support the F parameter for G2/G3 at all?!  

Here is the reprap definitions for G2/G3:  https://www.reprap.org/wiki/G-code#G2_.26_G3:_Controlled_Arc_Move

Here is the marlin definition:  https://marlinfw.org/docs/gcode/G002-G003.html

Notice that no mention is made of the F parameter!  Holy smoke, that would do it if it is true!  The solution would be to add a separate G1/G0 + F command before the arc, but I can hardly believe F isn't supported.  Going to dig into the code.

colinjohnm, 2020-04-29:
ok thanks

FormerLurker, 2020-04-29:
So I tried this on my printer, and F works as expected.  Can you try the following test on your printer to see if it works for yours:

```
G28 X Y ; home x and y
G0 X50 Y50 F1500 ; move to 50,50 slowly
G2 I20 J20 F600 ; VERY slow circle
G2 I20 J20 F3600 ; Reasonably fast circle
```

If you get around to it, let me know if the feedrate has an effect for the two G2 commands

FormerLurker, 2020-04-29:
I just edited my last comment with an additional command.  I wanted to make sure you saw it.

colinjohnm, 2020-04-29:
yes all worked as expected and the last line was at same speed

FormerLurker, 2020-04-29:
Man am I stumped..  I think we will need to produce a short gcode test that can trigger the same issue, and investigate exactly what the firmware is doing.  It would be great if this can all be done without any E parameters.  Perhaps you could identify the exact place (layer+ and as close to the G2/G3 that is causing the issue as possible) where a particularly bad slowdown is occurring, then we can try to create a very small gcode file that can consistently reproduce the issue.  Who knows, we might find that an F parameter is off somewhere (though I can't see that in the gcode analyzers I've used....)

colinjohnm, 2020-04-29:
Hmm I can look but not sure I will be able to Ill give it a go :)

colinjohnm, 2020-04-29:
Ill be honest Im not sure what Im doing with the gcode There was one bit in the AS Cura code that looks odd this is at one end. 

[image]

I think the stall in the middle might be about here but cant be sure
[image]

Im not sure how much time you want to look into this -its ok if you want to move on with more important items. Its not stopping me printing. I am going to be v busy over next few days as job change so I wont be able to look much. Let me know if you want me to close issue?

FormerLurker, 2020-10-17:
@colinjohnm, check out the latest version of Marlin.  Lots of arc enhancements for both speed and accuracy.  Going to close this sine the original issue is solved, and I believe the problem you are seeing is firmware related.  If you upgrade and have any problems with Arc Welder, please create a new issue.

Thanks for participating everyone!

closed 2020-10-17 by FormerLurker