FormerLurker/ArcWelderPlugin #7: [Feature Request]

KenLucke, 2020-04-26:
Please add a Before and After Filesize (or just bytes saved) to the results dialog, for information (and dancing in the streets) purposes.

FormerLurker, 2020-04-26:
I will add this soon.  Will update this issue once it is completed.  For now, just look in the file manager.

FormerLurker, 2020-04-27:
How is this:

[image]

I may add this to the progress display as well.

FormerLurker, 2020-04-27:
Here is is on the progress display:

[image]

FormerLurker, 2020-05-05:
This has been completed and released.  Closed!

closed 2020-05-05 by FormerLurker