FormerLurker/ArcWelderPlugin #8: Feature Request - Process existing files.

KenLucke, 2020-04-26:
Should be a way (am I missing it?) to process already uploaded files (and to prevent re-processing ones that have already been done), so they don't have to be deleted and re-uploaded.

FormerLurker, 2020-04-26:
No you are not missing this feature, it doesn't exist yet.  I thought about this feature from the beginning, but it's not yet available.  I'll leave this open until it is implemented.

FormerLurker, 2020-05-01:
I got done adding a 'process' button to the file manager.  It will be disabled for any files that have already been processed, and they will disable if you are printing.  Testing it out on my pi now.  Will let you know when it is ready for you to try.

FormerLurker, 2020-05-01:
OK, this as well as the other feature request you posted has been implemented.  Try installing from this URL and let me know as soon as you can if it closes both issues so I can create another release: ```https://github.com/FormerLurker/ArcWelderPlugin/archive/7801eebd91bfbd17213bf17952e274957ecfa4a3.zip```

Thanks!

KenLucke, 2020-05-01:
> OK, this as well as the other feature request you posted has been implemented. Try installing from this URL and let me know as soon as you can if it closes both issues so I can create another [...]

Hmmm.  De-installed the previous version and installed this one.   Can't see any process button for existing files.

Oh, DUH.  It was going by the disabled status of the previous plugin in the Plugin Manager.  Turning that back on made it come alive.  I forgot that even if you delete a plugin, unless you do a "clean data" or whatever it is, it just keeps your settings.

Looks good, however.  Hoping that we can get the mismatch between the gcode viewer and actual progress resolved, but I know that's a way more complicated issue in another thread.  That's a deal killer for me, I have to leave it disabled when not testing it, it's so irritating and distracting :)

Couple of things for the future:
1) the "Processing" dialog looks great now, but the "Competed" dialog doesn't specify to which file it is referring, as does the "Processing" one, so when several of the "Completed" dialogs stack up, there's no way to tell to which file they are referring, unless you remember the order in which you processed them.

2) "Processing" dialog should have an "Abort"or "Cancel" option which would bail out of that file's conversion.  If more than one file is queued, a "Cancel All" and "Cancel This File" choice would be appropriate.

FormerLurker, 2020-05-01:
Not sure what I can do to make the progress work.  That isn't connected to this plugin.  Not sure where that issue belongs honestly.  Octoprint issue maybe.  I will see if i can isolate the cause.

I will add the file name to the completed toast.  I just forgot it, since i had intended to add that.

The progress pop up does have a cancel button.  Are you not an admin?  I do not have a cancel all.  I might add that if it doesn't pose too many challenges.

KenLucke, 2020-05-01:
> Not sure what I can do to make the progress work. That isn't connected to this plugin. Not sure where that issue belongs honestly. Octoprint issue maybe. I will see if i can isolate the cause.

Have you cross-talked to the PrintTimeGenius author to see if you can collaborate on a fix?  Might not fix the main Octoprint, but if he can calculate G2/G3 times, it would certainly work around the issue (and, you can promote each others' plugins!  :) )

> 
> I will add the file name to the completed toast. I just forgot it, since i had intended to add that.
> 
> The progress pop up does have a cancel button. Are you not an admin? I do not have a cancel all. I might add that if it doesn't pose too many challenges.

I didn't see it (yes, I'm admin, and only user).  I'll look again

KenLucke, 2020-05-02:
> I will add the file name to the completed toast. I just forgot it, since i had intended to add that.

That's what alpha/beta testing is for, to find the little  nitpicky stuff we miss... if we could do and remember and think of it all, we'd all put perfect stuff out the first time :)

> The progress pop up does have a cancel button. Are you not an admin? I do not have a cancel all. I might add that if it doesn't pose too many challenges.

OK, I missed it, it's there.

FormerLurker, 2020-05-02:
Glad the cancel button showed up! Regarding the print progress issue, it is looking like the octoprint analysis needs to be rerun for overwritten files.  However, new files are working for me!  I dont have print time genius, but i will take a look.

KenLucke, 2020-05-02:
> Glad the cancel button showed up! Regarding the print progress issue, it is looking like the octoprint analysis needs to be rerun for overwritten files. However, new files are working for me! I dont have print time genius, but i will take a look.

PTG is FAR more accurate than OctoPrint for time estimates, from start to finish.  As long as it's not an AS file, that is.

> Edit:. Files with tye AS prefix seem to work is what I meant. How about on your end?
They work well on the curves, but long straights are stuttering and blobbing badly.

FormerLurker, 2020-05-02:
Regarding the AS files, the blobbing seems to be firmware related.  Im working on that.  I was talking about the stock gcode viewer without PTG.  That is working for me.

FormerLurker, 2020-05-02:
See #12 and #13.  Besides the file analysis issue and the firmware issues (stuttering when printing G2/G3), I think that should solve your feature requests.  The code has been pushed to the devel branch.

FormerLurker, 2020-05-05:
This has been completed and released.  Closed!

closed 2020-05-05 by FormerLurker