FormerLurker/ArcWelderPlugin #9: COMPLETELY messes up PrintTimeGenius

KenLucke, 2020-04-26:
Example:  I have a 77 layer, 3 mask build.  The normal error of the gcode being several layers ahead (have opened an issue there some time ago) is nothing compared to what's happening now.

The print *appears* to finish in Dashboard (except for the actual layer count indicator), OctoPod, gcode viewer,  etc., even though it is more than 10-12 layers from actually finishing (on this small file).  The print actually continues (and the dashboard's, and I can see it printing in the webcam stream, and finishes normally.

github will not allow me to upload files (even .jpg or .zip), and says "Something went really wrong, and we can't process that file.  Try again."

FormerLurker, 2020-04-26:
This issue probably belongs on the print time genius plugin page.  They probably haven't implemented G2/G3 commands (that is pretty common, as the vast majority of gcode files don't include g2/g3).  Try disabling that plugin and see what happens.

FYI, you can drag images into github (or at least you should be able to).  For other file attachments, go to gist.github.com and put each file in there individually, then share a link to the gist here.

KenLucke, 2020-04-26:
> FYI, you can drag images into github (or at least you should be able to). For other file attachments, go to gist.github.com and put each file in there individually, then share a link to the gist here.

Tried dragging and dropping, selecting (via file open),and pasting.  None work.  I'll try the gist.github.com method.

I'll disable PTG after these prints are finished and repoort back.

FormerLurker, 2020-04-26:
> I'll disable PTG after these prints are finished and repoort back.

Thanks!  I really appreciate it!  It's very important to know about plugins that don't work well together.

KenLucke, 2020-04-26:
Turned PTG off on Blue, printing now with a file re-uploaded and re-processed through AW (I'm ass/u/ming that PTG preprocesses the file in some way, so just reprinting with the same file, even with PTG off, might not be accurate)

I can't even seem to upload to gist.github.com.  Here's a dropbox link.  Late, but beter late than never, I guess.
https://www.dropbox.com/s/ei77ipr7y5n9a4z/CR-10v2%20%28Blue%29%20100%25%20L%3A56%3A77%20EET%3A12%3A49%202020-04-26%2012-49-38.jpg

KenLucke, 2020-04-27:
Disabling PrintTimeGenius makes no difference in the issue.  Right now I have a print that I was notified was "finished" over 5 minutes ago, still going (it was still 22 layers from finish according to the dashboard layer count indicator)

FormerLurker, 2020-04-27:
What version of octoprint are you using?  Also the interface in your photo, what is that exactly?

KenLucke, 2020-04-27:
> What version of octoprint are you using? Also the interface in your photo, what is that exactly?

I'm on the latest, 1.4.0.

The interface is the Dashboard plugin.

FormerLurker, 2020-10-17:
@KenLucke, this issue has aged, and I believe the issue is not related directly to arcwelder, but rather that the estimate calculations don't work well with G2/G3.  I did fix a bug that might have messed up announcement in print time genius though (ArcWelder now preserves all case), so you might want to try that.  

If you have any other issues, or would like me to reopen this one, please let me know!

closed 2020-10-17 by FormerLurker