FormerLurker/ArcWelderPlugin #10: Arc Welder plugin is not installing on OctoPrint running on Win10 or MacOSX

Prickly-Pear, 2020-04-29:
I tried to install Arc Welder on both my iMac and my Win10 laptop so that I could convert files faster than my old Rasperry Pi model 1B. OctoPrint installed well on both types of computers but Arc Welder did not install. 
This is just a small problem, at least for me, but there may be other people that would like to install Arc Welder onto PC/Mac based instance of OctoPrint. I thought I'd report it and provide you with the relevant OctoPrint logs. You can look at these on a rainy day.

[plugin_pluginmanager_console - Win10.log](https://github.com/FormerLurker/ArcWelderPlugin/files/4550535/plugin_pluginmanager_console.-.Win10.log)
[plugin_pluginmanager_console - MacOSX.log](https://github.com/FormerLurker/ArcWelderPlugin/files/4550537/plugin_pluginmanager_console.-.MacOSX.log)

KenLucke, 2020-04-30:
Worked fine for me  Mac OSX 10.5.4

FormerLurker, 2020-04-30:
Here is the issue for windows 10:
> 2020-04-28 12:19:09,737 ! c:\users\per\appdata\local\temp\pip-req-build-yb5hhv\octoprint_anti_stutter\data\lib\c\gcode_processor_lib\logger.h(7) : fatal error C1083: Cannot open include file: 'chrono': No such file or directory

Now to figure out how you could be missing this, or to just remove it from the application.

FormerLurker, 2020-04-30:
Here is the mac issue:

2020-04-28 11:11:22,902 ! octoprint_anti_stutter/data/lib/c/anti_stutter/anti_stutter.cpp:680:58: error: call to 'abs' is ambiguous
2020-04-28 11:11:22,902 ! if (!p.is_extruder_relative && utilities::greater_than(abs(absolute_e_offset_), 0.0) &&

abs is ambigious?  Two of these odd errors, lol.  Anyway, I think I can fix the abs error.  If I remove chrono we will loose MS in the error log timestamp.  Going to try in figure out a workaround.

Prickly-Pear, 2020-05-02:
Was there anything that I did wrong during installation of OctoPrint on Win10? Can I install it afterwards? Or is there a 'repair' feature that I can use?

Regarding the MacOSX issue, it sounds like it is an Arc Welder issue, right? Is there anything I can do to help with this?

FormerLurker, 2020-10-17:
@Prickly-Pear, I may have finally fixed the windows 10 issue.  I will be pushing a new version (v0.1.0rc1.dev5) to master soon.

FormerLurker, 2020-10-25:
OK, so I've been toying with the macOS build of the ArcWelderConsole application.  I do not have a mac to test, but if anyone is interested in trying it out, it can be found [here](https://github.com/FormerLurker/ArcWelderLib/actions/runs/326182285).  It very likely will not work, but it's worth a shot.

Prickly-Pear, 2020-10-26:
@FormerLurker, is there a trick to install the MacOS build? 
I tried to install through OctoPrint Plugin Manager from an uploaded zip file but it errors out. MacOS 10.15.7.
This is the error:
> 2020-10-26 18:49:10,842   /Users/iMac2011/OctoPrint/venv/bin/python -m pip --disable-pip-version-check install file:///var/folders/lv/7qg5qtvx2jxgxmt2f__hfktc0000gn/T/tmpac25aln9.zip --no-cache-dir
2020-10-26 18:49:11,857 > Processing /var/folders/lv/7qg5qtvx2jxgxmt2f__hfktc0000gn/T/tmpac25aln9.zip
2020-10-26 18:49:11,858 ! ERROR: Command errored out with exit status 1:
2020-10-26 18:49:11,858 ! command: /Users/iMac2011/OctoPrint/venv/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/lv/7qg5qtvx2jxgxmt2f__hfktc0000gn/T/pip-req-build-m_0ogz0d/setup.py'"'"'; __file__='"'"'/private/var/folders/lv/7qg5qtvx2jxgxmt2f__hfktc0000gn/T/pip-req-build-m_0ogz0d/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/lv/7qg5qtvx2jxgxmt2f__hfktc0000gn/T/pip-req-build-m_0ogz0d/pip-egg-info
2020-10-26 18:49:11,858 ! cwd: /private/var/folders/lv/7qg5qtvx2jxgxmt2f__hfktc0000gn/T/pip-req-build-m_0ogz0d/
2020-10-26 18:49:11,858 ! Complete output (5 lines):
2020-10-26 18:49:11,859 ! Traceback (most recent call last):
2020-10-26 18:49:11,859 ! File "<string>", line 1, in <module>
2020-10-26 18:49:11,859 ! File "/usr/local/opt/python/Frameworks/Python.framework/Versions/3.7/lib/python3.7/tokenize.py", line 447, in open
2020-10-26 18:49:11,859 ! buffer = _builtin_open(filename, 'rb')
2020-10-26 18:49:11,859 ! FileNotFoundError: [Errno 2] No such file or directory: '/private/var/folders/lv/7qg5qtvx2jxgxmt2f__hfktc0000gn/T/pip-req-build-m_0ogz0d/setup.py'
2020-10-26 18:49:11,859 ! ----------------------------------------
2020-10-26 18:49:11,859 ! ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

FormerLurker, 2020-10-26:
@Prickly-Pear, unfortunately I've never gotten the macOS version to build properly.  However, from your logs it looks like you might be missing the python 3 dev package.  See [this bit of the readme.md file](https://github.com/FormerLurker/ArcWelderPlugin#python-development-package-for-linux) for more details.

I really need to get a mac so I can figure these things out.

Prickly-Pear, 2020-10-31:
Thank you, @FormerLurker. 
I managed to break my OctoPrint setup on my Mac when I installed Python 3. I cannot get it to work even when I went back to Python 2.7. I have no idea of what I am doing so I will just quietly give up on ArcWelder (and OctoPrint) on my Mac.
On the other hand, ArcWelder installed happily on my Win10 laptop with OctoPrint 1.4.2 and Python 3.8 (and I had to install VSC with C++ before everything wanted to work). I will test it out and see how it compares.

FormerLurker, 2020-10-31:
Sorry to hear about your mac.  Regarding the windows install, I guess i could just compile a windows binary, but the c++ redistributable will still be required.  In any case I should probably update the installation instructions.