FormerLurker/ArcWelderPlugin #11: Possible gcode conversion problem.

Prickly-Pear, 2020-04-29:
I have converted a large gcode file using Arc Welder. The file size was reduced from 106.6MB to 36.6MB and when viewed in NC Viewer, it looked ok with no strange rapid movements.
Partway through, the print failed at layer 371 (or nearby) with a slight shift in the y-axis, about +2.5mm. The original code file printed just fine (printed from SD card, of course) and it was printed immediately after the failed print without needing to clean nozzle or anything else, thus ruling out any filament clogs or similar issues.

Original file on the left and converted file on the right.
[image]

Close-up picture shows the y-axis shift and there is also plenty of blobbing and what-not.
[image]
[image]
[image]

There is also an interesting effect in that the infill stopped printing, as can be seen in this picture:
[image]

I have inspected the converted gcode file but I cannot see anything strange. However, I am not an expert and I could easily have missed something. 

Link to converted gcode file: https://drive.google.com/file/d/12L_ssGwVAWxceoSo1AKa4dLeFE0VpFoJ/view?usp=sharing
Link to original gcode file: https://drive.google.com/file/d/17ZgFxpv9w-4E8iQ7mXbSPKGWo8V_GknQ/view?usp=sharing

Printing Info:
RaspberryPi model B with OctoPrint 1.4.0 on OctoPi 0.16.0
Slicer: Cura v4.5
Ender 3 Pro with BTT SKR mini E3 v1.2 and BLTouch
0.2 mm layer height with 0.4 mm nozzle
50 mm/s, 3 walls @ 0.4mm, infill 10%
max resolution 0.5mm, max deviation 0.05mm, relative extrusion
MM_PER_ARC_SEGMENT is 0.05mm

FormerLurker, 2020-05-01:
Thanks for posting this.  I will start investigating this as well as your mac/windows installation issues (I think I understand both of those, so it's just a matter of patching and testing).

FormerLurker, 2020-10-17:
This too should be solved.  I have run your super massive gcode through ArcWelder, and it looks like it is fixed.  I found that arcs with a radius that is too large cause overflow problems, so there is now a reasonable limit on the radius of large arcs.

FormerLurker, 2021-01-23:
FYI, this file has proved to be extremely useful for testing arc-welder in general.  It contains a lot of tricky situations, and I wanted to thank you for sharing it!  Going to close this out since you can be sure that any release of arc welder will be tested with this file :)

closed 2021-01-23 by FormerLurker