FormerLurker/ArcWelderPlugin #14: Cannot install plugin in octoprint

reinharm1959, 2020-05-05:
I tried installing Arc Welder via plugin manager as I previously was successful. Returns;

Installing plugin from https://github.com/FormerLurker/ArcWelderPlugin.git...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/FormerLurker/ArcWelderPlugin.git --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/FormerLurker/ArcWelderPlugin.git
ERROR: Cannot unpack file /tmp/pip-unpack-3KJdBY/ArcWelderPlugin.git (downloaded from /tmp/pip-req-build-qrmnTS, content-type: text/html; charset=utf-8); cannot detect archive format
ERROR: Cannot determine archive format of /tmp/pip-req-build-qrmnTS
Downloading https://github.com/FormerLurker/ArcWelderPlugin.git
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

What an I doing wrong?

FormerLurker, 2020-05-05:
You used the wrong link to install.  The .git link is for cloning source code.  You will want to use the .zip link.  Here is the link to the devel RC release I just pushed today:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/0.1.0rc1.dev2.zip```

reinharm1959, 2020-05-08:
Ok I installed the latest version and tried printing a few models with many arcs and the arc_welder version dev2 definitely reduces the file size but it prints exceptionally slow(er).  So I turned arc welder dev2 off and print times returned to "normal". I'm running arc_welder on Pi 4 with Octoprint v  1.4.0 with a Taz 6. Not sure if I need to do something or modify and code???

FormerLurker, 2020-05-19:
Sorry for the delay in responding.  It is likely an issue printing arcs with your firmware.  Unfortunately this is about as common as printers that have no problems printing arcs.  I suggest trying a test print containing many arcs directly from your SD card.  If it's still slow, it's a firmware issue.

I'm not sure what firmware LulzBot is running these days, but if you can find a link I will take a look and see if there is anything obvious in there.

FormerLurker, 2020-10-17:
This has aged a bit, so I'm going to go ahead and close this issue.  However, I wanted to mention that if you are using marlin you should check out the latest version.  There have been a lot of improvement to arc speed and accuracy!

Let me know if you want me to reopen this issue, and I will do so.

closed 2020-10-17 by FormerLurker