FormerLurker/ArcWelderPlugin #15: Conversion is stuck on Initializing...

benoit2600, 2020-05-06:
Hi, 

I'm using the 0.1.0rc1.dev2 version of this plugin, but I can't make it work :-( .

Everytime I try to do a conversion, It get stuck on  Initializing...

here is a gcode exemple that get stuck (it's on  Initializing since 15min now)
[Part Studio 1 - Part 1 (2)_0.3mm_PLA_MK3S_28m.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/4588028/Part.Studio.1.-.Part.1.2._0.3mm_PLA_MK3S_28m.zip)

I've activated the debug log, but there is nothing interesting  : 

`
cat plugin_arc_welder.log
2020-05-06 16:18:42,532 - arc_welder.__init__ - INFO - Received a new gcode file for processing.  FileName: Part_Studio_1_-_Part_1_(2)_0.3mm_PLA_MK3S_28m.gcode.
2020-05-06 16:18:42,630 - arc_welder.__init__ - INFO - Starting pre-processing with the following arguments:
        source_file_path: /home/pi/.octoprint/uploads/Part_Studio_1_-_Part_1_(2)_0.3mm_PLA_MK3S_28m.gcode
        resolution_mm: 0.100
        g90_g91_influences_extruder: False
        log_level: 10
2020-05-06 16:18:42,696 - arc_welder.gcode_conversion - INFO - Parsing GCode Conversion Args.
2020-05-06 16:18:42,698 - arc_welder.gcode_conversion - INFO - py_gcode_arc_converter.ConvertFile - Parameters received: source_file_path: '/home/pi/.octoprint/data/arc_welder/source.gcode', target_file_path:'/home/pi/.octoprint/data/arc_welder/target.gcode' resolution_mm:0.1, g90_91_influences_extruder: False

2020-05-06 16:18:42,699 - arc_welder.gcode_conversion - INFO - py_gcode_arc_converter.ConvertFile - Beginning Arc Conversion.
`

I'm using a raspberry pi 3 with octopi on it.

FormerLurker, 2020-05-06:
Are you running Python 3 by chance?  Also, can you reboot and see if the converted file actually exists in the file manager?  I'm trying to narrow the problem to the conversion or to the toast/ui updates, and there is a known issues in Python 3 (requires OctoPrint fix) that will prevent the progress bar and complete notifications from working.  Also, if you are using the 'overwrite existing file' setting, check the file you uploaded to see if there was a reduction in size (download it to make sure).

benoit2600, 2020-05-06:
Than you for your quick answer !

I'm not 100% sure I'm not using python3, but I'm pretty confident. octoprint is launched with /home/pi/oprint/python2, and python open python v2.7.13.

No unfortunaltely there is no new file after rebooting the pi :( Oh and by the way, this is the only way to get back octopi working. When it's on Initializing, nothing is responding anymore. I need to log on SSH and reboot from there.

And I'm not using Overwrite Source File.

FormerLurker, 2020-05-06:
@benoit2600, that means the C++ call is crashing for some reason.  I ran your gcode through my python 2.7 instance, and it is converting, which is unfortunate.  Had it crashed for me I could probably easily determine what is going on.  I will look again as soon as time permits (work day).  Perhaps in the meanwhile you could enable verbose logging and see if the log file changes?  I will likely have to add additional logging to the python extension to narrow down the problem.

benoit2600, 2020-05-07:
Hi @FormerLurker , 

The logs for this plugins are already set to debug. Do you want me to re-try with the same verbose level on octoprint.log ?

FormerLurker, 2020-05-19:
Hey, sorry it has been a while since I got back to you.  I'm just about to drop rc4 (in minutes, check the releases tab), and have made some changes that might have solved this.  If not, I added more logging to the conversion routine, which hopefully will help get this solved.

benoit2600, 2020-05-20:
It works now :-)  !!

Thank you ! 

I've attached the log of a conversion. I've not try yet to print it, but the conversion process go to the end !

[plugin_arc_welder.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/4659112/plugin_arc_welder.zip)

closed 2020-05-20 by benoit2600

FormerLurker, 2020-05-20:
Awesome 👍 let me know how it goes in the new issue i created for feedback for rc4 please!