FormerLurker/ArcWelderPlugin #16: Problems with d<3mm

EddyMI3d, 2020-05-19:
I recently printed some stringing tests.
While printing via Arc Welder it appears, that obviously Marlin can't handle diameters smaller than 3mm:
[image]
The red test was printed with Vivedino T-Rex 3 and Marlin 2.0.5 and the yellow one with Prusa MK2.5S and MMU2S with firmware 3.9.0-RC3.
The distortions are not random, they are similar on each pillar and similar on both objects. 
The gcode appears as normal circles in the OctoPrint gcodeviewer.
Or do I have to reduce the resolution in Arc Welder smaller the 0.05mm?

FormerLurker, 2020-05-19:
You can adjust the arc welder resolution to whatever you like in the settings, but the resolution is not your issue.  The number of arc segments generated by your firmware is likely the issue.

I'm quite surprised you got such horrible results from the MK2.5S with MMU2S, because that's exactly the printer I use!  What version of ArcWelder are you using?  If you didn't install from one of the release links, you might have just gotten bad code from a bad branch (I broke some things over the weekend, now fixed).  

Also, posting the original and modified gcode would be very useful (please use [gist.github.com](https://gist.github.com) to post the gcode).

Thanks!

EddyMI3d, 2020-05-20:
So I actually had dev2 running. I did it again with dev4 with the same result. 
I 100% agree to you that it is to be located in the firmware as both printers show the same result.

[OozingTest.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/4654875/OozingTest.zip)

[image]

[image]

Are circles are sent as complete circles? Maybe Marlin gets confused by that small numbers. The prints look as Marlin tries to print them - let me say - inside out.

FormerLurker, 2020-10-17:
Can you try out Marlin 2.0.0.6 with the latest version of Arcwelder (master branch is good)?  I think this should be fixed.  Let me know!

EddyMI3d, 2020-10-17:
Yep, I'll give it a try!

EddyMI3d, 2020-10-17:
So I also reduced the resolution to 0.01 mm and I got fine prints

FormerLurker, 2020-10-17:
That is AWESOME!  Getting so close to a release now that Marlin has MUCH better compatibility.  I'm adding some additional features and trying to prepare for the next release of Octoprint.  Then it's just some documentation and submission to the plugin repo.  

Thanks so much for participating and ESPECIALLY for testing.  That is super helpful.

Closed!

closed 2020-10-17 by FormerLurker