FormerLurker/ArcWelderPlugin #19: Feedback for v0.1.0rc1.dev4.

FormerLurker, 2020-05-19:
Please let me know here how your experience with this version went.  I'm interested in hearing about successes, failures, what you like, what you hate, what worked, what didn't, or anything else you are willing to share.  I'm getting close to a release, so all feedback is helpful and welcome!

xeddog, 2020-05-28:
I haven't done any comprehensive side-by-side testing yet, but so far I like what I see.  But I have a couple of feature requests.
1.  When uploading a file to Octoprint, ArcWelder puts out a dialog box with some conversion information.  Any way to put a timer on that so it will go away by itself after a while?

2.  Selective conversion.  It might be nice to be able to upload files without running AW.  Especially if your prints don't have any, or very few, non-straight surfaces.  Maybe just upload and then have a button to convert.

(did you ever consider ARCham Asylum for the name>   :-)   Probably not.  Copyright infringement.

FormerLurker, 2020-05-28:
For 1, you can just disable the popup.  Check the settings.  It is funny you ask about this, because someone else asked to have them stay open (it used to close after a while), lol!

2 is already implemented.  Check the settings and look for a drop down box.

xeddog, 2020-05-28:
Thanks for the reply, I guess this just shows that you can't please everyone.  

So, re: the popup, maybe a setting to allow the user to specify a timeout value, with a value of 0 to leave it open?

And the selective conversion, in the setting I only see a button to enable/disable the plugin.  There are no new buttons in the file browser.  Maybe I got an old version??  I just installed a couple of days ago.

[image]

FormerLurker, 2020-05-28:
@xeddog, what version are you using?  Also, I will consider adding an option to the popup controls controlling how long they are visible.

xeddog, 2020-05-28:
Plugin manager says "(0.1.0rc1.dev1)

xeddog, 2020-05-28:
Hang on.  Missed a step in the installation.

FormerLurker, 2020-05-28:
@xeddog, I notice you are using Print Time Genius.  A patch was just applied to improve print time estimation when using G2/G3 commands.  Check out the issue [here](https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/195#issuecomment-635135493), and perhaps consider helping to test the patch!

Also, you must have gotten an old link from somewhere.  Try installing from this release archive url:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/0.1.0rc1.dev4.zip```.  I've added this to the OP.

xeddog, 2020-05-28:
I will look into the Print Time Genius, thanks.

xeddog, 2020-05-28:
I guess having the right version helps a tad.  dev4 seems to have what I wanted.

jhstatewide, 2020-06-02:
> Please let me know here how your experience with this version went. I'm interested in hearing about successes, failures, what you like, what you hate, what worked, what didn't, or anything else you are willing to share. I'm getting close to a release, so all feedback is helpful and welcome!

I'm lucky enough to have a setup where enabling ARC is no problem. :) This plugin kicks ass... awesome idea, works great... EASY SETUP. I'm getting awesome results with curved surfaces now, no stutter at all with GCODE. I did have to patch the G2.cpp as you suggested in your docs, but that was pretty easy.

REALLY looking forward to V1.

PS. You should also release a static bin so people can just run GCODE through it without Octoprint. This idea is too cool!

FormerLurker, 2020-06-02:
@jhstatewide, thanks for your compliments, and enjoy this link: https://github.com/FormerLurker/ArcWelderLib/actions/runs/107579357

I don't think the mac code is working (I don't own one to test), buy the windows and linux builds work.

jhstatewide, 2020-06-07:
> @jhstatewide, thanks for your compliments, and enjoy this link: https://github.com/FormerLurker/ArcWelderLib/actions/runs/107579357
> 
> I don't think the mac code is working (I don't own one to test), but the windows and linux builds work.

Thanks for the link to the standalone bins. I was able to build them myself. I have your plugin in use across two different printers. After a few prints on each over a couple days, absolutely zero issues. Smaller gcode, everything working great. Keep up the good work!

FormerLurker, 2020-06-07:
@jhstatewide,

Fantastic, thank you for reporting this!  I am probably going to release this plugin within the next week or two since I haven't heard about any major issues with either the plugin or the resulting gcode (aside from some firmware issues, but I cannot fix all of those).

xeddog, 2020-06-10:
Minor detail, but I just noticed that when "arc'ing" a file in Octoprint, you apparently don't check to see if there is already an existing file with the same name for the output file.

FormerLurker, 2020-06-10:
This is a hard one.  I frequently reslice and upload files.  I asked myself, would I want to answer yes in a confirmation box every time I do this?  Since a reslice is usually due to a mistake, and by that time I am already frustrated, i figured it would annoy me. Then, i wondered if I would ever care if one was overwritten.  This may vary, but I figured no.  The final decision came when I realized that octoprint itself doesn't complain and will gladly overwrite an old upload with a file of the same name.

What do you think about my thought process here.  Should I revisit this?

xeddog, 2020-06-10:
Nah, let this one go.  It was trivial anyway, and now that I think about it, I would have had to respond to about 50 of those too.  I was also thinking that the ability to select a different folder might be a good idea, like maybe a setting in the configuration.  Then I thought about what a pita to be having to switch back and forth. 

Thank you for the response.

NickRno77, 2020-07-26:
I’m probably asking the impossible, I’ll ask anyway. Sending the gcode from Prusaslicer to octoprint to print directly doesn’t get interrupted by the ArcWelder. I’m guessing that is an impossibility?

 I’ve only been using ArWelder of a day and so far so good seems to have stopped the stuttering issues I was getting on my print, the print had text and a large radius. Thanks.

FormerLurker, 2020-07-26:
It may be possible actually.  I will look into it.  You aren't the first to ask for this, btw, so it is definitely worth looking into.

xeddog, 2020-07-27:
I think I am back in the camp of wanting that final statistics display you present after welding to go away after some period of time.  Perhaps a setting for a timer with a range of say . . . 0-60 seconds where 0 will not remove it.  I do most of my slicing and welding from a different room and start the print from that same place.  After a while I will go check just to make sure the print is going well, and sometimes I need to cancel it because it just ain't right.  But I can't get to the cancel button with first having to delete two or three or four of those panels.

Second item, is DEV4 still the latest?

FormerLurker, 2020-07-29:
I will take a look at adding a timer to the notifications.  You could simply turn them off if you go into the settings as a work around for the time being.  I'm also considering adding the statistics to the file metadata and making all popups disappear after a short time.  That would reduce the number of settings related to notifications.

Regarding the latest, Dev4 is nearly perfect bug wise (no confirmed arc rendering bugs at all), but there are some enhancements in the devel branch you are free to install using this url:  https://github.com/FormerLurker/ArcWelderPlugin/archive/70c2929bdaf5ee84d59c978979116c8354076664.zip

There are two primary enhancements in the devel branch:

1.  Some compiler fixes (still needs work for some older compilers and some OSs...)
2.  Greatly enhanced gcode case preservation.  The current dev release will change the case of the gcode in several cases.  Typically this doesn't cause any issues, but some plugins and scripts are not case insensitive, so I've attempted to preserve as much case as possible.

Both of these items have been tested by several users.  I am hoping to deploy an RC soon.  I think it's getting to be time to actually release this plugin.

NickRno77, 2020-07-29:
> I will take a look at adding a timer to the notifications. You could simply turn them off if you go into the settings as a work around for the time being. I'm also considering adding the statistics [...]

DEV4 is going great although I did get the stutters again with file attached, the file had a close honeycomb pattern, I was using a 0.6 nozzle. It’s not really a problem but thought you may like to look at.

https://drive.google.com/file/d/1AvOmJiRDEP9qYmJXABW0HL8rM3gCY3YU/view?usp=drivesdk

Also something you might be able to add in the “output file saving” options. What would be nice would have the option to rename the converted file with the .aw the deleted the original source file. 

I’ve just installed DEV5 so should be great 👍🏻....[](url)

FormerLurker, 2020-10-17:
Going to drop dev 5 final soon, closed!

closed 2020-10-17 by FormerLurker

FormerLurker, 2020-10-17:
@xeddog, I think I've solved the statistics display issue you were talking about.  Now all info panels close automatically after a few seconds, and you can view the final statistics for any newly welded file in the arcwelder tab by either selecting the file, or by clicking on the 'arc welder' button in the file manager panel.  Statistics will automatically display (with a show/hide button) after processing completes too.

FormerLurker, 2020-10-18:
@NickRno77, I just added these delete options per your request:

[image]

It cannot delete a printing file, and it won't delete the source file if it is overwritten, or if there is no target prefix/postfix (same as the overwrite option).

You can install from this commit archive link within the plugin manager:

```https://github.com/FormerLurker/ArcWelderPlugin/archive/62a38d3f25f6b3f59516a8598b89f0e8ecc148d5.zip```

Please let me know what you think in [this issue](https://github.com/FormerLurker/ArcWelderPlugin/issues/37).

xeddog, 2020-10-21:
> @xeddog, I think I've solved the statistics display issue 

As Chevy Chase once said "IIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIIiiiii likeit!"