FormerLurker/ArcWelderPlugin #21: Printing full circles ends up printing dots

girrrrrrr2, 2020-05-20:
I sliced a cylinder (testing z height calibration) and made it 15x15x100 tall.

The circles exterior seems to only print as a small dot. 
[Shape-Box.aw.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/4654507/Shape-Box.aw.zip)
I attached the full file before and after, converted using default settings.

(the file also has a 100mm box on x and y to calibrate all 3 at once)

I can get you Slic3r++ settings if you want.

not sure exactly what is causing it, but if its the fact that its a full-ish circle, i wonder if this can be fixed by just making it split the arc if its more then 80% of a full circle.

girrrrrrr2, 2020-05-20:
Its interesting that the preview seems to show that its attempting to print a full circle, but then the printer just doesnt.

I wonder if this is something in marlin. Ill take a look tomorrow

katherinehackworth, 2020-06-30:
There is certainly the possibility that Marlin arc support is buggy, and it just hasn't been caught until now because nothing supports arcs except this atm.

girrrrrrr2, 2020-06-30:
That too. I'm assuming it's because it's basically trying to make a full circle.

When I was testing it I don't think I had circle support on. Which could be the issue. Sadly I moved right after I started playing with it so I'm still setting back up. To test. 
On Monday, June 29, 2020 Katie Hackworth <reply@reply.github.com> wrote:

There is certainly the possibility that Marlin arc support is buggy, and it just hasn't been caught until now because nothing supports arcs except this atm.

FormerLurker, 2020-10-17:
Hey, I recommend trying the latest version of marlin.  Arc support has been greatly enhanced in both speed and accuracy.  Let me know how it goes!

girrrrrrr2, 2020-10-17:
Honestly I only had this issue around the time that I submitted the issue. Since then I dont think ive had the issue.

Plus I cant even replicate the issue, the printer that had this issue, has had a board swap and converted to Klipper.

FormerLurker, 2020-10-17:
Ok, cool.  Closing this out, thanks!

closed 2020-10-17 by FormerLurker