FormerLurker/ArcWelderPlugin #22: M117 lines unnecessarily capitalized

corporategoth, 2020-05-31:
I noticed the processed gcode from arc welder capitalizes the text on M117 lines.

This, while on one hand is cosmetic, DOES mean that plugins like DisplayLayerProgress, which inserts a M117 line at each layer (and stops itself re-processing files it has already processed) cannot recognize the layer M117 lines it put in, because they have changed from `M117 INDICATOR-Layer1` to `M117 INDICATOR-LAYER1`

There is no reason to change the capitalization (or indeed, anything) in M117 lines during the arc welding processing.

FormerLurker, 2020-05-31:
Can you post your source gcode file to gist.github.com and leave a link here?  I wasn't aware of the behavior you described, and have attempted to create a patch, but I'd like to run your gcode through my debugger to make sure it's working.

FYI, the code I'm using to reformat the gcode comes from Octolapse, which wasn't originally intended to rewrite, only to recognize gcodes, so I'm glad you pointed this out.   Frankly, I'm pretty surprised this hasn't been noticed before, so thanks again for your report!

FormerLurker, 2020-05-31:
Alternatively, you could also install Arc Welder from this link and run your file through the processor to see if this is resolved:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/devel.zip```

I'd sill appreciate it if you could post your file, but this may solve your problem.  I'm running some tests right now to make sure this didn't have any unintended consequences.

corporategoth, 2020-05-31:
I tried using gist.  It didn't work so well.

[test_print.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/4708688/test_print.zip)

Here is a print I did recently.  You can see that DisplayLayerProgress ran on test_print.gcode (this was re-downloaded from Octoprint), then ArcWelder went to work, and it capitalized all the indicators, which means DisplayLayerProgress had to go over it again, and add duplicate indicators :)

DLP is smart enough to skip running if it's already run.  So, while it's not THAT much CPU, it's more than needs be :)

FormerLurker, 2020-05-31:
Cool, i will run it through the debugger tomorrow.  In the meanwhile, try installing from the dev link i sent last time and let me know if it is working for you.  In any case, i will test with your gcode too.

FormerLurker, 2020-10-17:
Hey, I finally got this fixed.  Going to push the release here soon.  You should see it in version v0.1.0rc1.dev5, which I'm going to drop shortly.  Better late than never :)

closed 2020-10-17 by FormerLurker