FormerLurker/ArcWelderPlugin #23: Klipper: G2/G3 only supports relative extrude mode

Andrei-Pozolotin, 2020-05-31:
1. sample octoprint log:
```
Send: N24 G1 X70.976 Y50.090 E2.77133*84
Recv: ok
Send: N25 G3 X80.186 Y46.773 I23.480 J50.752 E3.40641*94
Recv: !! G2/G3 only supports relative extrude mode
Changing monitoring state from "Printing" to "Cancelling"
```

2. klipper docs:
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#g-code-arcs

3. kilpper host code:
https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/gcode_arcs.py

FormerLurker, 2020-05-31:
Isn't this a klipper issue and not an arcwelder issue?  Absolute extrusion is valid for all G1/G2/G3 codes according to most gcode definitions (though there really isn't a standard).  I recommend creating a feature request within the Klipper github page requesting absolute extrusion support for arcs.  It seems really strange to me that this wouldn't already be implemented, but so few people use G2/G3 currently this might have just slipped through the cracks.

To fix your own issue, just reslice your file using relative extrusion.

Andrei-Pozolotin, 2020-06-02:
klipper project confirmed they own this issue 
https://github.com/KevinOConnor/klipper/issues/2943#issuecomment-637079508

closed 2020-06-02 by Andrei-Pozolotin

FormerLurker, 2020-06-02:
Excellent, thanks for letting me know.  I am so glad arc support is becoming more widespread!

FormerLurker, 2020-06-02:
Wow, it looks almost exactly like the marlin code.  I may also attempt to implement a gcode (M214) for setting arc interpolation parameters.