FormerLurker/ArcWelderPlugin #24: Install Fails on Octprint 1.4.0 running on Windows 10 under virtualenv / Python

msteele999, 2020-06-08:
Hello,

I attempted to install your plugin version 0.1.0rc1.dev4 on OctoPrint 1.4.0 running on Windows 10 Home.

It looks like it failed trying to open a file called 'chrono'. I have attached my plugins log file. Is this something you think can be rectified? Am I missing another file that I need to install manually?

best regards,

Mark
[plugin_pluginmanager_console.log](https://github.com/FormerLurker/ArcWelderPlugin/files/4748754/plugin_pluginmanager_console.log)

FormerLurker, 2020-06-08:
Hmm.  What C++ compiler do you have installed?  VC++11 is supported in Visual studio 2010 with SP1 or in all versions of Visual Studio 2012 and above.

FYI, I may try to remove chrono since it has caused issues before, but I can't seem to find any other way to get milliseconds in all the targeted OSs.

msteele999, 2020-06-08:
I have Visual C++ for Python installed per the instructions I have on
installing OctoPrint under Windows (
https://www.microsoft.com/en-us/download/details.aspx?id=44266)

Looks like it's version 9.0.0.30729

msteele999, 2020-06-08:
I'm installing C++ 14.x now and will try compiling again

FormerLurker, 2020-06-08:
Hmm.  It looks like that is vs2008.  Let me think on this for a while and see if I can remove that dependency.  In the meanwhile, you could try upgrading to python 3, and install a newer version of visual c++ if you don't want to wait.

Fyi, that compiler has given me so much grief, lol!  I will be happy when python 2.7 is dead and gone :)

FormerLurker, 2020-06-08:
> I'm installing C++ 14.x now and will try compiling again

Rodger that.  Good luck, and let me know how it goes.  In any case I will try to make it work with the 2008 version.  Thx for letting me know!

msteele999, 2020-06-08:
I think I tried Python 3.0 before but let me see what happens

FormerLurker, 2020-06-08:
Yeah, it may just work.  The 2008 version is fully compatible, but that doesn't mean a newer one wont work.

Also, if you do try python 3, I recommend completely uninstalling python 2.  It is possible to get them both working at the same time, but it is a hassle and can cause some difficult to debug problems.

msteele999, 2020-06-08:
trying C++ update first - I actually installed it but on my host system and
not the Compute Stick where OctoPrint is running - derp

msteele999, 2020-06-08:
Gah - now to figure out which package to install - way too many options .

msteele999, 2020-06-08:
No joy - it keeps trying to use C++ for Python 2.7 - it must be buried in a
config file somewhere.

I'll give Python 3 a try again

FormerLurker, 2020-06-08:
You could uninstall the other compiler first

msteele999, 2020-06-08:
Trouble is I am not sure I am actually downloading the correct C++ compiler
- visual studio has several dozen options - not sure which one to select yet

msteele999, 2020-06-08:
No joy on the Python3 install - I'll keep playing around with it

msteele999, 2020-06-23:
Still no luck - any thoughts on removing that dependency to see if it will complete the installation? I'm happy to beta test this for you :-)

FormerLurker, 2020-06-23:
Turns out getting milliseconds in C++ in a cross platform way is nearly impossible without 1000s of lines of code.  I am actively working on this.  Your windows installation is the first that has had problems with chrono.  I believe I've found a fix for macos.

Once I get this done (it's been a crazy couple of weeks here) I'll send you a link to install.  If that fails, you can just uninstall the VCC for python that you installed and try again.  I'd much prefer to test this out before you do that because things are working properly on my machine (even using the old compiler), so the reason yours fails is a mystery.

msteele999, 2020-06-23:
Excellent - I'll wait to hear from you.

Mark

FormerLurker, 2020-06-23:
Mark, try installing from this URL:  https://github.com/FormerLurker/ArcWelderPlugin/archive/devel.zip

I can't promise it will work, and the MS aren't clock accurate (good enough I believe).  The MS will increment independently from the S unfortunately, but it may be good enough since this only affects console applications.

msteele999, 2020-06-23:
Failed again -

2020-06-23 12:57:07,703   c:\octoprint\venv\scripts\python.exe -m pip --disable-pip-version-check install c:/users/mstee/appdata/local/temp/tmpxo7rqr.zip --no-cache-dir
2020-06-23 12:57:10,750 > Processing c:\users\mstee\appdata\local\temp\tmpxo7rqr.zip
2020-06-23 12:57:18,766 > Requirement already satisfied: six in c:\octoprint\venv\lib\site-packages (from Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (1.15.0)
2020-06-23 12:57:18,766 > Requirement already satisfied: OctoPrint>1.3.8 in c:\octoprint\venv\lib\site-packages (from Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (1.4.0)
2020-06-23 12:57:18,766 > Requirement already satisfied: setuptools>=6.0 in c:\octoprint\venv\lib\site-packages (from Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (44.1.0)
2020-06-23 12:57:18,766 > Requirement already satisfied: filetype<2,>=1.0.5 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (1.0.7)
2020-06-23 12:57:18,766 > Requirement already satisfied: Flask-Login<0.5,>=0.4.1 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (0.4.1)
2020-06-23 12:57:18,766 > Requirement already satisfied: frozendict<1.3,>=1.2 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (1.2)
2020-06-23 12:57:18,766 > Requirement already satisfied: regex!=2018.11.6 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (2020.5.14)
2020-06-23 12:57:18,766 > Requirement already satisfied: pylru<1.3,>=1.2 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (1.2.0)
2020-06-23 12:57:18,766 > Requirement already satisfied: werkzeug<0.17,>=0.16 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (0.16.1)
2020-06-23 12:57:18,766 > Requirement already satisfied: tornado==4.5.3 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (4.5.3)
2020-06-23 12:57:18,766 > Requirement already satisfied: watchdog<0.10,>=0.9.0 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (0.9.0)
2020-06-23 12:57:18,766 > Requirement already satisfied: netifaces<0.11,>=0.10.9 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (0.10.9)
2020-06-23 12:57:18,766 > Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (1.6.5)
2020-06-23 12:57:18,766 > Requirement already satisfied: monotonic<1.6,>=1.5; python_version < "3" in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (1.5)
2020-06-23 12:57:18,766 > Requirement already satisfied: Jinja2<2.9,>=2.8.1 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (2.8.1)
2020-06-23 12:57:18,766 > Requirement already satisfied: netaddr<0.8,>=0.7.19 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (0.7.19)
2020-06-23 12:57:18,766 > Requirement already satisfied: cachelib<0.2,>=0.1 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (0.1)
2020-06-23 12:57:18,766 > Requirement already satisfied: semantic-version<2.9,>=2.8 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (2.8.5)
2020-06-23 12:57:18,766 > Requirement already satisfied: rsa<5,>=4.0 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (4.0)
2020-06-23 12:57:18,766 > Requirement already satisfied: wrapt<1.12,>=1.11.2 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (1.11.2)
2020-06-23 12:57:18,766 > Requirement already satisfied: blinker<1.5,>=1.4 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (1.4)
2020-06-23 12:57:18,782 > Requirement already satisfied: typing<4,>=3.7.4.1; python_version < "3" in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (3.7.4.1)
2020-06-23 12:57:18,782 > Requirement already satisfied: Flask-Assets<0.13,>=0.12 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (0.12)
2020-06-23 12:57:18,782 > Requirement already satisfied: sarge==0.1.5post0 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (0.1.5.post0)
2020-06-23 12:57:18,782 > Requirement already satisfied: emoji<0.6,>=0.5.4 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (0.5.4)
2020-06-23 12:57:18,782 > Requirement already satisfied: chainmap<1.1,>=1.0.3; python_version < "3" in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (1.0.3)
2020-06-23 12:57:18,782 > Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (1.5.0.1)
2020-06-23 12:57:18,782 > Requirement already satisfied: Flask-Babel<0.13,>=0.12 in c:\octoprint\venv\lib\site-packages (from OctoPrint>1.3.8->Arc-Welder==0.1.0rc1.dev5+u.49beaa2) (0.12.2)
[... cut ...] ; __file__='"'"'c:\\users\\mstee\\appdata\\local\\temp\\pip-req-build-4idej_\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'c:\users\mstee\appdata\local\temp\pip-record-bkfu1g\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\octoprint\venv\include\site\python2.7\Arc-Welder' Check the logs for full command output.

FormerLurker, 2020-06-23:
Your compiler really hates modern C.  If I fix these errors, there will be 100s more that show up next time.  to_string and std:array are apparently also not supported.  Just uninstall  C++ for Python 2.7 and see what happens.

FormerLurker, 2020-06-23:
Edit:

> Your compiler really hates modern C

If you consider a 10 year old version modern, lol.

msteele999, 2020-06-23:
Will do

msteele999, 2020-06-23:
Rock - meet hard place:

error: Microsoft Visual C++ 9.0 is required. Get it from http://aka.ms/vcpython2

Which then installs the version of C++ that your plugin hates :-)

FormerLurker, 2020-06-23:
Wait, aren't you running python 3?  Something must be wrong with your environment variables.  Search for 'Environment Variables' and look for this:

[image]

Make sure that is pointing to your python 3 installation.  then reboot.  Let me know how it goes!

FormerLurker, 2020-06-23:
OH, also, look at the path and remove all references to Python27 and replace with the python 3 paths.

msteele999, 2020-06-23:
No - I was not able to get P3 running - I'm re-doing that now - basically I
am installing a subset of Visual Studio now to get the proper 14.0 C++
compiler - I may get this licked before the end of the day

FormerLurker, 2020-06-23:
This might help with installation: https://wiki.python.org/moin/WindowsCompilers

msteele999, 2020-06-23:
I have python3 running!

Will test your plugin soon

msteele999, 2020-06-23:
The log shows the plugin installed - however during the install it said it
failed with an unknown error.

The plugin is showing as disabled. When I enable it the server restarts and
it looks like it stuck.

2020-06-23 15:09:17,414   c:\octoprint\venv\scripts\python.exe -m pip --disable-pip-version-check install https://github.com/StefanCohen/OctoPrint-Dashboard/archive/master.zip --no-cache-dir
2020-06-23 15:09:20,875 > Collecting https://github.com/StefanCohen/OctoPrint-Dashboard/archive/master.zip
2020-06-23 15:09:20,875 > Downloading https://github.com/StefanCohen/OctoPrint-Dashboard/archive/master.zip
2020-06-23 15:09:24,531 > Requirement already satisfied: OctoPrint in c:\octoprint\venv\lib\site-packages (from OctoPrint-Dashboard==1.14.0) (1.4.0)
2020-06-23 15:09:24,531 > Requirement already satisfied: psutil in c:\octoprint\venv\lib\site-packages (from OctoPrint-Dashboard==1.14.0) (5.6.7)
2020-06-23 15:09:24,531 > Requirement already satisfied: netifaces<0.11,>=0.10.9 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.10.9)
2020-06-23 15:09:24,531 > Requirement already satisfied: tornado==4.5.3 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (4.5.3)
2020-06-23 15:09:24,531 > Requirement already satisfied: netaddr<0.8,>=0.7.19 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.7.20)
2020-06-23 15:09:24,531 > Requirement already satisfied: Flask-Login<0.5,>=0.4.1 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.4.1)
2020-06-23 15:09:24,531 > Requirement already satisfied: frozendict<1.3,>=1.2 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.2)
2020-06-23 15:09:24,531 > Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.5.0.1)
2020-06-23 15:09:24,531 > Requirement already satisfied: watchdog<0.10,>=0.9.0 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.9.0)
2020-06-23 15:09:24,531 > Requirement already satisfied: rsa<5,>=4.0 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (4.6)
2020-06-23 15:09:24,531 > Requirement already satisfied: regex!=2018.11.6 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (2020.6.8)
2020-06-23 15:09:24,531 > Requirement already satisfied: Flask-Babel<0.13,>=0.12 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.12.2)
2020-06-23 15:09:24,531 > Requirement already satisfied: requests<3,>=2.22.0 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (2.24.0)
2020-06-23 15:09:24,531 > Requirement already satisfied: pylru<1.3,>=1.2 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.2.0)
2020-06-23 15:09:24,531 > Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.6.5)
2020-06-23 15:09:24,531 > Requirement already satisfied: flask<0.13,>=0.12 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.12.5)
2020-06-23 15:09:24,531 > Requirement already satisfied: semantic-version<2.9,>=2.8 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (2.8.5)
2020-06-23 15:09:24,531 > Requirement already satisfied: websocket-client<0.57,>=0.56 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.56.0)
2020-06-23 15:09:24,531 > Requirement already satisfied: sentry-sdk==0.13.2 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.13.2)
2020-06-23 15:09:24,531 > Requirement already satisfied: Jinja2<2.9,>=2.8.1 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (2.8.1)
2020-06-23 15:09:24,531 > Requirement already satisfied: werkzeug<0.17,>=0.16 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.16.1)
2020-06-23 15:09:24,546 > Requirement already satisfied: markdown<3.2,>=3.1 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (3.1.1)
2020-06-23 15:09:24,546 > Requirement already satisfied: sarge==0.1.5post0 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.1.5.post0)
2020-06-23 15:09:24,546 > Requirement already satisfied: blinker<1.5,>=1.4 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.4)
2020-06-23 15:09:24,546 > Requirement already satisfied: Flask-Assets<0.13,>=0.12 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.12)
2020-06-23 15:09:24,546 > Requirement already satisfied: future<0.19,>=0.18.2 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.18.2)
2020-06-23 15:09:24,546 > Requirement already satisfied: wrapt<1.12,>=1.11.2 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.11.2)
2020-06-23 15:09:24,546 > Requirement already satisfied: filetype<2,>=1.0.5 in c:\octoprint\venv\lib\site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.0.7)
[... cut ...]
2020-06-23 15:17:00,868 > Requirement already satisfied: itsdangerous>=0.21 in c:\octoprint\venv\lib\site-packages (from flask<0.13,>=0.12->OctoPrint->OctoPrint-PrettyGCode==1.1) (1.1.0)
2020-06-23 15:17:00,868 > Requirement already satisfied: Babel>=2.3 in c:\octoprint\venv\lib\site-packages (from Flask-Babel<0.13,>=0.12->OctoPrint->OctoPrint-PrettyGCode==1.1) (2.8.0)
2020-06-23 15:17:00,868 > Requirement already satisfied: chardet<4,>=3.0.2 in c:\octoprint\venv\lib\site-packages (from requests<3,>=2.22.0->OctoPrint->Octo

FormerLurker, 2020-06-23:
The log you posted doesn't appear to show the install log for ArcWelder.  Can you see if you can generate that for me?

msteele999, 2020-06-23:
interesting - that was the entire log - I'll look to see if I can find
something more

FWIW - it's working and I am running my first print after processing

FormerLurker, 2020-10-17:
FYI, I just added some additional compiler support, so hopefully we don't run into this issue again :)  

Going to close this now, but let me know if you have any other issues!

closed 2020-10-17 by FormerLurker