FormerLurker/ArcWelderPlugin #25: Feature request:  Make this a Cura plugin

xeddog, 2020-06-09:
This probably isn't the best place for this, but ArcWelder would be a great Cura plugin.

FormerLurker, 2020-06-09:
Yeah, there is already a command line version.  This has been discussed, and I'm glad it's now in the issues.

LeandroMarceddu, 2020-10-25:
> Yeah, there is already a command line version. This has been discussed, and I'm glad it's now in the issues.

Any more info on this? I'd like to include this plugin or well base ArcWelder into the postprocessing section of my slicer.

Charly333, 2020-10-26:
For me too as a cura or ideamaker plugin it would do a much better job.

FormerLurker, 2020-10-26:
I am currently creating a guide for integrating the console app with Cura, Slic3r and PrusaSlicer.  I plan to eventually make a video.  It's possible I will create a dedicated Cura plugin in the near future depending on how things go.

LeandroMarceddu, 2020-10-26:
To all looking for this: 

1) Clone https://github.com/FormerLurker/ArcWelderLib
2) Open up with visual studio and compile ArcWelderConsole
3) make sure it's set to x86 and not ARM, this was so in my case and had me going around for an hour as to why it wouldn't compile, heh.
4) A debug map will appear in the root of ArcWelderLib, inside will be the program ArcWelderConsole.exe. 
5) Open up a CMD and navigate to it; ArcWelderConsole.exe --help will get you started.

After that you're able to integrate it by postprocessing inside your preferred slicer (if supported, if not you'll have to do it manually).

> I am currently creating a guide for integrating the console app with Cura, Slic3r and PrusaSlicer. I plan to eventually make a video. It's possible I will create a dedicated Cura plugin in the near future depending on how things go.

I hope I'm allowed to write this?

FormerLurker, 2020-10-26:
> I hope I'm allowed to write this?

@LeandroMarceddu, you are welcome to contribute anything you would like!  Thank you for that!

I've posted this around several places, but lost track of who needed it.  This thread was missed.  For anyone who would rather not compile this themselves, check out the GithubActions [here](https://github.com/FormerLurker/ArcWelderLib/actions?query=workflow%3A%22CMake+Build+Matrix%22), and select the top action (the most recent build):

[image]

Then download the appropriate zip file (the ones I crossed out are known to have issues.... working on those):

[image]

FYI, github has no hosted ARM runner, so I can't yet pre-build an executable for the raspberry pi.  I'm looking into a self hosted runner (that's the ARM config that was confusing you) now.  The cross platform stuff has been a real education for me.  I learned one really must own the hardware to ensure everything works, lol!  Anyone got an old mac?

Trevor-Miller, 2020-10-28:
To integrate the .exe as a Cura post-processing script we'd need to write a Python script that executes the .exe, correct? I'm looking into this, but there's not a ton of information available on developing Cura scripts -- looking at https://github.com/Ultimaker/Cura/wiki/CuraAPI-and-SDK-Versions I'm not yet finding the information I'd need to dynamically pull the gcode file so I could pass it to ArcWelderConsole.exe as a parameter, but I'll update as I find any more info

FormerLurker, 2020-10-28:
Arc welder is wrapped in a python extension, so it can be called directly.  Just need to pre build the library.

Also, is there no way to call an exe directly from cura?  I thought there was a postprocessing plugin that did just that.

Anyway, I will develop a plug-in soon.  Just have to finish up a few things.  Also, i have successfully called arc welder from slic3r and prusaslicer if that is useful at all.

Trevor-Miller, 2020-10-28:
For those curious, some of the better resources I've found are:
https://community.ultimaker.com/topic/15655-how-to-make-a-plugin-anyone-have-a-plugin-to-do-x-action-every-layer-like-a-nozzle-wipe/

and digging through this plugin: https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/SearchAndReplace.py

However this appears to be a dead-end because obviously we can't pass the entire GCode into the ArcWelderConsole as a parameter, though that's how post-processing scripts receive it. 

I haven't found an official plugin that allows you to run .exes (certainly not in the [marketplace](https://marketplace.ultimaker.com/app/cura/plugins?page=1&search=))

The 'official' PostProcessing plugin that was incorporated into Cura only handles .py files: https://github.com/nallath/PostProcessingPlugin/blob/master/PostProcessingPlugin.py

FormerLurker, 2020-10-28:
Wow, it is getting the entire gcode file in memory?  I didn't imagine it would work like that.  I could write a wrapper to deal with that, I suppose, but it will be a pain.

FormerLurker, 2020-10-28:
I think the best way to handle things would be to use writelines to write the layers to a file, process them with the arc welder python extension, then convert it back into a list.  It will be tricky to return the layers properly, and all that will slow things down unfortunately.

Trevor-Miller, 2020-10-28:
At that point it starts to seem it's just a more convoluted version of the octoprint plugin, and would be more straightforward to run the .exe outside of Cura on the generated GCode

FormerLurker, 2020-10-28:
The octoprint cura plugin writes a file.  I need to dig in a bit.  As a pligit at least it would be seem less.

JohnEdwa, 2020-11-04:
The official post-processing plugin also runs every time after the file is sliced, which if the user has auto slicing enabled is 0.5 seconds after they chance a slicing setting. From the few tests I've done with the Octoprint plugin, ArcWelder doesn't seem to be particularly fast so it would increase the slicing time by quite a bit, and, Cura doesn't actually support displaying arcs in the preview so it would mess that up too.
So it should only kick in after the user presses "Save to File" or "Sent to Octoprint".

FormerLurker, 2020-11-04:
@JohnEdwa, you were running it on a pi.  It's extremely fast on my PC because the whole python wrapper is eliminated.  It is over 10x faster.  It's not multi-threaded ATM, so it could be sped up further potentially, especially if I know where the layers start and end.  But, I get what you are saying since every second counts.  I may end up just calling an executable on the pi to do the conversion to bypass all the slowness of python completely, but that will take a while.  It processes 100mb in around 30sec right now, and there are some higher priority items to deal with first.

Also, you are correct that cura doesn't display arcs.  This will hopefully be remedied in time.  The only program I know of that always seems to work correctly is Simplify.  OctoPrint itself is definitely getting better too, likely because ppl are using G2/G3 more now.

Regarding sending only after 'Send To Octoprint', I have no idea how I could detect this without some modifications to the Cura Octoprint plugin or the cura plugin framework itself.  It's easier to use the new (unreleased, but soon-to-be released) print after conversion feature in the arc welder plugin.  It does have some drawbacks though.  Specifically, I'm trying to figure out a way to detect that the file was sent with the cura plugin so it doesn't auto-start after every upload, but so far no dice.

It would be so much easier if cura allowed one to run an arbitrary file after slicing was completed as you can in Slic3r and PrusaSlicer:

[image]

The simplicity of that is very nice.  You can pass parameters too, but I hear there were some issues in earlier versions doing that.  One could always call a script file that calls the exe and passes the appropriate parameters though.

JohnEdwa, 2020-11-05:
Well, true on the Pi performance part. It's just that the parts I tested it on were quite small (such as [this Rift S mount](https://www.thingiverse.com/thing:3678066) that sliced is 9.3MB and it takes 20 seconds to Weld and an additional 15 until it pops up as it being ready on a Pi3). 

As for the file handling, I guess adding a "Weld and Send to Octoprint" option to the save menu that would then pass the welded file to the Octoprint plugin would be the correct way do it. Also, the Octoprint plugin has settings (Preferences -> Printers -> Connect to OctoPrint) that tell it if you want it to start or select the print after uploading. AFAIK those just control the [OctoPrint file operations API parameters.](https://docs.octoprint.org/en/master/api/files.html#upload-file-or-create-folder)

FormerLurker, 2021-01-23:
The cura plugin written by @FieldOfView is now available in the marketplace, so this can be closed.  Enjoy!

closed 2021-01-23 by FormerLurker

xeddog, 2021-01-23:
YAAAAAAAAY.  What took you so long???    

(Seriously, ThankyouThankyouThankyou)

FormerLurker, 2021-01-24:
LOL!  I didn't create the plugin, so thank @FieldOfView

xeddog, 2021-01-24:
@fieldOfView<https://github.com/fieldOfView> certainly deserves a thank you also.  But it would not have happened without your work.

Wayne
[https://avatars2.githubusercontent.com/u/143551?s=400&v=4]<https://github.com/fieldOfView>
fieldOfView (Aldo Hoeben) · GitHub<https://github.com/fieldOfView>
fieldOfView has 88 repositories available. Follow their code on GitHub.
github.com