FormerLurker/ArcWelderPlugin #26: Question about statistics after "arcing"

xeddog, 2020-06-27:
In the attached screenshot are the stats for a T-Tex tooth I sliced, uploaded, and welded.    I get all of the stats except Points Compressed.  Is that the number of small straight moves that have been made into the given number of arcs?  In this case, 1550266 small straight move segments have been converted into 440127 arcs?  

Wayne

[image]

xeddog, 2020-06-28:
And a second question - do you have any plans to support G2/G3 commands with the "P" operand for full circles???

closed 2020-06-28 by xeddog

FormerLurker, 2020-06-29:
Yes, 1550266 points were turned into 440127 arcs.  Also, full circles are not supported because that could lead to overlap in general models.  Arcwelder will always generate semi-circles, though they may be very close to a full circle.