FormerLurker/ArcWelderPlugin #28: Klipper Supported?

DroneMang, 2020-07-12:
Hi,

I am using Klipper, and just wondering if this plugin will work.

So far I have found to enable the below in Klippers printer.cfg, but what should the resolution be?  It defaults to 1.

[gcode_arcs]
#resolution: 1.0

Also, will firmware retract and pressure advance still work as intended?  How about tool changes (done in the slicer or via macros in Klipper)?

I have had some stuttering on circles so this should be awesome!

DroneMang

FormerLurker, 2020-07-13:
It should work as long as arcs are enabled.  Try a resolution of 0.25mm for starters.  Increase if arcs cause slowdown, but you probably have more than enough processing power to handle 0.25mm segments.

All of the other options should still work (lin advance etc), because behind the scenes klipper is generating small segments (the resolution setting) to generate the arcs.

changedsoul, 2020-07-24:
Im sorry to bring this back up, but I am using klipper "without" octoprint.
How would I get this to convert some files.

FormerLurker, 2020-07-24:
@changedsoul, what OS?  I have windows/linux/linux arm binaries of a command line version.

FormerLurker, 2020-07-24:
Check [here](https://github.com/FormerLurker/ArcWelderLib/actions/runs/107579357)

I have an arm binary for raspberry pi too somewhere, but I would have to manually build it since git actions dont support that yet.  Also, IOS and Ubuntu may have issues (still ironing out the kinks there)

changedsoul, 2020-07-24:
> @changedsoul, what OS? I have windows/linux/linux arm binaries of a command line version.

Its for a raspberry pi....what ever OS is installed on that....I think debian?

FormerLurker, 2020-07-24:
Rasbian.  I will have yo manually compile that.  Give me a day or two and I can send it.  If you happen to have access to a windows machine you could use that in the meanwhile.

changedsoul, 2020-07-24:
I downloaded the Linux one and ran it on my arch distro and it seem to be working nicely. I am going to compare a before and after print, but my first impressions are that just using arcs seems to have made my printer much quieter, lol. 
Very promising.

FormerLurker, 2020-07-24:
Awesome!  Star the repos if you think they are useful.  Enjoy the app, and post any feedback or thoughts please, good or bad.

DroneMang, 2020-07-27:
I am happy to report that this has been working great in Klipper.  I have even made the resulution smaller in Klipper, half as small as recommended as a start point, no issues.  I am printing with an SKR PRO.  

[gcode_arcs]
resolution: 0.05

FormerLurker, 2020-07-27:
Wow!  At that resolution you should have next to no deviation at all.  Do you notice any change at all in print quality or speed?

DroneMang, 2020-07-27:
Well I didn't really notice anything different, but the log for how the mcu
and connection is performing did seem like things were getting more taxed.
The good thing with Klipper is it will throw an error or shutdown if it
can't keep up, and it kept right on going.

FormerLurker, 2020-10-17:
Since things appear to be going well with klipper, I'm going to go ahead and close this.  Thanks for participating!

closed 2020-10-17 by FormerLurker