FormerLurker/ArcWelderPlugin #31: extruder skipping at all arcs and klipper gives move exceeds maximum extrusion error

tobyfu, 2020-08-06:
Hi I'm running klipper on a bigtreetech skr 1.4 turbo and thought I would give this plugin a try while I was printing some ear savers but it seems wherever there is an arc the extruder just skips and eventually klipper errors and stops the print. The original gcode prints without issue and as far as I can see from the modified gcode the same amount of material is being extruded and from the movement of the machine the toolpath isn't noticeably different so I assume the arcs are working as they should. I thought the skipping might have to do with pressure advance but disabling it didn't help. 

Klipper and Octoprint and ArcWelder are all up to date.

I'm just using the defaults for resolution and maximum arc radius but I also tried 0.1 for resolution and that still resulted in skipping. 

The print just has gaps and blobs where the curves should be.

Here is the before and after code:

[beforeAndAfterGcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5034129/beforeAndAfterGcode.zip)

FormerLurker, 2020-08-06:
The gcode looks correct.  I ran it through two analyzers.  Are you sure you have arcs enabled in the klipper config?  To test this you could send some test G2/G3 commands via the Octoprint Terminal.

tobyfu, 2020-08-06:
I was able to do some big non-extruding circles in both directions via the terminal so the arc movements themselves seem to be working okay. At this point seems like it probably isn't the plugin being an issue but I'm at a loss as to what it could be... Maybe klipper is doing something funny with E movements on arcs but I haven't seen anyone else with the same problem so far.

tobyfu, 2020-08-07:
It seems like what is actually happening is that the arc moves do not extrude at all for some reason and the error happens when it gets to the next non-arc move and tries to extrude it all at once because of absolute positioning. 

Anyway it is either a bug in klipper or my printer itself and not the plugin so I will close this issue.

closed 2020-08-07 by tobyfu

tobyfu, 2020-08-07:
Just in case anyone runs in to this I found my issue in klipper and fixed it, there was an issue with how klipper dealt with extrusion on arcs in absolute extrusion mode. Possibly you can use relative extrusion to avoid this entirely but I have a pull request in so hopefully it will be fixed in master very soon. The arcwelder gcode printed perfectly once this was addressed.

FormerLurker, 2020-08-07:
Hmm. I think i saw a PR to fix this in klipper.  I will take a look later.  Fyi, arcwelder will work on a file with relative extrusion just fine if you slice it that way.  Also, thank you for posting about your resolution!