FormerLurker/ArcWelderPlugin #32: Support for object cancelling

girrrrrrr2, 2020-08-07:
As far as I can tell the conversion script removes the lines that declare what part is which. 

Would it be possible to leave those lines in so that its possible to use the cancel object part of Marlin/Octoprint?

FormerLurker, 2020-08-15:
Hmm..  Can you submit a gcode file that contains the object cancel codes?  That way I can see what they look like before and after conversion.  Thanks!

girrrrrrr2, 2020-08-15:
here is a before and after of a project im working on.

[Coupler_Cage.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5079630/Coupler_Cage.zip)

FormerLurker, 2020-08-16:
Excellent, this will help, thx!

girrrrrrr2, 2020-08-16:
No problem! and if you want any more i can get you more!

andreibobirica, 2020-09-08:
IT deletes M486 too?

FormerLurker, 2020-09-09:
@girrrrrrr2 and @andreibobirica,

FYI, my investigation into this issue is still ongoing.  Sorry this has gotten pushed back, but I had some family stuff to deal with the last couple weeks.  I WILL get to this ASAP.

girrrrrrr2, 2020-09-10:
> @girrrrrrr2 and @andreibobirica,
> 
> FYI, my investigation into this issue is still ongoing.  Sorry this has gotten pushed back, but I had some family stuff to deal with the last couple weeks.  I WILL get to this ASAP.

Nah you are fine take your time. I can still use it when I need it. The part cancelation thing isn't needed that often. But when it is. It's nice to have.

FormerLurker, 2020-10-17:
I think I fixed it.  Looks like the cancel object's ```@Objectinfo``` tag was case sensitive.  I fixed the case sensitivity issue (had TONS of problems with this, lol!), and that seems to have fixed lots of other random issues.

I have NOT actually tried a print with cancel object, but looking at the plugin and the before/after gcode, I think it's working.  You can try the fix by installing v0.1.0rc1.dev5 from master as soon as I drop the release, which should be soon.  Please let me know if it's working!

girrrrrrr2, 2020-10-17:
Ill try this weekend!

girrrrrrr2, 2020-10-23:
So I forgot to check last weekend (Mostly because I was doing a bunch single object prints.) But it works perfect for me!

Thank you for fixing it!

FormerLurker, 2020-10-23:
Awesome!!!!  I figured it would work, but confirmation is extremely valuable.  Thanks for that!  Closed!

closed 2020-10-23 by FormerLurker