FormerLurker/ArcWelderPlugin #33: Corrupt gcode after conversion

NickRno77, 2020-08-15:
Prusa Mk3s. OctoPi 3b

I printer a part yesterday with 0.6 Nozzle Layer Height 0.3 I got corruption after AW convert, this part printed ok using a 0.4 Nozzle and Layer Height of 0.15.

When I viewed the gcode in ideamaker I can clearly see the corruption lines in the view, when I view the same file not converted there is no corruption?

I've attached the GCODE and photos of screengrabs and part photos
[image]
[image]
[image]
[image]
[image]

[ArcWelder Problem Gcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5078725/ArcWelder.Problem.Gcode.zip)

FormerLurker, 2020-08-15:
I will run this through my analyzer.  Thanks for the report.

NickRno77, 2020-08-15:
> I will run this through my analyzer. Thanks for the report.

Great thanks, I should of said ArcWelder has been great up till now although 0.6 nozzle does stutter a little, does that make sense ?

FormerLurker, 2020-08-15:
Of course!  Bugs happen, an I am glad to find/fix them.

FormerLurker, 2020-08-15:
OK, so I don't see the corruption when viewing in Simplify 3d.  That doesn't mean there is some issue, but in order to find it we will have to isolate which layer is having the issue.  Any way you could measure up on your print to identify the layer height from your first image where the exterior perimeter didn't print correctly?  If we can do that I'd recommend extracting that layer and only printing it to see if the problem happens again in the same spot.  Knowing exactly where the failure is would save a couple hours of manually reviewing hundreds of lines of gcode :)

[image]

NickRno77, 2020-08-15:
> OK, so I don't see the corruption when viewing in Simplify 3d. That doesn't mean there is some issue, but in order to find it we will have to isolate which layer is having the issue. Any way you [...]
 I hope this helps, its slightly above the female part of the latch, maybe 2 or 3 layers. I thought the latch projecting across the print might be something to do with the problem?
[image]
[image]

FormerLurker, 2020-10-17:
Hey, since this doesn't appear to be an arcwelder issue, I'm going to go ahead and close this.  Let me know if you have any other issues!

closed 2020-10-17 by FormerLurker