FormerLurker/ArcWelderPlugin #34: [improvement] Keep file thumbnails after welding

pietchaki, 2020-08-16:
I'm using the "Ultimaker Format Package" plugin (https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage) to get a cool thumbnail of the file in Octoprint. I believe there is something like this for PrusaSlicer too.
It would be great if you could copy these thumbnails into the processed file.
If I understood correctly, the thumbnail is not part of the GCODE itself, they are both part of a container, so they shouldn't impact on the print.

Thanks

FormerLurker, 2020-08-16:
Interesting idea. I will take a look at that plugin and see how it might be copied.

FormerLurker, 2020-10-17:
Done for both Cura Thumbnails and PrusaSlicer thumbnails!  Check out the 0.1.0rc1.dev5 release, which I will be dropping shortly once I clean up these issues.

closed 2020-10-17 by FormerLurker