FormerLurker/ArcWelderPlugin #35: Install Fails - unknown error

DaveZA, 2020-10-10:
Trying to install version 0.1.0rc1.dev4

No other version previously installed

System :
OctoPrint 1.4.2
Python 3.7.3
OctoPi 0.17.0

Installing thru OctoPrint Plugin manager
 
Takes a while, and then reports 'unknown error'

Have tried a few times over the last 2 days. How do I identify the cause of the error ?

Using url :
https://github.com/FormerLurker/ArcWelderPlugin/archive/0.1.0rc1.dev4.zip

Any help greatly appreciated.

[plugin_pluginmanager_console.log](https://github.com/FormerLurker/ArcWelderPlugin/files/5358572/plugin_pluginmanager_console.log)
[octoprint(2).log](https://github.com/FormerLurker/ArcWelderPlugin/files/5358573/octoprint.2.log)

FormerLurker, 2020-10-10:
Can you reboot and check the plug-in list?  Sometimes it "fails", bit really it works.

DaveZA, 2020-10-10:
tried that and check. not installed.

tried running from the command line :
~/oprint/bin/pip install https://github.com/FormerLurker/ArcWelderPlugin/archive/0.1.0rc1.dev4.zip

and it go to Building Wheel and sat there unresponsive for over an hour - I assumed that should have been enough time. 
Had to reset the Pi to get access again.

Now trying command line : 
~/oprint/bin/pip install https://github.com/FormerLurker/ArcWelderPlugin/archive/0.1.0rc1.dev4.zip --no-cache-dir

as I read that the no-cache-dir will install without the wheel ?

nope - it still stops at :   Building wheel for Arc-Welder (setup.py) ... \

DaveZA, 2020-10-10:
here's a copy of the command line output  :  

[install cmd line.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5359192/install.cmd.line.txt)

DaveZA, 2020-10-10:
Update - got it installed.

My printer is using a Pi B+, so moved the microSD to a Pi 3B+, did the install with the plugin manager in OctoPrint, and then moved the SD back to the printer.

Now to test the results. .....

FormerLurker, 2020-10-10:
Wow!  Did you ever figure out what was going on?  It likely has something to do with the C++ extension that actually does the conversion. I still need to improve this for several OSs, though it has genearlly worked well for OctoPi and windows installations.  MacOs and other Linux flavors need some work.

FormerLurker, 2020-10-10:
OH, also, I may push another pre-release today.  Will let you know if this happens.  I have added a bunch of statistics for the conversion that are pretty useful imho.

DaveZA, 2020-10-10:
I think it was that my printer is using a really really old Pi ( Pi B+ 512mb ), so that could have affected it.

The install when the SD was in the 3B+ was less than 2 minutes to do.

BUT ... I have got the SD back in the printer, and uploaded a file, converted.
Tried with the default resolution 0.05 and also with 0.025 - results seem the same.

I have a test cylinder that I have been using that gives a load of stuttering when run from OctoPrint, but perfect curves when run from the printer SD card.  The search for a solution left to ArcWelder.

But converting this model causes a problem result.
[image]

If you're able to take a look, here's my before & after gCode files, and a screenshot of the settings
[gcode1.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5359643/gcode1.zip)

FormerLurker, 2020-10-10:
Interesting.  Let me finish up what I'm working on, and I'll run your source file through the debugger.

FormerLurker, 2020-10-10:
It looks like the problem is related to extensive use of G92 offsets.  I thought I had that working :)

I'll see what can be done.

FormerLurker, 2020-10-10:
How is this:

[image]

[See if this file works for you](https://gist.github.com/FormerLurker/7aa47ee4aad46e7f8e3b877c47fe9e57)!

I'm finishing up some changes (adding detailed conversion statistics), then I'll push out a new dev release for you to try.

DaveZA, 2020-10-10:
Awesome. Thanks. Almost 10pm here, so won't get to try anything until later tomorrow.
Thanks so much for taking a look at this. Really appreciate it.

DaveZA, 2020-10-10:
last question for today : in the notes, you say :+1: 

Important Notes: If you are using a pre-release version of Arc Welder, I recommend completely uninstalling and cleaning the files before trying to upgrade.

Do you have any instructions for the non-technical users on how to do this ?

FormerLurker, 2020-10-10:
It probably isn't necessary anymore, buy you can do it inside the octoprint plugin manager.  Click uninstall and select the clean files option.  But, I wouldn't bother.

FormerLurker, 2020-10-11:
I think I have this fixed.  See if you can install the new version from this location:

```https://github.com/FormerLurker/ArcWelderPlugin/archive/devel.zip```

Let me know if/when you have it installed.  I will try not to make any breaking changes to the devel branch in the meantime.

DaveZA, 2020-10-12:
Thank You
 
sorry for delayed reply. time difference - only 6:30am here now.
 
just started a new print from the sd card, so will be a few hours until that
is done and I can test.
 
I have downloded the zip so will update it from my downloaded file, so feel
free to go ahead and brek the devel.zip if you want :)
 
Regards
Dave

 
  _____  

(#35)

https://github.com/FormerLurker/ArcWelderPlugin/archive/devel.zip

<https://github.com/FormerLurker/ArcWelderPlugin/issues/35#issuecomment-7067
IU73ANCNFSM4SK37E4A> .
<https://github.com/notifications/beacon/ADAWYVRGASAORXPNHCDAFLDSKIU73A5CNFS
M4SK37E4KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFIQIWPI
.gif>

DaveZA, 2020-10-12:
feedback :
ran a print from the printer SD card this morning.

Then updated to latest version of ArcWelder ( had to put the sd in my Pi 3B+ as it still would not update on the Pi B+, then moved the microSD back to the Pi B+ on the printer )

Printed the exact same part using ArcWelder from the Pi, and there are very very very little difference between the 2 models. Certainly not enough difference to stop me from using ArcWelder and the Pi to print, instead of the printer SD card.

What I did notice, was that the print from the Pi ran a lot faster than it used to ( same speed settings in Cura ), and I can only assume that the previous max speed was caused by the limitations of the serial comms speed.

Will be testing more models tomorrow. Thanks for the Dev Update.

FormerLurker, 2020-10-12:
@DaveZA , I have a B+ lying around.  I will test on that as soon as I have a chance.  I'm not surprised there are some issues, as the install is quite complicated.

> What I did notice, was that the print from the Pi ran a lot faster than it used to ( same speed settings in Cura ), and I can only assume that the previous max speed was caused by the limitations of the serial comms speed.

That's EXACTLY why I created ArcWelder!  So glad to hear it :)  Looking forward to the update!  Be sure to check out the new 'Stats' button on the ArcWelder tab.  It should show up when you select a previously welded file, though the new statistics will only show up with newly welded files since I just started adding the metadata.  I still have a few things to improve, namely extracting the metadata from the client instead of the server side (I wasn't aware all the metadata was pushed down to the client before).  I believe this info should be quite illuminating!

DaveZA, 2020-10-12:
Looking at the time that the Pi takes to process this, and then having to re-download the gcode file back to my PC for viewing and verifying that it is going to print what is intended, has got me thinking .. Is there a way to convert / process the gcode file on my Win PC before uploading to OctoPrint ?

DaveZA, 2020-10-12:
Running win 8.1 - 64 bit and just installed Python 3.9.0

Is there a way to process the gCode file on the PC ?

DaveZA, 2020-10-12:
OK. I thought that I could also use Octoprint installed on my NAS Pi, to process the files, and then pull them back to the PC, and upload to the printers Pi

Yes, I will order a new Pi 3B+for the printer soon :)

Just installed OctoPrint on my Pi3B+ which runs OpenMediaVault.

Installed in a Docker Container, and the UI appears correct, I can upload files, etc.

I also have the ArcWelder Plug-in active.

I am wanting to find the location, on the Pi, where the Uploaded ( and converted ArcWelder ) files are stored.

The OctoPrint settings show the Folders for Uploads as
/root/.octoprint/uploads

If I 'test' the path in OP Settings, it displays that the path is valid.

but I can not find this location when I ssh to the Pi using Putty

c-born, 2020-10-15:
Thanks for the info given in the comments here. I had the same problem with my RPi (Model B Rev 1), the system was thrashing due to lack of memory. To get around it I stopped the OctoPrint server and did the command line manual install as shown above, ie:
```
wget https://github.com/FormerLurker/ArcWelderPlugin/archive/devel.zip
sudo /etc/init.d/octoprint stop
~/oprint/bin/pip install ~/devel.zip
sudo /etc/init.d/octoprint start
```
which did the build/install successfully and gave me a working ArcWelder plugin.
(I now have ab RPi 3B+ on order, but it is at least a week away!)

FormerLurker, 2020-10-17:
> Looking at the time that the Pi takes to process this, and then having to re-download the gcode file back to my PC for viewing and verifying that it is going to print what is intended, has got me [...]

Yes, there should be.  Take a look at [ArcWelderLib](https://github.com/FormerLurker/ArcWelderLib).  There is a console project, and I have some pre-built binaries [here](https://github.com/FormerLurker/ArcWelderLib/actions/runs/311628599).  It looks like the Mac OS build failed, but it isn't working anyway :)

FormerLurker, 2020-10-17:
@DaveZA, you could also check the gcode viewer tab.  It might need some cleanup (report any issues on the Octoprint Github page), but if it looks good in there it should definitely work.

Also, just FYI, I agree you should be checking these files.  However, if it makes you feel better, your cylinder gcode was the first I've seen in a long while that didn't process correctly :)  If you do find any other issues, be sure to report them and I'll figure out what's going on ASAP!

c-born, 2020-10-17:
I built the console version from GitHub head on my PC, no problems. Processing the same gcode file as on my OctoPi took 74.95 seconds with the debug build, but only 1.29 seconds with a release build, producing identical output, a remarkable difference.
The RPi output differed only in a space before the comment semicolon.
I'll probably add it as a post-processor script to my Slic3r, it makes a huge difference in getting rid of stutter and the resulting blobs in my builds!

FormerLurker, 2020-10-17:
Yeah, it's WAYYY faster running as a console app.  It took quite a while to make it run that fast.  So glad you are getting some use out of this!

Also, I pushed a big update to the ArcWelderLib repo, so you might want to try it again.  I have CI (though the mac build isn't working), so you can snatch one the windows binaries [here](https://github.com/FormerLurker/ArcWelderLib/actions/runs/311628599).  

Let me know if they work, I haven't tested them directly since the last push, but they should be OK.

c-born, 2020-10-17:
Thanks, just did a sync and rebuilt release from master.
Processing time down from 1.29s to 1.21s, file size up from 6,498KB to 6,500KB, mainly due to the added extra space before comment semicolons.
It also dropped a few of the comments from the code, eg:
`M201 X2000 Y500 ; Limit acceleration (Y was slipping at default 3000)`
became
`M201 X2000 Y500`

No changes to the actual code itself, though?

Slic3r/PrusaSlicer post-processing script instructions say they require the script to replace the file in-place, which makes sense. Easy enough to add a script to do this, moving the original to a backup. But perhaps if the console version had a flag to do this itself, it would remove one extra step for users wishing to use it?

DaveZA, 2020-10-17:
Hi FormerLurker
Will try the latest release on the Pi and send you feedback.
For the console version on a ( win ) PC, can you point me ( not very technical user ) in the direction on how to install & use this ?
Thank You for all the effort you have put in to this plugin

FormerLurker, 2020-10-17:
@DaveZA, download one of the windows builds from [this link](https://github.com/FormerLurker/ArcWelderLib/actions/runs/311628599).  You probably want the MSVC .zip file.  Copy the ArcWelderConsole.exe file out of the bin\ folder in the zip file.  Open up a command prompt (type cmd in the search box) and navigate to the folder where you copied the exe to.  Then type:

```ArcWelderConsole.exe --help``

to see usage instructions.  Here is an example:

```ArcWelderConsole.exe "C:\3DPrinterFiles\SourceGcodeFileName.gcode" "C:\3DPrinterFiles\TargetGcodeFileName.gcode"```

You might also need to set the 'G90/G91 Influences extruder' option (false by default) like so:

```ArcWelderConsole.exe -g "C:\3DPrinterFiles\SourceGcodeFileName.gcode" "C:\3DPrinterFiles\TargetGcodeFileName.gcode"```

If you need more help, I'm planning to create some tutorials and videos soon.

FormerLurker, 2020-10-17:
@c-born, I will think about adding a flag to overwrite the source file.  I JUST added this to the plugin and pushed to devel, lol!  However, the console app will require different modifications.  I'll have to write to a temp file, then replace the original, which isn't all that difficult.

Also, can you post the gcode you used in your last test?  I want to figure out why it's dropping the comment on this line:

```M201 X2000 Y500 ; Limit acceleration (Y was slipping at default 3000)```

It should NOT be doing that!

DaveZA, 2020-10-17:
> @DaveZA, download one of the windows builds from [this link](https://github.com/FormerLurker/ArcWelderLib/actions/runs/311628599). You probably want the MSVC .zip file. Copy the ArcWelderConsole.exe [...]

tells me it's missing VCRunTime140_1.dll
I'm using win 8.1
I guess that's something that would be installed with the development platform ?

FormerLurker, 2020-10-17:
Interesting.  Yeah, you need the visual studio c++ runtime installed for it to work.  You can find the latest installer [here](https://aka.ms/vs/16/release/vc_redist.x64.exe) for the x64 version (recommended) and [here](https://aka.ms/vs/16/release/vc_redist.x86.exe) for the x86 version.  I need to add that to the installation instructions.  Maybe at some point I can create an installer, or create a statically linked version.

FormerLurker, 2020-10-17:
@c-born, I added source file overwrite.  Just set the source and target path to the same location, and it should detect it, create a temp file, copy the target to the source, then delete the temp file.  Hopefully this works on the other platforms.  Multi-platform c++ seems to be pretty difficult.

Please let me know if you have any issues OR if it works :)

DaveZA, 2020-10-17:
> Interesting. Yeah, you need the visual studio c++ runtime installed for it to work. You can find the latest installer [here](https://aka.ms/vs/16/release/vc_redist.x64.exe) for the x64 version [...]

Thanks for the installer link. will try that next.

Created a .bat file so I could drag & drop files to it for conversion ( i am lazy when it comes to repetitive tasks ).

Content of the .bat file : 
```
@echo off
echo %1
ArcWelderConsole.exe %1 %1_AW.gcode
pause

```
Worked 100%. Processed a file in a fraction of the time compared to the plugin in octoprint.

Used Notepad++ with a Compare Plugin and the 2 files are identical in content.

There was a small difference in the file sizes. Viewed again in Notepad++with 'Show All Characters' enabled, and there are differences in the CR LF line endings.

IMAGE : left was procesed in Octo on a Pi, right was thru the batch file on Windows PC.
[image]

Will the Windows Build on your link be updated whenever you upgrade the plugin ?

c-born, 2020-10-17:
@FormerLurker I went back and checked previous builds, and found  that dropped comment was fixed in https://github.com/FormerLurker/ArcWelderLib/commit/4fd38897fd66c245991a4066c7bb3db373087e70 

I tried the source file overwrite, which works as advertised, thanks!
One suggestion, if you use the C++ `rename()` function you should get multi-platform support, and also save on copying the file in the case where `tmpdir` is not on the same device as the source/dest (as in my case), along the lines of:
```
process(source.gcode, source.aw.gcode)
rename(source.gcode,source_bk.gcode)
rename(source.aw.gcode,source.gcode)
```

FormerLurker, 2020-10-18:
@c-born, are you a c++ dev?  In any case, I am happy for any advice! This and a very similar library for Octolapse have been my first c++ projects in decades, and are the first multiplatform c++ work I have ever done.  It has been pretty painful to say the least.  Additionally, this has to work in C++98 to maintain python 2.7 compatibility in windows (though oddly some 11 stuff works), so I will have to verify that it works in that case.  I cannot wait until I can drop python 2.x support entirely!

FormerLurker, 2020-10-18:
@DaveZA, when you say a small reduction, how much exactly?  I will add statistics output to the console tomorrow if you want to wait.  Just trying to see if your file converted correctly.  Perhaps it has no/few curves?  If you can, perhaps you wout be willing to link to the source and target gcode?

Also, i will be posting some instructions for integrating this with cura and prusaslicer/Slic3r.  At some point I may create a plug-in for the former.

DaveZA, 2020-10-18:
> @DaveZA, when you say a small reduction, how much exactly? I will add statistics output to the console tomorrow if you want to wait. Just trying to see if your file converted correctly. Perhaps it [...]

the file convert resulted in identical gCode on both the Pi and the PC files. Only reason for the difference in the file names is the line endings.

c-born, 2020-10-18:
@FormerLurker You have done a very impressive job, congratulations!
Most of my work has been on embedded systems, in C since the mid 80s, assembler before that, some C# in the last decade. But a bit of C++.
As ```rename()``` is part of the standard library, you should be fine with cross-platform compatibility.

FormerLurker, 2020-10-18:
@daveza, can you post your gcode please?  I would like to run it through the debugger.

DaveZA, 2020-10-18:
> 
> 
> @DaveZA, can you post your gcode please? I would like to run it through the debugger.

[gcode sample.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5397145/gcode.sample.zip)

c-born, 2020-10-20:
@FormerLurker Not the right place for this, however...
I have a part that fits over a 32mm diameter pipe, it is an interference fit with a 32.5mm ID.
I printed it using ArcWelder and it looked great, all the outside blobs due to stutter gone. But when it came to fitting it there was no way I could get it to slide onto the pipe.
I figured it could be due to Marlin taking the arc and converting to inscribed line segments, hence tightening the circle up a bit, did some searching, and found you were all over this topic! I did a bit of work on Marlin years back, fixed up some buggy servo-driver code, but haven't looked at it for a while. You are really getting into it! From what I read it looked like there was a problem reducing the segment size below 1mm (at least on older hardware like my RAMPS 1.4), do you know if this has been resolved now?
(My quick-fix was to increased the ID on my model by 0.5mm, which makes a snug fit, but that is only a temporary workaround.)

FormerLurker, 2020-10-25:
@DaveZA, thanks for the gcode!  It looks like everything is in order.

@c-born, it is possible if there is a really tight fit that the inscribed circles could cause issues.  Arc Welder can also move the inner and outer diameter by up to the resolution indicated in the plugin.  However, this should never happen for actual circular parts, but only for ovals or other non-circular organic shapes.

The inscribed segments issue has been discussed before, along with some potential ways of dealing with it.  They all seem to require more math, which can slow older boards down.  Using smaller segments will reduce, but not eliminate the issue.  I'll spend some time and hopefully will be able to come up with a solution, though I can't be sure this is what's causing your issue.  The deviation is quite small even with 1mm segments.

Another possible problem is that the segment interpolation is done with an approximation of sin and cos.  Marlin 2.0.6 added another term to the approximation, which greatly increases the accuracy.  However, it is still just an approximation.  I will run some tests on this as well.

closed 2020-12-18 by DaveZA