FormerLurker/ArcWelderPlugin #36: [Feature] Improve user experience when sending file to print directly through slicer plugin

dmarx, 2020-10-12:
I use Cura and generally send files to my printer using the Cura octoprint plugin, which adds a "print with octoprint" button in cura. When I hit this button, the file gets sent to my octopi where arcwelder automatically converts it to a compressed `fname.aw.gcode` file, but the original `fname.gcode` ends up being the one sent to the printer. I imagine this isn't specific to cura, and other slicers with octoprint plugins probably experience similar behavior. It would be nice if somehow arc welder could interrupt the behavior of the cura plugin to automatically print the `fname.aw.gcode` file. 

My current workflow (when I remember to do it) is to hit the "print with octoprint" button in cura, open octoprint in my browser, cancel the print, find the `fname.aw.gcode` file, and print that one instead. Would be nice if this was all just wrapped up in that first button push.

FormerLurker, 2020-10-17:
So, I've been looking at this, and I don't think it can be done without some modification to the Cura plugin.  It seems to select the file and start the print, and *Arc Welder* might not know anything about the file yet.  However, I think the workaround would be to edit your printer profile, and deselect the 'Start print Job after upoading' option:

[image]

dmarx, 2020-10-23:
Maybe the solution is to go even further upstream and add a cura plugin to the roadmap? I bet just adding a script to the existing postprocessing plugin would be lower effort.

FormerLurker, 2020-10-24:
@dmarx, there is a console version of arc welder that should do the trick.  However, does the cura octoprint plugin wait for postprocessing to complete before sending the file.  I will give it a try today.

FormerLurker, 2020-10-24:
[Here is a link to the pre built binaries](https://github.com/FormerLurker/ArcWelderLib/actions/runs/314225565) in case you would like to give it a go.  Add a -h parameter to the ArcWelderConsole exe to see the manual.

dmarx, 2020-10-24:
The cura octoprint plugin definitely waits for post-processing to complete. Thanks for all the ideas and the cool project!

bavery79, 2020-10-27:
I can confirm deselecting the "start print job after uploading" works just fine to prevent auto-printing of the file before arc welder processes it.  The only remaining step would be for arc welder to load and print the new file after its own processing, but I see that's already been proposed in another issue thread here!

FormerLurker, 2020-10-27:
I'm working on this as we speak.  It's been much trickier than I'd even imagined, but I'm getting closer.  I'm implementing a 'Print After Processing Complete' setting that allows you to print automatically after automatic or manual processing.  Here are the issues I'm trying to solve:

1.  In the normal case when you process via OctoPrint and nothing is printing, print after processing is completed.
2.  If a print starts during processing, cancel whatever is currently processing so that it does not negatively affect print quality (this is from another issue).  As a bonus I may add the file back to the queue so that it will restart when the print is complete, but I haven't gotten that far yet.
3.  If a print starts before preprocessing is started, ignore the 'Print After Processing Complete' setting for ALL queued items.  We don't want to start a new print right after another print has finished and there is a part on the build plate!
4.  If Arc-Welder starts a print, and there are more gcode files in the processing queue, print the first one and do NOT print the rest of them (see item 3).  In this case, after the print finishes, arc welder will continue to process the remaining files in the queue, but will not print them when they complete.
5.  If cura's 'print after upload' is selected, Arc welder might be able to beat the cura plugin and start printing the welded file, but it might not.  If it doesn't beat cura, preprocessing may stop or it may not stop depending on how close it is to finishing, but in any case Arc Welder will NOT start a new print.

This is a really big can of worms :)

FormerLurker, 2020-10-28:
@dmarx @bavery79,

I've implemented a feature that potentially resolves this feature request without requiring any modifications to the slicer or slicer plugins.  It still may need some tweaking, but I think it's a good start.  

If you want to test it, you can install from this url within the plugin manager (get more->From Url...): ```https://github.com/FormerLurker/ArcWelderPlugin/archive/1.0.1.rc1.dev1.zip```.  Please read the release notes [here](https://github.com/FormerLurker/ArcWelderPlugin/releases/tag/1.0.1.rc1.dev1).  You can comment in [this thread](https://github.com/FormerLurker/ArcWelderPlugin/issues/55) and if there are problems I may ask you to create a new issue.  Thanks!

closed 2020-10-28 by FormerLurker