FormerLurker/ArcWelderPlugin #38: [Suggested Doc Update] G90/G91 Feature Selection is on Features menu item, not in OP printer profile (OP v1.4.2)

TodWulff, 2020-10-23:
Title says it all - :)

Following the guidance here: https://github.com/FormerLurker/ArcWelderPlugin#printer-settings

I bounced around the printer settings to find this setting and was rather cornfuzed.  Ended up finding it on the Features settings page under Features(!), as depicted here:  https://i.imgur.com/6qWoXN9.png

Take care.

-t

FormerLurker, 2020-10-25:
@TodWulff, thanks for pointing that out!  I have modified the name of the base setting and have updated the help file and the readme.md file.  Please let me know what you think.

LMS0815, 2020-10-26:
Please find my G90/91 testing gcode below:

```
G90; Switch to absolute coordinates
G0 X10 Y10; Move to origin at 10, 10
G91; Switch to relative coordinates
G0 X0 Y0; no move if G91 is working - if not move to 0, 0
G90; Switch back to absolute coordinates
```

If G91 is working, `G0 X0 Y0` will not move the head.
If not, the head is moved from X10 Y10 to position X0 Y0.

FormerLurker, 2020-10-26:
@LMS0815, I'm confused.  Why are you testing G91?  Is there a firmware that doesn't support this command?  Arc Welder needs to know if G90/G91 changes the axis mode of the extruder (usually set with M82, M83), but I've not seen a case where either G90 or G91 doesn't work.  Am I missing something?  If so, sorry.

LMS0815, 2020-10-26:
I am probably the one who is confused. 
Why do we need the G90/G91 awareness. What is it doing?

Seems I misinterpreted the help text.

FormerLurker, 2020-10-26:
@LMS0815, for some firmware (Marlin 2, Smoothieware, and maybe and others) sending a G91 is equivalent to sending both a G91 and M83.  For slicers, this isn't much of an issues because they can always just insert either an  M82 or M83 after sending a G90 or G91 if they want to use absolute/relative extrusion.  However, if you are reading gcode to try to figure out what the printer is doing, you can't tell if the e axis is in relative or absolute mode after seeing this gcode:
```
M83; Set Extruder To Relative
G90; Set XYZ to Absolute
; What mode is the extruder in now?
```

The only way to know this currently is to assume either G90 does or does not affect the e axis mode.  If the gcode is ambiguous (not explicitly using M82/M83 after G90/G91) you will sometimes guess wrong.  This can have some pretty big impacts on the conversion algorithm, since it needs to know what exactly is going on with the extruder (is it retracting or extruding mainly, but other stuff too).

Hopefully at some point you will be able to query the printer to figure out exactly have G90/G91 behaves.

LMS0815, 2020-10-26:
Thank you for the detailed explanation. I think I am now on the same page. 
The flag actually tells you if G90/G91 overrides M82/M83 mode.
As this is  printer pending, I would have expected this flag being in the printer profile (and not under functions). 
_We should raise this question at OctoPrint forum if not already done._

Never the less one could evaluate manually by sending some gcode (not tested yet):
```
M83; Set Extruder To Relative
G1 E-10 F500; slow Retract 10mm
G90; Set XYZ to Absolute
G1 E0; if no extraction G90 overrides M83 -> ENABLE FLAG - otherwise Extrude 10mm -> DISABE FLAG
```

FormerLurker, 2020-10-26:
@LMS0815, Nice script!  I have created something similar in the past, but have never figured out a way to tell if it's working without looking at the extruder gears.  Any ideas about how to detect it without watching the printer?

LMS0815, 2020-10-26:
The only option I can think of is `M503; Print a concise report of all current settings` but I am not sure if the output is easy to interpret throughout the various printer models.

```
M83; Set Extruder To Relative
M503; Report
... get values of interest
G90; Set XYZ to Absolute
... get and compare values of interest
```

FormerLurker, 2020-10-26:
I just checked my printer, and it doesn't have any axis mode info.  I'm wondering if I can use M114.  It appears to work on my machine:

```
Send: G1 E-10 F1000
Recv: echo: cold extrusion prevented
Send: M114
Recv: X:0.00 Y:0.00 Z:0.15 E:-10.00 Count X: 0.00 Y:0.00 Z:0.15 E:0.00
```

The E axis went from E:0 to E:-10, which is a bit expected since the firmware rejected the motion (probably a bug).  I also realized we need to add a ```G92 E0``` to the beginning of the script.  Next I tried the rest of your code:

```
Send: G90
Send: G1 E0
Recv: ok
Send: M114
Recv: X:0.00 Y:0.00 Z:0.15 E:-10.00 Count X: 0.00 Y:0.00 Z:0.15 E:0.00
```

No Change to E there, which is good.  Now we can test absolute mode to make sure E changes as expected:

```
Send: M82
Send: G1 E0
Send: M114
Recv: X:0.00 Y:0.00 Z:0.15 E:0.00 Count X: 0.00 Y:0.00 Z:0.15 E:0.00
```

Looks good to me.  This will work if 114 works as expected.

LMS0815, 2020-10-26:
Yes  M114 seems to report the position!!!

FormerLurker, 2020-10-26:
@LMS0815, regarding M400, this will need to be done behind the scenes (Octolapse uses this trick extensively) so that it knows the all gcode has been executed before any further commands are sent.  So, I have two new buttons I need to add: Auto-Configure firmware

Hopefully I will be able to figure out if arc commands are enabled, if they are configured properly, what the ```G90/G91 influences extruder``` setting should be, and if the firmware version has any known issues (say, Marlin 2.0.5 vs 2.0.6).

LMS0815, 2020-10-27:
This is to confirm that my printer is having the same cold extrusion firmware _bug_.

```
G1 E-10 F1000
echo: cold extrusion prevented
ok P7 B3
M114
X:162.50 Y:188.00 Z:4.98 E:-10.00 Count X:13000 Y:15040 Z:2000
ok P7 B3

```

I am looking forward to the test button.

LMS0815, 2020-10-27:
> @LMS0815, regarding M400, this will need to be done behind the scenes (Octolapse uses this trick extensively) so that it knows the all gcode has been executed before any further commands are sent. So, I have a new button I need to add: Auto-Configure firmware
> 
> Hopefully I will be able to figure out if arc commands are enabled, if they are configured properly, what the `G90/G91 influences extruder` setting should be, and if the firmware version has any known issues (say, Marlin 2.0.5 vs 2.0.6).

Is it worth to evaluate the Marlin version being  2.0.6 or higher `M503` ?

FormerLurker, 2020-10-27:
> My printer extrudes to E0 -> G90 has no impact to M83 -> DISABE FLAG

I think the test shows the opposite, that g90 DOES influence the extruder in your firmware, else the ```G1 E0``` wouldn't have had any effect, no?

On my printer G90/G91 does NOT influence the extruder, as shown by this test:
```
Send: M83
Send: G1 E-10
Send: M114
Recv: X:0.00 Y:0.00 Z:0.15 E:-10.00 Count X: 0.00 Y:0.00 Z:0.15 E:0.00
Send: G90
Send: G1 E0
Send: M114
Recv: X:0.00 Y:0.00 Z:0.15 E:-10.00 Count X: 0.00 Y:0.00 Z:0.15 E:0.00
```
Note that I have removed OK and Cold Extrusion Prevented messages.

Also, and I just thought of this, It is likely NOT a bug that the extruder position changes even when cold extrusion is prevented.  I think it's to maintain the absolute position so that subsequent gcodes don't fail if the extruder is still heating.  It may eventually reach an acceptable temperature.

> Is it worth to evaluate the Marlin version being 2.0.6 or higher M503 M115 ?

Yes, definitely.  I would flag any instance of Marlin < 2.0.6 as having 'potential problems' with G2/G3.

> Should I not use ArcWelderPlugin (although I do not have any issues - yet)?

You aren't guaranteed to have issues with an older version.  It depends a lot on other factors too.  If you aren't having problems, it's probably fine to use Arc Welder.  I still recommend that you upgrade when possible.

I will be working on this button over the weekend.  I'll probably create a new issue for this and will report my progress there.  I'll update this ticket with a link when I do so.

FormerLurker, 2020-10-28:
So, I'm going to close this out since the initial fix is now implemented and released.  I will add an issue for the firmware configuration button later.  

If you want to test it, you can install from this url within the plugin manager (get more->From Url...): ```https://github.com/FormerLurker/ArcWelderPlugin/archive/1.0.1.rc1.dev1.zip```.  Please read the release notes [here](https://github.com/FormerLurker/ArcWelderPlugin/releases/tag/1.0.1.rc1.dev1).  You can comment in [this thread](https://github.com/FormerLurker/ArcWelderPlugin/issues/55) and if there are problems I may ask you to create a new issue.  Thanks!

closed 2020-10-28 by FormerLurker