FormerLurker/ArcWelderPlugin #39: [Feature Request] When printing from slicer direct to OctoPrint (via slicer plugin) offer option to print AW's output

TodWulff, 2020-10-23:
When printing from the slicer direct to OctoPrint, via a slicer plugin, the AW plugin does its thing, but the print is the original gcode sent by the slicer.  Can AW hook a callback to offer the user to print either the original .gcode file, or the .aw.gcode that is generated by Arc Welder?

girrrrrrr2, 2020-10-24:
Or what about something like where if it detects a line of code in the welding, like "Print_After_Weld" in the beginning of the file (could be done automatically via start Gcode) and its a new file, it welds and automatically tries to start the print

Idk just throwing my idea out here

FormerLurker, 2020-10-25:
@TodWulff, I've been discussing this with several people, and a hook may indeed be the best way to proceed.  I've also considered adding a 'Print after welding' option to Arc Welder, though that also presents some problems.  I'll be working on this.

@girrrrrrr2, it is very tricky to intercept a gcode and stop the print before anything is actually sent to the printer.  Octolapse had to deal with this, and the solution is..  very ugly.  At some point I may create a feature request for OctoPrint (or maybe a PR) that adds a 'before-print-start' synchronous callback that happens before any gcodes are queued.

girrrrrrr2, 2020-10-26:
I was thinking more of a it being uploaded but not being auto started. And then when you see the line in the code, it starts the print right away, or if its printing its ignored... something like that

FormerLurker, 2020-10-28:
Hey @TodWulff and @girrrrrrr2, I've added a feature that may work for this that doesn't require modifications to slicer plugins.  If you want to test it, you can install from this url within the plugin manager (get more->From Url...): ```https://github.com/FormerLurker/ArcWelderPlugin/archive/1.0.1.rc1.dev1.zip```.  Please read the release notes [here](https://github.com/FormerLurker/ArcWelderPlugin/releases/tag/1.0.1.rc1.dev1).  You can comment in [this thread](https://github.com/FormerLurker/ArcWelderPlugin/issues/55) and if there are problems I may ask you to create a new issue.  Thanks!

closed 2020-10-28 by FormerLurker