FormerLurker/ArcWelderPlugin #41: Errors near 'outside' ring of delta printer

EliW, 2020-10-25:
I have a Rostock Max V3 — I was excited by this plugin (and still am) because I'm often frustrated at how I have a circular delta printer, and yet it does all these straight movements inefficiently.

However on my first attempt, I was printing multiple circular(ish) objects at once.  The inner one looked to be printing very nicely.  But the outer object was having issues and I stopped after the second layer.

Basically, the infill and the 'walls' were not lining up.   they were offset to each other, causing a rather unsightly mess.

FormerLurker, 2020-10-25:
I need the before and after gcode please (source and arc welded).  Also, a snapshot of what the printer is doing may be helpful

EliW, 2020-10-26:
Before and after copies shared (.aw is after version).   As for snapshot what do you want?  Are you wanting a visual video recording of the 'miss', or are you wanting a printing log?   Either way, it'll have to wait until I finish the project and I can re-run this for a couple of layers.  But let me know and I'll get it to you! 👍 

[Dominion_Body_(18hr).aw.gcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5438816/Dominion_Body_.18hr.aw.gcode.zip)

[Dominion Body (18hr).gcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5438805/Dominion.Body.18hr.gcode.zip)

FormerLurker, 2020-10-26:
If possible i would like to see a photo of the build plate after you cancel the print.  I want to compare that with the gcode for those layers.  Thanka!

FormerLurker, 2020-10-26:
FYI, I looked at both the source and arc-welded gcode, and they look the same to me.  I'll wait for your photo, but it looks like this could be a firmware issue.

EliW, 2020-11-03:
Hey, so I still never got a chance to reprint that big boy just to wait for it to fail.   But I was printing something small (and circular!  some caps to fill in drill holes) and went 'Hey, good option to try again'.

So I did.  And it failed spectacularly.  So basically on the arcwelded version, the walls completely didn't print at all.  Only the infill & supports   And the support was weird.  Where it normally would be a square 'snake' shape, it instead became a backgammon board of triangles.   The solid bottom of the infill shows it really well also.

Photo showing the print, and then when I printed the unmodified form:
[image]

Video showing failed print off better:
https://d.pr/v/97PxP0

The files:
[Archive.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5481880/Archive.zip)

EliW, 2020-11-03:
FWIW, one print done right after the other with no changes to any settings.

EliW, 2020-11-03:
(and obviously this isn't near the outside ring.  But right in the middle.  So the title of this ticket is slightly off now.   Seems likely that this is a firmware issue as you suggest.  Unless something 'weeeeird' is going on.

FormerLurker, 2020-11-03:
I think the chances are almost 100% that arcs are disabled.  Send an ```M115``` command to your printer and paste in the response here.

Next, run this test (it's also at the bottom of the Arc Welder page in the plugin repository):

```
G90
G28 X Y
G0 X0 Y0
G2 X40 I20 F400
```

Enter those into the octoprint terminal one at a time.  When you enter the final command (```G2 X40 I20 F400```) your printer should move in an arc slowly to Y0 X40.  If it does not move, or it returns 'Unknown Command' or something like that, arcs are disabled.  If it works, something else may be going on.

FormerLurker, 2020-11-03:
It's also possible that arc movements just aren't supported for delta printers.  For example, it looks like G2/G3 wasn't working in marlin for a while on Deltas.:  https://github.com/MarlinFirmware/Marlin/issues/2371.

EliW, 2020-11-06:
Thanks for the details.  Just did them and low-and-behold.  Yeah, G2 support doesn't exist (command not found).   But I swore I saw it in the firmware.

So I went and started looking more deeply and found the following:

1) Yes, there IS arc support, but it's not compiled in by default.  I could recompile the firmware to add it.  (Though there's some 'tight memory' issues on my machine anyway and it would be adding a significant amount of code.)  But more importantly ....

2) It appears that the arc support, is 'faked'.  Basically inside of Repitier firmware, if arc support is enabled, it accepts arc commands, and turns them into many many small linear movement commands.   So basically we'd be in a bit of a 'copy of a copy' situation.  Where I'd be taking the output from Simpify3D that is full of small linear movements.  Having ArcWelder convert those into arc commands.   And then having my firmware re-convert them into small linear movements.

Not really ideal.   Thanks for the help in figuring out what was up however.

closed 2020-11-06 by EliW

FormerLurker, 2020-11-06:
I figured it was at least disabled.  However, I wanted to comment about something you said:

> It appears that the arc support, is 'faked'. Basically inside of Repitier firmware, if arc support is enabled, it accepts arc commands, and turns them into many many small linear movement commands. [...]

It's not really faked.  Steppers can only move in discrete increments anyway, so drawing 'true' curves is impossible.  Technically drawing non-horizontal/vertical straight lines is also impossible due to the discrete nature of the steppers!  They operate sort of like pixels on a screen.

There are several main benefits for using arc commands that don't rely on them being interpolated by the firmware:

1.  Fewer instructions sent which frees up some very expensive operations (serial communication, buffer space, parsing, just to name a few).
2.  The arc resolution can be controlled by the firmware (I'm working on creating a gcode to configure arcs ATM), which can increase or decrease resolution based on the speed of your board.
3.  As arc support is improved, we may be able to produce finer and finer details when using arcs, so this may 'future proof' the gcode.

There are others, but I wanted to make it clear that the fact that arcs are interpolated does NOT make them any less useful than linear segments.  Quite the opposite, in fact, depending on your firmware and the configuration.  Obviously if the firmware isn't handling arcs well this is negated, but if they didn't handle straight segments well, they wouldn't be useful either!

Does that make sense?

EliW, 2020-11-12:
Totally makes sense and thanks for the info :)