FormerLurker/ArcWelderPlugin #42: Improvement GUI

LMS0815, 2020-10-26:
If the browser has a high horizontal resolution the main GUI configuration tab is out of order.

[image]

FormerLurker, 2020-10-26:
Looks like a css issue.  Would you be willing to disable your other UI plugins and see if this page looks better?  I'd like to narrow it down to a particular plugin so I can see what is causing this, and potentially fix it.  Here is what it should look like:

[image]

I do want to show the current run configuration, but I may add a show/hide button, and will store the state in the browser storage so it remembers.

LMS0815, 2020-10-26:
I am sure the _issue_ it is related/cause to Themify which i am using to update the css (as you mentioned).
I like this setting as it enables OctoPrint to use the whole 16:9 screen and not only a fraction of it.

[image]

FormerLurker, 2020-10-26:
Looks like your .span8 selector is too broad, and is affecting all span8s on the page.  Try this instead:

Selector: ```.tabbable.span8```
CSS-Rule: ```width```
Value: ```calc(100% - 300px)```

The result should be something similar to this:

[image]

LMS0815, 2020-10-26:
Unfortunately myscreenshot is cutting the content. The setting is `calc(95% - 360px)`

I disabled it and added your recommendation. 
100% setting is not working, But 95% did the trick.

Selector: `.tabbable.span8`
CSS-Rule: `width`
Value: `calc(95% - 300px)`

Selector: `.container`
CSS-Rule: `width`
Value: `95%`

@FormerLurker **THANK YOU**

Please consider offering an option to remove this information from this tab and only provide the statistics.

closed 2020-10-26 by LMS0815

FormerLurker, 2020-10-26:
I will add a show/hide button.  It is a reasonable request :)