FormerLurker/ArcWelderPlugin #43: [Feature Request] Option to automatically print the file after processing

xkuyax, 2020-10-26:
Hi,

thank you for your awesome plugin!
At the moment im using the handy feature of Superslicer to start the print after uploading, because of that i dont have to open octoprint to start my print. Would it be possible to include that feature in the arc welder plugin?

Thanks!

FormerLurker, 2020-10-26:
This has been discussed a bit over the last couple days.  There is an issue that specifically asks for a print to start after uploading from a slicer, but since the title of this asks for automatic printing after processing, and because this is something I am considering anyway, I'm going to mark this as an enhancement.  

FYI, There are some issues getting this to work that I need to solve.

martonbaksa, 2020-10-27:
I'll second this one. I upload directly from PS to octoprint, and having to open the web interface to start the print adds a couple of extra steps to the workflow, or I end up accidentally printing the wrong one because I leave the "start printing after upload" ticked. Thanks @FormerLurker for your amazing work in this plugin and Octolapse as well!

FormerLurker, 2020-10-28:
Hey @xkuyax  and @martonbaksa, this has been implemented.  If you want to test it, you can install from this url within the plugin manager (get more->From Url...): ```https://github.com/FormerLurker/ArcWelderPlugin/archive/1.0.1.rc1.dev1.zip```.  Please read the release notes [here](https://github.com/FormerLurker/ArcWelderPlugin/releases/tag/1.0.1.rc1.dev1).  You can comment in [this thread](https://github.com/FormerLurker/ArcWelderPlugin/issues/55) and if there are problems I may ask you to create a new issue.  Thanks!

closed 2020-10-28 by FormerLurker