FormerLurker/ArcWelderPlugin #44: Slowdowns in certain curved areas while performing Infill

war4peace, 2020-10-26:
I have started using ArcWelder for a model which has a round base.
The nozzle slows down a lot while performing an arc between infill lines.
Video of the issue:
https://www.youtube.com/watch?v=t7hgM0Zzhjk
GCode file: https://drive.google.com/file/d/1zNAy_NOukkugZuJofqBfKqFfHPxth7GM/view?usp=sharing
Printer: Creality CR 10 MAX
Firmware: Marlin 2.0.5 customized by TinyMachines3D for this printer model.
Printing using OctoPrint.

For any other details, please let me know and I'll provide them.

FormerLurker, 2020-10-26:
> Firmware: Marlin 2.0.5
This is most likely the issue.  I recommend asking the folks at tinymachines3d to pull in changes from Marlin 2.0.6, which has fixed many issues with G2/G3.  It shouldn't be too difficult.  

The creator of Marlin says [he doesn't recommend arcs pre 2.0.6](https://github.com/MarlinFirmware/Configurations/pull/107#issuecomment-643808597).  I've got lots of anecdotal evidence that supports this, but some people don't have issues on prior versions.  My guess is that it is situational.

war4peace, 2020-10-26:
Since I haven't bought the printer from them directly, I doubt there will be changes. I had submitted a firmware bug to them and they were kind to reply, but haven't released any new firmware just yet.
Until that happens, I will disable the plugin and revisit when it will work on my printer.
About the print I mentioned earlier, I canceled it because the results were disastrous - therefore please mark the CR 10 MAX with 2.0.5 TinyMachines3D firmware as NOT compatible.

FormerLurker, 2020-10-26:
Will do, and I really appreciate you trying this out.  Where did you report the bug, btw?  I found a github page, but the repositories are either private or do not exist.

war4peace, 2020-10-26:
Their github:
https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0
Their documentation page:
https://tiny-machines-3d.myshopify.com/pages/resourcepagecr10max
Their new user guide:
https://docs.google.com/document/u/0/d/1TiwFGTbX3R9_RI275Kq_-tfqDOhl-XXogsCqUaA_tB8/mobilebasic

FormerLurker, 2020-10-26:
So, the current CR10-Max release does NOT have the G2/G3 fixes, but their most recent branch does.  I would expect that to make it into the next official CR-10 firmware.  I will keep an eye on this.

rizwansarwar, 2020-11-07:
So I have noticed something similar not to this extent however. I have noticed that when doing infill, the printers stutters at the end of an infill line (where direction change is about to take place). When printing with gcode that is not converted by ArcWelder, i do not see this issue. 

And I have seen this in Klipper not on Marlin. maybe something to investigate further.

FormerLurker, 2020-11-08:
If you are interested in trying an alpha, I've implemented a new firmware check and a bunch of other goodies.  The check still needs work and there are still some things I need to fix, if you want to try it out you can install from the plugin manager by clicking 'Get More' then adding this line to the 'From Url...' box and clicking install:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/d92528cdebf158f67b906caf99c50a35c7fcbc4e.zip```

Once the install is finished, restart octoprint and make sure your printer is connected (this is one of the things that's broken).  Then go to the arc-welder tab and click on the new 'Check Firmware' button.  It is possible that a check has already completed.  

Here is what it looks like:

[image]

If you have any problems running the checker, go to the arc welder settings and turn on verbose logging for the firmware checker like so:

[image]

Clear the current logs:

[image]

Save your settings changes:

[image]

Click on the firmware check in the Arc Welder Tab:

[image]

Then download and post your log file at gist.github.com.  You can download it via the Octoprint logging menu, or go to the arc welder settings and click download log:

[image]

I've still got to add help files for various firmware versions with known issues, but for now you can just send me a snapshot of the checker and I'll let you know what your options are.

Holzniggl, 2020-12-03:
could it be possible that the problem is eeprom related? i´ve updated my printer to marlin 2.0.7.2 and had the same problems with a regular benchy and a wallthickness test. since then i had multiple problems with my z-offset when i was using M500. now i changed all the eeprom settings in my firmware and made a factory reset, it works just fine. i will print a benchy tomorow to prove this. 
hope itll help

FormerLurker, 2020-12-03:
@Holzniggl, as far as I can tell all of the arc settings are #defines, so nothing to store in EEPROM.  However, several other things affect this, like junction deviation settings and linear advance.  Anything that slows down planner calculation can yield slowdowns.  It would be really interesting to see what was in your EEPROM that changed.

Holzniggl, 2020-12-04:
@FormerLurker, I guess that the problem was that i updatet to the 2.0.7.2 firmware, but didn´t factory reset the machine afterwards, so i had the EEPROM settings from the 2.0.4.3 firmware with enabled arc support AND enabled linear advance loaded.
Since then i´ve altered the firmware some more till the EEPROM wasn´t compatible any more with the FW so i was promted to do a factory reset. 
 
[image]

o-nix, 2020-12-15:
I am on Marlin `2.0.7.2`, definitely see the blobs. Probably at the end of some (not all) infill lines.
Standard `ARC_SUPPORT` properties (`MM_PER_ARC_SEGMENT=1`, `MIN_ARC_SEGMENTS=24`, `N_ARC_CORRECTION=25`) with `EEPROM` disabled. `LIN_ADVANCE` is enabled (along with some default flag `EXPERIMENTAL_SCURVE`) but setting `K=0` does not change the picture.

FormerLurker, 2020-12-15:
@o-nix, can you try printing that gcode directly from SD to eliminate the possibility that this has something to do with OctoPrint?

FormerLurker, 2020-12-15:
also @o-nix, have you tried resetting your EEPROM as @Holzniggl did?  Not sure if you upgraded or not.  Can't hurt.  You can backup your old EEPROM setting with this plugin:  https://plugins.octoprint.org/plugins/eeprom_marlin/

o-nix, 2020-12-15:
@FormerLurker SD print is problematic, I believe my SD card support is not working properly, but I'll try to work around.
I have NO `EEPROM` enabled (`// #define EEPROM_SETTINGS`), I code changes into the firmware.

FormerLurker, 2020-12-15:
@o-nix, perhaps you could reboot OctoPrint into safe mode and try re-printing?  I always like to see what happens with no plugins enabled to rule out any kind of conflict, especially if SD prints are not an option.

FormerLurker, 2020-12-15:
OH, and could you post your source gcode?  If the problem persists, I want to make sure the extrusion lengths are correct.  This is a painstaking process, so I'd rather not do it unless it is absolutely necessary.

o-nix, 2020-12-16:
I'm still verifying, but it looks like migrating from `2.0.7.2` to `bugfix-2.0.x` worked! 🚀
Definitely the problem in Marlin itself.
Anyway, I was trying things on `2.0.7.2`:
* Figured out proper settings and enabled SD support, started printing from SD only - same picture
* Disabled curve support and linear advance - no effect
* Tried to experiment with different `ARC_ENABLED` settings - slight improvements but still 0.5s slowdowns on some specific points
* Downgraded jerk to classic - no changes
* Set min segments - no way

Attaching the files (contains .gcode, movement analysis and sample's photo with blobs): [marlin_arc_blobs.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5699476/marlin_arc_blobs.zip)

SageCrispin, 2020-12-19:
FYI:
Tried out Arc Welder for the last couple of days. Drastically shrunk some files, but slowed printing (hence my arrival here). Dl'd and installed the alpha. Reported:  

Version:
2.0_DW5
Errors
Your printer's firmware is not supported.

So, it worked for me. Now if Tiny Machines would just step up. 
Thanks anyway for what will be a great plugin I'm sure.

RURon, 2020-12-22:
> I'm still verifying, but it looks like migrating from `2.0.7.2` to `bugfix-2.0.x` worked! 🚀
> Definitely the problem in Marlin itself.

I can confirm this. I just flashed the latest Bugfix over 2.0.7.2 keeping the same settings in Marlin - from blobby curved arcwelded arcs to non-blobby nice and clean arcs!

FormerLurker, 2020-12-28:
@RURon and @o-nix, any idea what changed in Marlin?  I may need to update my firmware detection library to show issues in 2.0.7.2 if this is a universal problem with the release.

Also, I just dropped a new rc/devel version of ArcWelder.  Feel free to check it out.  The software update is broken for all previous versions if you are running OctoPrint 1.5+, so you'll have to install it manually.  Check out [this issue](https://github.com/FormerLurker/ArcWelderPlugin/issues/106) for details.