FormerLurker/ArcWelderPlugin #45: [Feature Request] Ignore certain directories

adoeller, 2020-10-26:
Hi,

i move printed files to a "done" dir using the filemanager. If automatic AW is switched on, the process starts automatically.
It would be useful to have a blacklist of dirs to exclude from the automatic conversion.

Cheers,
Alex

FormerLurker, 2020-10-26:
Yuck, I didn't know the 'file added' event was triggered for a move...  I will look into this, thanks for reporting.

FormerLurker, 2020-10-28:
Hey @adoeller, this fix has been implemented.  If you want to test it, you can install from this url within the plugin manager (get more->From Url...): ```https://github.com/FormerLurker/ArcWelderPlugin/archive/1.0.1.rc1.dev1.zip```.  Please read the release notes [here](https://github.com/FormerLurker/ArcWelderPlugin/releases/tag/1.0.1.rc1.dev1).  You can comment in [this thread](https://github.com/FormerLurker/ArcWelderPlugin/issues/55) and if there are problems I may ask you to create a new issue.  Thanks!

closed 2020-10-28 by FormerLurker