FormerLurker/ArcWelderPlugin #46: [Feature Request] Wait processing before auto-starting job

Daniel4182, 2020-10-27:
in Super slicer i usually, export direct to the octopi and set it to start printing, but arch welder will start processing the code and octopi start with the non processed version as i can tell, i think that maybe i got a print failure because of it, can not prove it tough, if it's not a problem, i would like it to see octoprint waiting for the processed code, before the print start

FormerLurker, 2020-10-27:
For the time being, please disable automatic processing in arcwelder.  Since arc welder will not start a job while printing, I suspect that ot started processing first, then the print started.  The only solutions I can think of are spawning a thread in C++ to do the processing (technically difficult) or cancelling preprocessing if a print starts during processing.  I will think about it

FormerLurker, 2020-10-28:
Hey @Daniel4182 , this fix has been implemented.  If you want to test it, you can install from this url within the plugin manager (get more->From Url...): ```https://github.com/FormerLurker/ArcWelderPlugin/archive/1.0.1.rc1.dev1.zip```.  Please read the release notes [here](https://github.com/FormerLurker/ArcWelderPlugin/releases/tag/1.0.1.rc1.dev1).  You can comment in [this thread](https://github.com/FormerLurker/ArcWelderPlugin/issues/55) and if there are problems I may ask you to create a new issue.  Thanks!

closed 2020-10-28 by FormerLurker

o-nix, 2020-12-06:
As I came to GitHub issues specifically for this feature request I should say GREAT JOB @FormerLurker!
I became a patron and everybody should! @Daniel4182