FormerLurker/ArcWelderPlugin #47: Percentage Goes Greater than 100

martonbaksa, 2020-10-27:
[image]

Hi!
I think I might've found a bug. Octoprint doesn't like the file size for some reason, and displays the file size incorrectly (6.2mb instead of 8.1mb actual). This causes the print time and the percentages to break. I've attached the G-Code. 
Marton

GCodes:
[panda_dual_black.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5447452/panda_dual_black.zip)

FormerLurker, 2020-10-27:
I'm honestly not sure where this issue is coming from.  I don't set the file size anywhere.  I'm wondering if it is added automatically somehow by some other OctoPrint code.  I will look into it.

FormerLurker, 2020-10-27:
So, I'm not seeing what you are seeing when I weld and print this file:

[image]

It looks like you have a lot of plugins running.  My best guess right now is that it's being caused by some other plugin.  Are you willing to disable them until you figure out which one is causing the problem?  I think that will be the only way to track this down.

martonbaksa, 2020-10-28:
I believe this might have just been a really weird one time error where it didn't get the file size right as deleting the file and manually rewelding it fixed it.

FormerLurker, 2020-11-21:
I think this is fixed now.  Will try to release an RC soon.  Resubmit if you see it again!

closed 2020-11-21 by FormerLurker