FormerLurker/ArcWelderPlugin #48: Marlin SkyNet3D-V2.3.2

jjvelar, 2020-10-27:
Hi!
Could you please add support for Marlin SkyNet3D-V2.3.2?
I know it is old but it is working quite well in my Anet A8 :-)
Thanks and best regards,

José

FormerLurker, 2020-10-27:
Do you mean I should add your firmware to the list of supported firmware, or that you'd like arc-welder to work with your firmware?

jjvelar, 2020-10-28:
I would like arc-welder to work with my firmware.
Thanks

José

FormerLurker, 2020-10-28:
So, I took a look at the firmware, and you are correct:  it is really old.

> I would like arc-welder to work with my firmware.

There isn't any modification I can make to Arc Welder that will make it work with your firmware.  I looked online, and there are several guides for upgrading the Anet A8 to Marlin 2.  However, the example configuration file on [Marlin's github page](https://github.com/MarlinFirmware/Marlin/tree/2.0.x/config) has arc support disabled.  I'm not sure if the A8 has enough memory to enable that feature without further configuration changes.  Perhaps someone with an A8 and experience with Marlin 2.0 can chime in?

jjvelar, 2020-10-28:
Not unexpected. What a pity!
Thanks for taking the time to review this.
Kind regards

José

jjvelar, 2020-11-01:
Hi!
I just wanted you to know I have finally updated my Anet A8 to Marlin 2.0 and I have been able to activate arc support.
I am re-configuring steps and the new fancy stuff (power loss, etc) but I will start soon to "play" with arc welder.
Thanks again and best regards,
José

FormerLurker, 2020-11-01:
Awesome!  Please let me know if it starts working!

FormerLurker, 2021-01-23:
Going to go ahead and close this out for inactivity.  If you have any future problems, please create another issue, and thank you for your post!

closed 2021-01-23 by FormerLurker