FormerLurker/ArcWelderPlugin #50: Disable Arc-Welder Buttons while printing

FormerLurker, 2020-10-27:
Disable Arc-Welder Buttons while printing, except the 'show stats' button.

codefaux, 2020-10-28:
Related; I noticed the button to 'weld arcs' on the "Files" panel of Octoprint indicates "Cannot weld arcs during print" etc etc, but the plugin seems to have no hesitation to automatically start processing while printing, if you upload a file while printing.

Rationale;
Manual Arc-Weld trigger not available during print; sane
Automatic Arc-Weld trigger not blockable during print, always enabled; not sane

Resolutions;
Queue-but-pause automatic processing when a file is uploaded during printing?
Trip an error dialog stating that automatic processing was skipped and will need to be manually triggered?

Also, fantastic plugin. Spiral Vase stuff looks -incredible- now, it's blowing my mind. I've ALWAYS been mad that slicers seem to ignore arcs. Thank you!

FormerLurker, 2020-10-28:
Yeah, i actually already fixed this.  It will now also cancel processing is you start a print, and will re-queue it to process after the print completes. I am testing this stuff now.

Also, thanks for your compliments!  I'm very glad you are finding it useful!

FormerLurker, 2020-10-28:
Hey @codefaux, this fix has been implemented.  If you want to test it, you can install from this url: ```https://github.com/FormerLurker/ArcWelderPlugin/archive/1.0.1.rc1.dev1.zip```.  Please read the release notes [here](https://github.com/FormerLurker/ArcWelderPlugin/releases/tag/1.0.1.rc1.dev1)

closed 2020-10-28 by FormerLurker