FormerLurker/ArcWelderPlugin #55: Feedback for V1.0.1.rc1.dev1

FormerLurker, 2020-10-28:
If you have any feedback for the V1.0.1.rc1.dev1 release, please leave that here.  Let me know what you like, what you love, what you hate, or any ideas about the future of Arc Welder. If you believe you have found a bug, feel free to mention it here. I may ask you to create a new issue.

Thank you in advance for your contribution!

FormerLurker, 2020-10-28:
Hey, if anyone notices that the version they installed is showing up at 1.0.1.dev1 instead of 1.0.1rc1.dev1, or if the new *Print After Processing* setting as displayed on the tab under *Current Run Configuration* is actually showing the *Select After Processing*, please reinstall from this url:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/1.0.1.rc1.dev1.zip```  I accidentally targeted the wrong branch for about 15 minutes, but it's fixed now.

Thanks, and sorry about that!

martonbaksa, 2020-10-28:
So far, I really like this rc. Thanks again for automatically starting the print, it works a charm :>!

NameOfTheDragon, 2020-10-29:
ArcWelder is AWESOME. Thank you for making this. I have always been disappointed that slicers don't use the arc commands to full advantage. Perhaps some of them will reconsider this in the light of the clear advantages demonstrated by ArcWelder, but meanwhile thanks for a great plugin. It makes a huge difference with Klipper firmware because GCode throughput across the USB-serial link has been a long-standing issue with Octoprint. I'm using an Ender 3 with a BigTreeTech SKR Mini E3 running Klipper and I was having a lot of stuttering issues that made for poor print quality. ArcWelder has eliminated almost all of these, so a massive "thumbs up". I will be sending you a few bucks to support the project.

The auto-print-after-processing enhancement is a step in the right direction but, in my opinion, its not quite the right solution. Auto-start printing really needs to be controlled from the slicer and then ArcWelder needs to be completely transparent to the user, avoiding the need to interact with OctoPrint at all in order to start a print. The current solution gives that, but takes away the users control at the slicer end. You don't _always_ want to start the print automatically and the the ability to control that in the slicer is, I think, quite important.

I feel that this must be possible, because the Display Layer Progress plugin does something similar. DLP pre-processes the gcode to add M117 placeholder messages throughout the file, which are then later further processed in-line as the file is streamed. So it must be possible to preprocess a file before it prints and replace it with the processed gcode. I would be fine if OctoPrint never saw the "raw" file and only the ArcWelder pre-processed file was ever visible in OctoPrint. I hope you are able to find a way to do this in a future version, it would be the cherry on top of the cake.

FormerLurker, 2020-10-29:
@NameOfTheDragon,

I believe *Display Layer Progress* uses the gcode analysis hook.  Somehow the Cura Octoprint plugin can wait for this to finish before starting a print.  I initially built the Arc Welder plugin using the analysis hook, but was having some major difficulties due to the way Arc Welder works.  I may give this another go at some point, but it would require a pretty massive rewrite.  I have thought about the scenario you mentioned (you want to upload, but not print), and it is definitely a weakness.

First, I plan to add a *Firmware Check/Configuration* button to deal with printers that either have no arc support, or that have poor arc support.  This may also be able to deal with the *G90/G91 Influences Extruder* setting.  That setting isn't as important for Arc Welder as it is for Octolapse, but it could cause some problems if it isn't set correctly.

Also, if you are willing to submit some before/after photos of prints using Klipper by itself, and Klipper with Arc Welder.  The before/after GCode and statistics (including before/after print times and a snapshot of the Arc Welder conversion statistics) would be AWESOME.  I'd love to post some of these.  I'm working on some of these using my MK2 with Prusa's fork of marlin.

FormerLurker, 2020-10-29:
@hamster65,  That's great!  If you happen to do any before/after prints, I'd love to see photos, print time comparisons, a snapshot of the Arc Welder statistics, and before/after gcode.

NameOfTheDragon, 2020-10-29:
@FormerLurker 
> Also, if you are willing to submit some before/after photos of prints using Klipper by itself, and Klipper with Arc Welder. The before/after GCode and statistics (including before/after print times [...]

I will see what I can do. I have a print in progress that is going to take me well into tomorrow so it will not be quick.

BTW, I am using Prusa Slicer (2.3.0-alpha.1), which also lets you control whether to start printing after upload and works with DLP, so that's not special to Cura.

FormerLurker, 2020-10-29:
@NameOfTheDragon, you could use the console version of ArcWelder and enter that as a post processor within PrusaSlicer.  You can find the console version [here](https://github.com/FormerLurker/ArcWelderLib/actions?query=workflow%3A%22CMake+Build+Matrix%22).  For now, select the most recent action and pick the Windows-MSVC.zip file if you are using windows (I broke the mingw version recently).  The mac versions don't work ATM, and the Linux version won't work on the pi (no git hosted ARM runners ATM, working on that).  I plan to update the application and push out a formal release including the artifacts soon.

To get it running in prusa slicer, you can use something like this:

[image]

Example Post Processing Script Setting: ```{path_to_arcwelder_exec}\arcwelderconsole.exe -r=0.05 --g90-influences-extruder;```

I'm not sure what the semicolon is for.  PrusaSlicer apparently added that by itself after I saved the profile.

You can use this to run arc welder with the default settings: ```{path_to_arcwelder_exec}\arcwelderconsole.exe```

You can get help by opening a command prompt, navigating to the directory containing the exec, and executing this: ```arcwelderconsole --help```

Note:  you may need to remove the --g90-influences-extruder flag depending on your firmware.  Leave it for Marlin 2.0 and.. i think..  [smoothieware](smoothieware.org) (double check that).

NameOfTheDragon, 2020-10-29:
> @NameOfTheDragon, you could use the console version of ArcWelder and enter that as a post processor within PrusaSlicer. 

Ooooh, now that's interesting, I will give that a go. BTW, I was just trying to make some sample prints using Klipper and in an attempt to maximize the difference I used a cylinder in spiral vase mode. No arcs are being produced when in spiral vase mode, hence #56 :(

FormerLurker, 2020-10-29:
@NameOfTheDragon, I updated that issue, and will summarize here for others.  I don't think it's possible move to another Z coordinate currently when creating an arc.  It would also require some changes to the shape detector so that it can operate in 3 space, though that wouldn't be particularly challenging.  It may slowdown the conversion a tiny bit.

However, I just looked at the marlin 2 source, and [there is definitely something more than just bed leveling going on with the Z axis](https://github.com/MarlinFirmware/Marlin/blob/ac7ee4b94a200590a566fd6979f4738976de12b6/Marlin/src/gcode/motion/G2_G3.cpp#L80).  It looks like it does take z travel into account.  My guess is that this is for [CNC Workspace Planes](https://marlinfw.org/docs/gcode/G017-G019.html), but I know nothing about those.  Marlin's [G2/G3 docs](https://marlinfw.org/docs/gcode/G002-G003.html) don't talk about allowing a Z parameter for arcs, though it should be easy to implement in theory.  If a Z parameter is ever added, I could add support for it within arc welder.

jcharaoui, 2020-11-02:
I've done a few tests with ArcWelder and I want to say: well done! I've been struggling with stutters ever since I got my Ender-5,  printing with OctoPrint on a Pi ZeroW. I shared a [report of my findings](https://www.thingiverse.com/groups/creality-ender-5/forums/general/topic:39719) on Thingiverse, essentially pointing toward increasing baud rate, buffers and toying with the slicer maximum resolution/deviation settings with the goal of reducing GCODE complexity. Nevertheless, whenever I let my guard down in terms of model/GCODE complexity, the stutters would come back to haunt me.

I chose just recently to overhaul my Ender-5 with a TH3D EzBoard Lite 1.2 and Pi 3B+. Admittedly, the upgraded Pi performed MUCH better than the ZeroW in terms of stuttering. I had no idea how much the ZeroW was underpowered in that sense! So finally, I went to work printing some pretty arc-gruesome models I was putting off. Although the print quality was excellent, I could still observe perhaps one or two very clear [movement stutters](https://imgur.com/a/NayQm5A) per layer. I went back to my old habit and increased the baud rate and firmware buffers, but this time, it didn't have any effect. Could I be hitting some bottleneck in the firmware and/or the serial interface itself? Well, this looks to be the case, because with ArcWelder, the print is 100% totally smooth with no stutters at all that I could observe, while preserving print accuracy.

I think currently ARC_SUPPORT is way underrated because before ArcWelder, there was no real way to compare both methods (pure linear moves vs arcs) head-to-head. Your plugin changes this completely and I expect that quite quickly, some major slicers will introduce arc support thanks to your work! The benefits of this plugin, by dramatically reducing serial connection traffic, isn't only limited to better print quality. It means more GCODE bandwidth between host and printer than can be leveraged to deliver improvements in many areas like better print progress display granularity, jerk/acceleration control and more!

ETE-Design, 2020-11-10:
@FormerLurker So to understand this correct, we only need to insert `{path_to_arcwelder_exec}\arcwelderconsole.exe -r=0.05 --g90-influences-extruder;` in Post Processing Script in Prusa Slicer / Super Slicer, then it will generate the right GCode with ARC Support... So we can copy that to an SD-Card and print is that correct?
Could you mabye make a Post Processing Plugin for Cura?

FormerLurker, 2020-11-10:
@ETE-Design, yes and yes :)  Working on the plugin.

However, I did change the name of the console app to ArcWelder or ArcWelder.exe depending on the OS.  You can get the latest artifacts in the ArcWelderLib repo.

ETE-Design, 2020-11-11:
@FormerLurker Have just tryed it out, but it seems like it does nothing to the gcode file... All commands are still G1, only thing is that the size is a little bigger now :-(

FormerLurker, 2020-11-11:
@ETE-Design, Hmm...  I will take a look.  It works in my debugger.  The additional size is probably due to comments added at the top.

ETE-Design, 2020-11-11:
@FormerLurker Your right it is caused by the comment in the top, I was just told when using ARC the file will be reduced in size.... BTW I use SuperSlicer

FormerLurker, 2020-11-11:
Yeah, I did a huge refactor, but it was for the OctoPrint plugin, so I probably broke something.  I think it has to do with overwriting the target file..  Give me 10 minutes.

FormerLurker, 2020-11-11:
Ok, pretty close to 10 minutes :)  It takes a while to build the artifacts.  Try [this link](https://github.com/FormerLurker/ArcWelderLib/suites/1482612945/artifacts/25666557) to the msvc artifact.  I tried it out on a benchy and it looks good.

I think I made a copy paste error, but didn't notice it since I was testing with two parameters (source and target).

FormerLurker, 2020-11-11:
Here is what Benchy looks like after processing with a 60.2% reduction in extrusion/retraction commands:

[image]

ETE-Design, 2020-11-11:
@FormerLurker That was it, now it's working :-) Thanks for being so fast to solve it...

ETE-Design, 2020-11-11:
@FormerLurker  But seems like there is something wrong this is the Original:
[image]

And this is after ARC:
[image]

FormerLurker, 2020-11-11:
Much more likely the slicer can't visualize G2/G3.  Send me your file and I'll run it through simplify.  To my knowledge it is the ONLY slicer that can properly visualize arc commands.

For an alternative, try ncviewer.com.  It has some issues that cause some random commands to looks like bad arcs/segments, and renders arcs with a small number of facets, but it is pretty darn good for a free online tool!

ETE-Design, 2020-11-11:
@FormerLurker As long as it is only the Preview that isen't correct no harm is done then :-) Any your right only minor error when using ncviewer.com

susisstrolch, 2020-11-11:
@FormerLurker can you also point where to find a Linux build?

ETE-Design, 2020-11-11:
@susisstrolch https://github.com/FormerLurker/ArcWelderLib/actions/runs/356847098

FormerLurker, 2020-11-11:
@susisstrolch, I have to warn you, the linux build there will NOT work for ARM (raspberry pi).  If you need a build I can generate one manually.  Github has no hosted arm runners.  I'm planning to set up a self hosted runner for that, but I've got a big backlog :)

Also, there are some mysterious issues on some linux flavors.  I've not pinned this down yet, but I'm a relative noob regarding cross platform c++ development.  Eventually I'll get to the bottom of it.  Let me know if you have problems.

@ETE-Design, thanks for posting that link!

susisstrolch, 2020-11-12:
@FormerLurker I‘ll will try it on x64-platform, Linux Mint. Long time goal is to use ist as Vura Plugin. 
Thanks for your ongoing effort!

ETE-Design, 2020-11-16:
@FormerLurker I use a plugin from BigTreeTech who make it possible to get a 3D Review on the TFT of your gcode... Do you know if it work with your console app?
Right now the Plugin is only for Cura, but a version for SuperSlicer is also soon out...

FormerLurker, 2020-11-17:
What is 'TFT'?  Put the script on gist.github.com if it will fit, and I'll take a look.

ETE-Design, 2020-11-17:
@FormerLurker MKS TFT / BigTreeTech TFT... Rename script to *.py

[biqu_convert_new.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5554439/biqu_convert_new.txt)

ETE-Design, 2020-11-17:
@FormerLurker It male it possit possible to male 3D Review like this...
[image]

FormerLurker, 2020-11-17:
Looks like it just needs a loader (.bat file) that calls the python code.  Remove this line from the python script:

```
#!/usr/local/bin/python3
```

Save the pthon script as ```tft.py```

Create a file called tft.bat that contains the following:

```
@echo off

python tft.py
```

Save it in the same directory as the tft.py script.  Now run it and it should work.  Note, you will need to make sure python is in your environment variables (it should be).  you can test this by just typing python into a command prompt and hitting enter.  It should say something like this:

```
Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:19:30) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
```
Also, you may need to enter the full path for the .py script in the .bat file depending on how you are calling it.  Something like this:

```
@echo off

python "c:\path_to_the_file\tft.py"
```

Should do it.

ETE-Design, 2020-11-17:
@FormerLurker After removing #!/usr/local/bin/python3 it got possible to run the script from command prompt, but only if I use the original name of the file... biqu_convert_new.py g-code file
It can't be run trough the tft.bat no matter what i call the *.py, then it just return "./biqu_convert inputfile" so right now I can't use it as post-processing script :-(

FormerLurker, 2020-11-17:
Ahhh, you need to forward the arguments to that file!  Unfortunately my fiber is down (new install, bad hardware, go figure).  Appt is tomorrow evening.  Will look at it again once the tubes are up again.

ETE-Design, 2020-11-17:
@FormerLurker That's what happend :-( But I'll just wait then :-)

ETE-Design, 2020-11-19:
@FormerLurker Did you get your fiber up and running again? ;-)

FormerLurker, 2020-11-19:
@ETE-Design, Yes I did.  Yesterday afternoon.  Been playing some pretty difficult catch up.  It's easy to take the internet for granted, but my gosh I'm dead in the water without it.

So, to fix your issue you need to pass in all of the arguments to the python call.  You can do this with ```%*```, which contains all the args sent to the batch file.  Here is a sample:

```
@echo off

python tft.py %*
```

Here is a test python file I made (test.py):
```
import sys

if __name__ == "__main__":
   print (sys.argv[1:])

```

And the sample bat file:
```
@echo off

python test.py %*
```

I called it with some random parameters, and here's what I get:
```
C:\Temp>test.bat Test TEst TEst1
['Test', 'TEst', 'TEst1']
```
Let me know if that fixes it.

ETE-Design, 2020-11-19:
@FormerLurker That did it :-) Thanks... Will delete my posts since it has nothing to do with feedback for your ARCWelder :-)

FormerLurker, 2020-11-19:
Cool, I will do the same with mine.  Can you do me a favor since I did one for you??  I am about to release 1.0.1rc1.dev2, and would like someone other than me to try it beforehand.  If you are interested, you can install from this url:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/be0c56ccf7699a0da88c678ac434f45a5cea0c74.zip```

I'm particularly interested in feedback on the 'Firmware Checker' and the new UI.  If it all works I hope to push a release this weekend.

I can't promise that you won't have to delete and reinstall when the next release comes out, but I'll do my best to prevent that :)  I'm going to be removing the 'Slicer Upload' options that you might see, so don't use those (they won't work properly).

ETE-Design, 2020-11-19:
@FormerLurker As you micht know I don't use OctoPi, that's the reason I asked for the Command Line version of your briliant work :-) But I can post it in the OctoPrint Group and ask for feedback there?

FormerLurker, 2020-11-19:
lol!  NP.  I will reach out.  Sorry, I interact with so many ppl I sometimes forget who I'm talking to.  I used to work retail way back in the day, and it was the same there.

NameOfTheDragon, 2020-11-19:
> Ooooh, now that's interesting, I will give that a go. BTW, I was just trying to make some sample prints using Klipper and in an attempt to maximize the difference I used a cylinder in spiral vase [...]

I did try this and it works rather well! Having done a lot of prints with it, I'm at the point now where I am confident in Arc Welder to the point where I feel OK trusting it to automatically process every file, so I will most likely use this method once the next release is stable. The command line version seems to operate much faster than the OctoPrint plugin.

FormerLurker, 2020-11-19:
@NameOfTheDragon, so glad!  

> The command line version seems to operate much faster than the OctoPrint plugin.

Yes it does, and though it will probably always be significantly slower running on a raspberry pi (It is only a pi afterall), I am working on this issue now in a separate branch.  I'm going to attempt to add a lot of parallel processing, which could potentially speed both apps by a lot (2-3x on a pi with 4 cores running octoprint and a lot more on a PC with numerous cores).  It's going to be a challenge, but should be fun :)

Also, your point earlier:

> The auto-print-after-processing enhancement is a step in the right direction but, in my opinion, its not quite the right solution. Auto-start printing really needs to be controlled from the slicer [...]

I have added a feature (needs a bit of testing still) that allows you to control arc-welder via gcode comments.  It's not the best solution, but adding a line to the start gcode for a particular profile can trigger printing on OctoPrint, and a bunch of other stuff.  Will include all that in the docs.

I'm also finishing up a hook that will allow slicer plugins to tell ArcWelder what to do.  I don't really think this will be used much except maybe for the Cura Octoprint plugin (been going back and forth with the author.  A good solution to this is difficult).

I think the ultimate solution is to bypass this problem by processing in the slicer then uploading to OctoPrint in the regular way.  For the PrusaSlicer and many others this is moot, since you can already do that (as you are) with the console app.  Cura is the odd one out.  It is different in every way, but does add some potential opportunities.  It delivers the gcode in a list of layers, with each layer containing a list of GCodes.  This will be SLOW to process, but can be parallelized pretty easily once that is done.

So, lot of movement.

ETE-Design, 2020-11-21:
@FormerLurker Do we have to add -Z To the Terminal App to use ARC on Z-Axis?

FormerLurker, 2020-11-22:
@ETE-Design, yes.  You can add the --help parameter to get a full list of args and the default values

bvrielink, 2020-12-02:
@FormerLurker Didn't see there was a rc and have no problem to do some testing if needed. But as this rc is already more then a month old I guess I am a bit late to the party. Are there any bugs stopping the V1.0.1 release or is the release around the corner?

FormerLurker, 2020-12-02:
@bvrielink, I'd hoped to get the next version out like two weeks ago, but it's been a rough couple weeks.  I already had a huge backlog and didn't get around to testing with OctoPrint 1.5 until rc3, where I discovered Octolapse was broken (18K+ confirmed users, so it's been a pain for the entire community), so I spent my time fixing those problems and arc welder (2k users about) unfortunately took a back seat.

If you'd like to get a jump on testing you could install from this commit url via the plugin manager:  https://github.com/FormerLurker/ArcWelderPlugin/archive/2a95fab3a7797b3dad3664bece27f08137d0f46c.zip

There is one known issue and two potential issues I'm going to deal with before rc1.dev2 drops:

1.  If your printer responds with 'Unknown Command' in the new firmware tester, it will disconnect.  I've fixed this already for 'G2/G3 Bad Parameter' response.  If your firmware definitely supports G2/G3 you will be fine, and the worst case is a printer disconnect, in which case you can disable the firmware test in the settings.
2.  The new 'Firmware Compensation' might be breaking up too many arcs in some cases, but it does seem to fix the 'flat edge' issue for firmware that has that problem.  If you need that I recommend setting min arcs to 14 (the current default is 12, but is too low in some situations)
3.  The new '3D' arcs setting seems to cause a seam for some prints or firmwares.  This has not happened to me, so it could be a firmware issue.  However, it could also be an issue with the 3d arc code.  I need to review that.

I would really like to UI feedback.  I rearranged stuff and killed a bunch of popups, like the progress popup (now in the tab)

FormerLurker, 2020-12-02:
OH, @bvrielink, one more thing I'd like to mention.  Performance should be improved quite a bit in the devel branch.  I found that log strings were being constructed when logging was disabled for the c++ code.  They weren't being logged, but creating the strings and sending them to python was nearly doubling processing time on a pi.  The console app and windows didn't seem to have much performance difference at all, maybe a few percent, but on a pi it was taking nearly 2x as long for me in the old version.

ETE-Design, 2020-12-13:
@FormerLurker What is the optimal options to set in Marlin under ARC to get perfect circle or close to? Cause have just activated ARC with standard settings, and without ARCWelder the Circle is more perfect than when ARC is used :-(

cp2004, 2020-12-18:
@FormerLurker just thought I'd let you know, run a fair number of prints with this version of AW enabled, and all seems to be going just fine. Did I see somewhere you fixed the software update hook problem?

Other than that I can't report any issues, prints coming out very smooth.

FormerLurker, 2020-12-18:
@cp2004, thanks fot the report!  Yes, the hook was fixed.  Still working on a few bugs before I deploy.

FormerLurker, 2020-12-18:
@ETE-Design, what version of marlin?  Some ppl have said factory resetting eeprom fixed their issues after an upgrade.  The default arc settings should be fine, depending on the version.  Can you post a pic?

ETE-Design, 2020-12-18:
@FormerLurker Use Marlin Bugfix 2.0.x - 17.12.2020, always resetting after updating :-) Is some settings more recommended more than others? And is there some settings who is best for perfectly round things?
My printer is down right Noe, so will test again when it is up and running again...

FormerLurker, 2020-12-28:
OK, finally have the next devel rc released.  Please leave feedback [here](https://github.com/FormerLurker/ArcWelderPlugin/issues/106), and note that the software update is BROKEN in all previously released versions of arc welder for OctoPrint 1.5 +!  You will have to install the new release candidate manually by going to *OctoPrint Settings->Plugin Manager->Get More->From URL..* and paste this link to install the most recent pre-release:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/1.0.1rc1.dev2.zip```

Thank you for all your contributions!  I think the next release is much better for it.

closed 2020-12-28 by FormerLurker

ETE-Design, 2021-01-23:
@FormerLurker Sorry to highjack you again... But seems like you know how to set up actions and other things here to compile for Windows... I understand if you don't want to use your time on my question... But here it comes, https://github.com/Zip-o-mat/Slic3r/ is only compiled for Linux cause the owner don't know how to comlile it on Windows... Can you help, mabye even if I can compile it on my locale machine it would be fine, cause running an emulator inside Windows makes the Slicer really slow... Thanks for your great work :-)

FormerLurker, 2021-01-24:
I will take a look.  I am assuming this is for automation (github actions), right?  When i have some time I will look at he make (hopefully cmake) script.

ETE-Design, 2021-01-24:
@FormerLurker I'm not good at Github Actions, so if it can be done direct from my computer that would also be fine... The creater seems like he' dosen't know it either... Can IT be done in VSCode mabye?

FormerLurker, 2021-01-24:
So, if a github action is added to the repository, the binaries can be built automatically when changes or releases are added to github.  This isn't all that difficult since there is already a cmake script.  I'm sure you can build it with vscode, but you will have to install the cmake extension.  Also, cmake might need to be modified so that a windows build will work.

ETE-Design, 2021-01-24:
@FormerLurker Already have VSCode + CMake Extention :-) Just don't know how to compile stuff like this :-( And thanks for looking at it :-D

FormerLurker, 2021-01-24:
Looks like [boost](https://www.boost.org/users/download/) needs to be available in order to compile.  I have one error showing this is missing on my PC, and two others probably due to missing boost.

FormerLurker, 2021-01-24:
Hey, @ETE-Design, perhaps we should move this to the [discussions](https://github.com/FormerLurker/ArcWelderPlugin/discussions/) page?  I just realized that people who commented on this issue are getting emailed about our posts.