FormerLurker/ArcWelderPlugin #57: Edit G-Code results in second-processing of file

codefaux, 2020-10-29:
Me again. Still loving the plugin but I broke a wire on my heater sensor replacing a heater core so I'm down for a few days. Unrelated, but agony must be shared, hah...

_Anyway_ - after uploading a g-code file, I occasionally need to edit it for small corrections.

I use Klipper, and change acceleration/corner speed/offset values via its extensible G-Code. Occasionally, I need to edit something like the Z-offset if I've just re-applied a bed coating or changed temperatures etc. I've found that upon saving, AW will process the file it had already processed.

1 - This results in filename.aw.aw.gcode (maybe check for the configured prefix/suffix in the filename in their enforced locations before processing, add option to block, repeat-into-new, or repeat-and-overwrite)

2 - Interestingly, without changing any geometry it will sometimes find and process new arcs - usually not many - which doesn't really so much matter but I thought you might find it interesting.

Keep up the good work!

FormerLurker, 2020-10-29:
How are you editing the gcode?  If you are using a plugin, it should propagate the file metadata if it creates a new file and deletes the original.  This sounds like an issue with another plugin.

codefaux, 2020-11-09:
Apologies for the delay, I must've missed the email.

The "Edit" I speak of is actually a plugin; it integrates well so I had forgotten it wasn't stock OctoPrint. I'll raise an issue with them.

To be fair, though - let's pretend I did something via CLI like pulling up nano and adjusting a number....or I triggered an external script, or re-uploaded a file that had been previously processed in order to re-print it, etc etc..

The addition of the "is my prefix/suffix already present" check within your plugin before processing a file wouldn't interfere with literally any functionality, and would provide a layer of "defense" against incorrectly written plugins, unexpected/unusual user actions, etc.

FormerLurker, 2020-11-10:
So, arc welder adds metadata to indicate that a file was welded.  Unfortunately there is no way to add metadata when a file is added.  My guess is that the gcode editor plugin readds the edited file, then copies metadata.  The alternative would be to rewrite the file in place and manually trigger analysis.  Im not sure what the plugin is doing (haven't looked at the source), but that is a reasonable guess.

Also, the file prefix and suffix isnt the best way to check for welding since they can be changed at any time.  However, I may be able to read the first few lines of the gcode file.  I add some info when the file is welded.  Will see what can be done there.

FormerLurker, 2020-11-21:
I believe I've fixed this without relying on changes to the gcode editor plugin.  I now scan the gcode for any indication that ArcWelder has processed the code, and abort if I find any.  You're welcome to try it out!  Let me know.

codefaux, 2020-11-24:
I would be more than happy to give it a test - I just don't know how without a proper plugin repository update.

I'm deeply familiar with self-compilation, Github usage, I spend most of my time in terminal windows, etc -- I would just need a plain english description of how to staple it onto Octoprint. Is there a mechanism to add github/source-repository plugins to Octoprint, or would I need to compile a local copy and add it myself?

FormerLurker, 2020-11-24:
Oh, it's really easy.  Open the plugin manager, click 'Get More' then enter this into the 'From Url...' box:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/d940e574a65398a4b90a46c7e74b890ccf8eb4fc.zip```

If you ever want to install from a specific branch, you can follow the above steps and find the URL you want like so:

[image]

1.  Pick branch.
2.  Click the code drop down.
3.  Right click on the 'Download Zip' link and copy the link.

codefaux, 2020-12-07:
Super handy - sorry for disappearing, my main PC went down and I got lost, lol. I'll install that now, but testing might be delayed a bit further - I'm doing a hardware modification to the printer and that's pending a parts order.

codefaux, 2020-12-07:
Reporting back sooner than expected - literally installed the plugin, loaded Thingiverse to find a print, and the roomie walked in, "Hey man, did you design that fan riser for me?"  Guess the modification can wait, lol.

SO - after whipping up a model real fast, I used the Edit plugin *( https://github.com/ieatacid/OctoPrint-GcodeEditor ) to adjust the Z-offset the 0.025mm I needed with the new coating, and it half-worked. Nice job with the detection - not sure where metadata is getting lost, but that's out of your hands now I assume, unless you'd like further information anywhere.

[image]

FormerLurker, 2021-01-23:
Sorry it took me so long to respond, but I have fixed this issue.  Try installing with this URL in the plugin manager:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/devel.zip```

It should hopefully now report that arcs are enabled.  Basically I'm replacing the ```G2/G3 NEITHER I NOR J GIVEN``` message with an OK, and using that to determine if arcs are enabled.

codefaux, 2021-04-02:
Hey. Yeah uh, sorry for disappearing myself. My dog died so I uh... yeah.

I assume by now those changes have gone mainline, or is it in a to-be-tested state due to Klipper's limited deployment? Would you still like feedback on that version? Is there any other way I could be of help to you? If not, I believe I can close this issue.