FormerLurker/ArcWelderPlugin #61: Ender 3 with SKR Mini E3 1.2

locka99, 2020-10-31:
As per this [discussion](https://www.reddit.com/r/octoprint/comments/jhoani/arc_welder_anyone_try_it_out/).

My experience of using the plugin with the SKR Mini E3 didn't start out too well since I thought the print was coming out blobby. 

But actually my firmware was ignoring the arc commands and just printing the infill. BigTreeTech supplies prebuilt firmware for the board on their [github page](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3). None of these prebuilt firmwares contain arc support, so in order to utilise the plugin you will must build your own firmware.

Since I have a BL-Touch and need to build auto bed levelling I had to optimize the build to make the firmware fit in memory. The optimizations described below. You may or may not need these optimization depending on your need to include ABL support or some other feature.

Read a good tutorial on installing Visual Studio code and PlatformIO before proceeding.

1. `git clone https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3.git`
2. Open Visual Studio Code on the folder `BIGTREETECH-SKR-mini-E3/firmware/V1.2/Marlin-2.0.7.2-SKR-mini-E3-V1.2`
3. Edit `platformio.ini` and add the compiler optimizations below.
4. Edit `Marlin/Configuration.h` and enable the settings described below.
5. Edit `Marlin/Configuration_adv.h` and enable ARC_SUPPORT described below.
6. Hit Ctrl+Alt+B to build.
7. Copy the `.pio/build/firmware.bin` to an micro SD, insert into the printer and turn it on.

With the changes below I was able to squeeze the functionality I needed from the firmware.

# Changes to platformio.ini

Space-optimization - Add this line under the section in `[platformio]` in `platformio.ini`. This enables compiler optimization flags that reduce the space of the binary.

```
build_flags   = ${common.build_flags} -fno-tree-scev-cprop -fno-split-wide-types -Wl,--relax -mcall-prologues -finline-limit=3
```

# Changes to be made to `Marlin/Configuration.h`:

1. Space-optimization - Comment out "#define SHOW_BOOTSCREEN"
2. Space-optimization - Comment out "#define SLIM_LCD_MENUS" (this removes some more advanced menu options)
3. For auto bed levelling uncomment "#define BLTOUCH", "AUTO_BED_LEVELING_LINEAR" (or another ABL option) and "Z_SAFE_HOMING"

# Changes to be made to `Marlin/Configuration_adv.h` 

1. Uncomment "#define ARC_SUPPORT" for arcs.

FormerLurker, 2020-10-31:
Awesome 👍

I am currently working of a method for testing firmware support, including help files for specific printers.  Your post will be quite useful.

Can you send your printer an M115 and paste in the response?  That will help me identify the firmware fork.

locka99, 2020-10-31:
Send: M115
Recv: FIRMWARE_NAME:Marlin 2.0.7.2 (Oct 31 2020 14:58:34) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:0
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:1
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:1
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok
Send: M155 S2
Recv: ok
Recv:  T:168.28 /0.00 B:47.11 /0.00 @:0 B@:0
Recv:  T:165.83 /0.00 B:46.38 /0.00 @:0 B@:0

FormerLurker, 2020-10-31:
Excellent, thank you!

FormerLurker, 2020-11-08:
So, this check isn't perfect, and there are still some things I need to fix, if you want to try it out you can install from the plugin manager by clicking 'Get More' then adding this line to the 'From Url...' box and clicking install:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/d92528cdebf158f67b906caf99c50a35c7fcbc4e.zip```

Once the install is finished, restart octoprint and make sure your printer is connected (this is one of the things that's broken).  Then go to the arc-welder tab and click on the new 'Check Firmware' button.  It is possible that a check has already completed.  

Here is what it looks like:

[image]

If you have any problems running the checker, go to the arc welder settings and turn on verbose logging for the firmware checker like so:

[image]

Clear the current logs:

[image]

Save your settings changes:

[image]

Click on the firmware check in the Arc Welder Tab:

[image]

Then download and post your log file at gist.github.com.  You can download it via the Octoprint logging menu, or go to the arc welder settings and click download log:

[image]

I've still got to add help files for various firmware versions with known issues, but for now you can just send me a snapshot of the checker and I'll let you know what your options are.

Deses, 2020-12-20:
> Since my Ender 3 has a BL-Touch, I also need to build support G29 auto bed levelling and the firmware wouldn't fit with both options enabled. 

I have the exact same board and it has space for 512 kb firmwares... I have everything and the kitchen sink enabled without problem.

FormerLurker, 2020-12-28:
@Deses , thanks for pointing this out!  I'm learning a lot about specific boards and firmware, and it all helps!

Also, I just dropped a new rc/devel version of ArcWelder.  Feel free to check it out.  The software update is broken for all previous versions if you are running OctoPrint 1.5+, so you'll have to install it manually.  Check out [this issue](https://github.com/FormerLurker/ArcWelderPlugin/issues/106) for details.

K, gonna close this, thanks both of you, and let me know if I should reopen.

closed 2020-12-28 by FormerLurker

Deses, 2020-12-29:
Thanks for the heads up! I'll try it soon.