FormerLurker/ArcWelderPlugin #63: Ender 3/SKR Mini E3 - Pausing during print?

kylane, 2020-11-03:
Ahoy hoy,

Have manually converted a number of files now, and they all have weird pauses in them during print.

Here is the google drive link with the original STL, Cura generated gcode, the arcwelder generated gcode, and a video of the issue.

[https://drive.google.com/drive/folders/1azFBl4Nw46e0tY-k-awuaLikPpCOyzOP?usp=sharing](https://drive.google.com/drive/folders/1azFBl4Nw46e0tY-k-awuaLikPpCOyzOP?usp=sharing)
Any advice would be awesome. Cheers!

FormerLurker, 2020-11-04:
You provided a lot of info, but I need more (LOL!  Always need more...)

Can you do the following:

1.  Send an M115 command via the octoprint terminal and attach the printer's response.
2.  Send a naked ```G2``` command (no parameters) and paste in the response (I suspect this will work based on your video).
3.  If you have your firmware configuration files, or a link to the firmware you installed (source is best, link to the binaries will do).

I have some suspicions as to what is going on (running marlin < 2.0.6), but it's hard to tell without the above.  If I'm right, you should be able to fix it pretty easily.  I'm not an expert in the Ender world, but I am pretty sure your board has plenty of mem to get all the goodies compiled and installed.

Thanks for your submission!

kylane, 2020-11-04:
@FormerLurker - Thanks for that - in the middle of a print atm - will get back to you.

FWIW: Its the stock SKR Mini E3 1.2 (Ive never flashed it).

kylane, 2020-11-04:
@FormerLurker 

M115 returned:
Send: M115
Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:SKR-mini-E3-V1.2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFE1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:0
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:CHAMBER_TEMPERATURE: 0

G2 returned:

Send: G2
Recv: Error:G2/G3 bad parameters
Changing monitoring state from "Operational" to "Error: G2/G3 bad parameters"
Send: M112
Send: N208765 M112*31
Send: N208766 M104 T0 S0*28
Send: N208767 M140 S0*89
Changing monitoring state from "Error: G2/G3 bad parameters" to "Offline (Error: G2/G3 bad parameters)"
Connection closed, closing down monitor

Trevor-Miller, 2020-11-07:
For reference @kylane, I have an SKR Mini E3 1.2 (which I have configured and flashed) and this is what M115 returns for me (note the firmware version and ARCS:1). Take a look at [this section of the Arc Welder readme](https://github.com/FormerLurker/ArcWelderPlugin#marlin) as well and compare your results to what's mentioned here

Generally if you see weird behavior with Arc Welder it's important to first upgrade Marlin to 2.0.6.x or greater because they added significantly improved arc support.

```
Recv: FIRMWARE_NAME:Marlin 2.0.6.1 (Sep 27 2020 16:48:53) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 Pro EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRA1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:0
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
```

FormerLurker, 2020-11-08:
The firmware check is almost ready.  Just debugging some final issues.  I will add your posted responses to the lookup.  Also, kudos for having the extended capability report option enabled, it makes everything easier

FormerLurker, 2020-11-08:
So, this check isn't perfect, and there are still some things I need to fix, if you want to try it out you can install from the plugin manager by clicking 'Get More' then adding this line to the 'From Url...' box and clicking install:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/d92528cdebf158f67b906caf99c50a35c7fcbc4e.zip```

Once the install is finished, restart octoprint and make sure your printer is connected (this is one of the things that's broken).  Then go to the arc-welder tab and click on the new 'Check Firmware' button.  It is possible that a check has already completed.  

Here is what it looks like:

[image]

If you have any problems running the checker, go to the arc welder settings and turn on verbose logging for the firmware checker like so:

[image]

Clear the current logs:

[image]

Save your settings changes:

[image]

Click on the firmware check in the Arc Welder Tab:

[image]

Then download and post your log file at gist.github.com.  You can download it via the Octoprint logging menu, or go to the arc welder settings and click download log:

[image]

I've still got to add help files for various firmware versions with known issues, but for now you can just send me a snapshot of the checker and I'll let you know what your options are.

FormerLurker, 2021-01-23:
This issue has been inactive for a while, so I'm going to go ahead and close it out.  Just message me if you want it re-opened, and thanks for posting!

closed 2021-01-23 by FormerLurker