FormerLurker/ArcWelderPlugin #64: Corrupted GCode

DunklerPhoenix, 2020-11-07:
Heho.
I tried to print the little Testship first normal and after this with arcwelder.
Slicer is Ultimate Cura:

**Normal:**
Gcode: [AI3M_3DBenchy.gcode.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5505230/AI3M_3DBenchy.gcode.txt)
[image]
Prints also normal

**Arc Welder:**
Gcode: 
[AI3M_3DBenchy.aw.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5505235/AI3M_3DBenchy.aw.txt)
[image]
[image]

Specs:
RPI4 8GB
Octopi: Version 0.17.0, running on Raspberry Pi 4 Model B Rev 1.2
Octoprint: Version 1.4.2
Printer: AnyCubic Mega-X with TMC2208 and Marlin 2.0.5.4

FormerLurker, 2020-11-08:
Your visualizer appears to not support G2/G3.  This is what I see when I run your welded code through simplify 3d:

[image]

I'm guessing arc support is disabled in your firmware.  I've been working on a firmware checker for the past couple weeks, and I'm almost ready to push a development release.  I'll send you a link when I'm done, along with instructions for installing in OctoPrint so you can try the 'Check Firmware' button and see what it says.

FormerLurker, 2020-11-08:
So, this check isn't perfect, and there are still some things I need to fix, if you want to try it out you can install from the plugin manager by clicking 'Get More' then adding this line to the 'From Url...' box and clicking install:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/d92528cdebf158f67b906caf99c50a35c7fcbc4e.zip```

Restart and make sure your printer is connected (this is one of the things that's broken).  Then go to the arc-welder tab and click on the new 'Check Firmware' button.  It is possible that a check has already completed.  

If you have any problems running the checker, go to the arc welder settings and turn on verbose logging for the firmware checker like so:

[image]

Clear the current logs:

[image]

Save your settings changes:

[image]

Click on the firmware check in the Arc Welder Tab:

[image]

Then download and post your log file at gist.github.com.  You can download it via the Octoprint logging menu, or go to the arc welder settings and click download log:

[image]

I've still got to add help files for various firmware versions with known issues, but for now you can just send me a snapshot of the checker and I'll let you know what your options are.

DunklerPhoenix, 2020-11-09:
The first log was directly after pressing Check Firmware (after the little loading icon disappear)
The second log is after a quick octoprint restart and another check.
As you can see in the screenshot the printer is connected and ready, but your tool didn't recognize it.

[image]
[plugin_arc_welder (1).log](https://github.com/FormerLurker/ArcWelderPlugin/files/5507549/plugin_arc_welder.1.log)[plugin_arc_welder (2).log](https://github.com/FormerLurker/ArcWelderPlugin/files/5507548/plugin_arc_welder.2.log)

FormerLurker, 2020-11-09:
Ok, thanks for the logs!  They should be very helpful.  I will get to work

FormerLurker, 2020-11-21:
FYI, I think I fixed this on the latest devel branch.  Doing a bit more testing then I'll shoot you a link.  LOTS of new goodies in there :)

FormerLurker, 2020-11-24:
OK, it's nearing devel/rc.  Here is a link to the latest devel commit you can use to install via the plugin manager: ```https://github.com/FormerLurker/ArcWelderPlugin/archive/d940e574a65398a4b90a46c7e74b890ccf8eb4fc.zip```

I recommend you uninstall and clean Arc Welder via the plugin manager first, as the firmware types dictionary has changed forms, and might not work correctly otherwise (I only finalize/test migrations from release to release).

Let me know how it all looks (new UI) and if the firmware checker is doing what it's supposed to.  Also, I'd be interested in what you think about the default settings.

DunklerPhoenix, 2020-11-24:
Thanks for your work. There are so many changes \*-\*

Ok. This is the Firmwarecheck. The tableview is a bit strange. What do you mean with "Edit the Arc Welder settings, enable the G90/G91 Influences Extruder setting, and run the firmware check again."? Is this setting in your Plugin or should I search this in the firmware.
[image]

And now the benchy. Looks still a bit broken in viewer, but I can't print it at the moment.
[3DBenchy.aw.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5591901/3DBenchy.aw.txt)
[image]

FormerLurker, 2020-11-24:
Hmm.  I wonder what is up with the tab.  Doesn't look right at all!?  Do you have any custom themeify rules?  Try disabling that and see if it fixes the layout.  

G90/g91 influences extruder is in the arcwelder settings.  You may have 'use octoprint settings' checked (octoprint has a setting for this too, and that needs to be fixed as well!).  Click the help button in the arc welder settings (blue question mark) for more info.

What did you use to visualize benchy?  Most don't support g2/g3 yet.  Post the gcode and I will run it through simplify 3d for you.  Ncviewer.com may work too, but it has some issues.

FormerLurker, 2020-11-24:
FYI, the firmware checker should look like this:

[image]

If it does look like the issue is related to themeify, send me your custom CSS and I'll take a look.  I've seen some rules floating around that affect ALL bootstrap columns, and have been giving people a less intrusive rule that seems to play nicely with bootstrap rows.  Look within the themeify settings here for that:

[image]

Copy out the selector, css rule and value.

DunklerPhoenix, 2020-11-24:
Ah I see. Didn't realize that both are your plugins ^^'

Yep its the viewer... https://github.com/Kragrathea/OctoPrint-PrettyGCode/issues/52
The GCode is over the image in my last post. But seems good in Simplify3D

I changed the G90/G91 Setting under Function in Octoprint Settings, but it shows _Arcs Enabled:Unknown_. 
As you seen in the Issue https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/38 it should be enabled. 
For which value are you looking?

I disabled themify and yes its one of its rules, but its not an big issue for me.

Sidenote: Found a lil typo 
[image]

I try print it now.

FormerLurker, 2020-11-24:
It does look like G2 is enabled, but the checker wasn't 100% sure so it reported 'unknown'.  Once your test print is complete, try sending these two commands in the OctoPrint terminal and paste in the responses:

```
G2 ; First thing
```

and

```
M115
```

I might be able to make it report enabled if I know how it's responding to the above.

DunklerPhoenix, 2020-11-24:
I'm not printing currently cause octolapse don't want to:
[image]
I set the values from 0,75mm / min 0,05mm to zero, but it still don't like the gcode.

It's caused by the G90/G91 setting.

DunklerPhoenix, 2020-11-24:
Ahhhhh... I think Arc is not enabled...
```
echo:Unknown command: "G2"
echo:Unknown command: "G3"
```

FormerLurker, 2020-11-24:
See this:

[image]

These are both my plugins, lol!  That setting is critical unfortunately and most ppl don't have it set correctly.  

I recommend you disable Octolapse for your first test print with Arc Welder.  It's best to see how it runs on its own for testing.  Once you verify that benchy prints OK, enable Octolapse and see what happens.  If the G90/G91 setting is correct (and my firmware checker detected it properly) you might have better quality with Octolapse, unless for some reason this feature is different on the anycubic.  When you paste in the M115 response I should have more info about that.

DunklerPhoenix, 2020-11-24:
```

G2
echo:Unknown command: "G2"
ok
 T:28.22 /0.00 B:21.25 /0.00 @:0 B@:0
 T:28.19 /0.00 B:21.17 /0.00 @:0 B@:0
 T:28.12 /0.00 B:21.25 /0.00 @:0 B@:0
 T:28.00 /0.00 B:21.25 /0.00 @:0 B@:0
 T:28.00 /0.00 B:21.25 /0.00 @:0 B@:0
 T:28.00 /0.00 B:21.21 /0.00 @:0 B@:0
M115
FIRMWARE_NAME:Marlin 2.0.5.4 (knutwurst, Github) KNUTWURST_AI3M_VERSION:1.1.6 DISTRIBUTION_DATE:2020-09-11 SOURCE_CODE_URL:https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S PROTOCOL_VERSION:1.0 MACHINE_TYPE:Knutwurst's Anycubic i3 MEGA EXTRUDER_COUNT:1 UUID:b6153152-86fb-4cd8-bda7-91c6a04c467d
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:1
Cap:PROMPT_SUPPORT:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:CHAMBER_TEMPERATURE:0
ok
M155 S2
ok
```

FormerLurker, 2020-11-24:
Looks like arcs are disabled!
```
G2
echo:Unknown command: "G2"
```

The firmware checker should have been reporting this as completely disabled.  I will try to fix this and will send you a link to see if it reports a failure.  

Then maybe you can ask  @knutwurst to pull in the changes from at least Marlin 2.0.6 (preferably the newest version available) since arc support was greatly enhanced.  However, Arc Welder will not work until arcs are enabled on your firmware.  Let's see what we can do after I fix the firmware checker.

FormerLurker, 2020-11-24:
Ok, I think I may have figured out what's going on with the 'Arcs Enabled' detection.  Looks like your printer added 'echo' to the beginning, which is fine but my checker wasn't expecting that.  Try installing from this link:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/a564139f54dc4c7d2a513b5f1f9bbf9c86ecd3a3.zip```

It should now report an error and say that arcs are disabled.  Then we can figure out what we can do to get them enabled for you.

DunklerPhoenix, 2020-11-24:
Printer is connected and the Logfiles of your plugin (after enabling verbose) are empty. I did also purge your plugin before installing it again.
[image]

```

 T:32.76 /0.00 B:33.75 /0.00 @:0 B@:0
M115
FIRMWARE_NAME:Marlin 2.0.5.4 (knutwurst, Github) KNUTWURST_AI3M_VERSION:1.1.6 DISTRIBUTION_DATE:2020-09-11 SOURCE_CODE_URL:https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S PROTOCOL_VERSION:1.0 MACHINE_TYPE:Knutwurst's Anycubic i3 MEGA EXTRUDER_COUNT:1 UUID:b6153152-86fb-4cd8-bda7-91c6a04c467d
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:1
Cap:PROMPT_SUPPORT:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:CHAMBER_TEMPERATURE:0
ok
M155 S2
ok
G2
echo:Unknown command: "G2"
ok
 T:32.76 /0.00 B:33.75 /0.00 @:0 B@:0
 T:32.74 /0.00 B:33.75 /0.00 @:0 B@:0
 T:32.48 /0.00 B:33.75 /0.00 @:0 B@:0
 T:32.44 /0.00 B:33.75 /0.00 @:0 B@:0
 T:32.41 /0.00 B:33.75 /0.00 @:0 B@:0
```

FormerLurker, 2020-11-24:
OK, I may have been a bit too hasty with that update.   Nothing in the logs?!  That's unfortunate

FormerLurker, 2020-11-24:
So, the problem must be in the 'Arcs Enabled' detection when 'Unknown Command' is received.  You didn't have that error the first time because the check wasn't working.  My printer doesn't respond that way, so it will take me a bit of time to get a simulation set up based on the M115 response you posted.  I'll message you back as soon as that's done.  I MAY have it done overnight.  If not, it will be finished this weekend for sure..  I need a few test printers (only have one)...

FormerLurker, 2020-11-26:
OK, I figured it out.  When i was trying to fix the G2/G3 detection for you earlier I included a syntax error (was thinking in c#, whoops).  That should be corrected, and hopefully will teach me to test my quick fixes, lol

Here is a link you can install from:  https://github.com/FormerLurker/ArcWelderPlugin/archive/5d89642fb879a6e7af34db3a3e2fed37b2f65e40.zip

When you test your firmware, you should now see this:

[image]

Please let me know if that works (or rather that arcs do NOT work), then we can take a look at the firmware.

DunklerPhoenix, 2020-11-27:
Sorry that i answer so late.
Its working.

FormerLurker, 2020-11-28:
Awesome, thanks!  Now maybe we can get your firmware working.  Might just need to reflash.  If it was a while since you flashed the firmware, maybe it has changed?  Otherwise I will review the code to see what is goit on.

DunklerPhoenix, 2020-11-28:
I have this https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.1.6 release of the firmware. It's the last stable. The next would be 1.1.7beta, but the last time I had some problems with it.

DunklerPhoenix, 2020-11-29:
Ok Knutwurst pushed now version 1.1.7 as stable. 
With this version G2 and G3 exists, but the printer is instantly disconnecting after sending G2 or G3 without anything with this message:
[image]

FormerLurker, 2020-11-29:
Hmm, does it report that when running the firmware checker?  That shouldn't be happening.  If it reports bad parameters, that is actually good.  What version of octoprint are you running (in case you upgraded)?  Will have to look into that because the printer should NOT be disconnecting because of that.  Maybe there is an octoprint setting somewhere.  

In the meantime, disabled the firmware checker in the arc welder settings and it should work..  can't quit now when we are SO close!

DunklerPhoenix, 2020-11-29:
I have OctoPrint 1.4.2 .
With FW Check set to manually it doesn't disconnect

FormerLurker, 2020-11-29:
OK, I've found a workaround in the OctoPrint settings for this, but I believe I can do better.  I'm going to try to catch the 'Bad Parameters' response and trap it before Octoprint gets wind of the error.  Might be able to handle it gracefully.  give me an hour or two to get this working and I'll send you a link.

FormerLurker, 2020-11-29:
OK, hopefully this fixes the disconnect error when using the firmware checker.

By default, if OctoPrint receives an 'ERROR' from your printer, it assumes it is serious and disconnects.  You can disable this, but it's not recommended since many times that is the desired behavior.  In this case, I was using the error itself for diagnostic purposes (does G2/G3 exist??), but it was triggering this safe guard.  What I've done is to catch the error message and re-write it as an OK since the response was expected (and actually helpful).  Now I have to say this:  I've only been able to test this by pretty much hacking the system.  My printer doesn't respond like yours, nor does the virtual printer I've used to debug.  There may be problems.  However, you've stuck through this for 22 days (quite the achievement), so I am hopeful we can finally achieve a resolution!

Here is the link:  https://github.com/FormerLurker/ArcWelderPlugin/archive/0a553aa3418498e631e7fd9f814334433ea53840.zip

DunklerPhoenix, 2020-11-29:
🥳 It works

FormerLurker, 2020-11-29:
Oh my gosh, AWESOME!!  Now to try a print :) i am looking forward to closing this issue and I will bet you are too.  Fingers crossed....

DunklerPhoenix, 2020-11-30:
Sorry for the waiting time. I had some slicer problems :D 

Here are the print results of the normal and the "arced" gcode.
I think the Firmware or my board doesn't support it correctly.
[3DBenchy (1).txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5618304/3DBenchy.1.txt)
[3DBenchy.aw (1).txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5618305/3DBenchy.aw.1.txt)

[image]
[image]
[image]
[image]

FormerLurker, 2020-11-30:
OMG, so I hate to ask you this, but:

1.  Were you using firmware compensation?
2.  If so, would you be willing to try it one more time?

Bad slicer settings are OK as long as the benchy on the right doesn't look like...   well like it does, lol!  The reason I'm asking about firmware compensation is that a problem was discovered that could be affecting things.  It's now fixed and I can send a link, but I understand if you are tired of working on this.

DunklerPhoenix, 2020-11-30:
I'm not tired. The printer is the most time not in use xD

nope the firmware compensation is disabled.

I can try it as often as you want :D

FormerLurker, 2020-11-30:
OK, in that case let ME test some stuff for a change.  I'll see if I can adjust your gcode to work on my printer (with as little adjustment as I can get away with), and will try it there.  If it succeeds then we know it's firmware related.  If not, back to the drawing board.  I'm sad that the firmware checker is reporting success, but it doesn't work when you print :(  All indications are that it should work!

Give me a few days to get back with you, and thanks again for sticking with this!

DunklerPhoenix, 2020-11-30:
I'm sorry for the problems that I create ^^ It's not intentionally

FormerLurker, 2020-11-30:
> I'm sorry for the problems that I create ^^ It's not intentionally

LOL!  You definitely didn't make them.  Might be me, might be firmware, who knows, but not you, lol.

FormerLurker, 2020-12-05:
OK, so I suspect the mm_per_arc_segment = 0 is causing this issue (as was noted in the firmware issue thread).  Still needs some testing.

DunklerPhoenix, 2020-12-16:
So I think I close this Issue here now. 
Thanks for your help.

Summary:
There were two different problems.
1. I used an Octoprint plugin (PrettyGCode) [ https://github.com/Kragrathea/OctoPrint-PrettyGCode/issues/52 ] as a viewer that does not support arcs.
2. Arcs in Knutwurst's firmware were deactivated.

closed 2020-12-16 by DunklerPhoenix

FormerLurker, 2020-12-16:
> I used an Octoprint plugin (PrettyGCode) Kragrathea/OctoPrint-PrettyGCode#52 as a viewer that does not support arcs at this moment. (should be fixed now)

Yes, I patched that myself ;)

> Arcs in Knutwurst's firmware were deactivated.

Not actually deactivated, but misconfigured.  mm_per_arc_segment was set to 0 :(

Very glad this is taken care of!  Let me know how things go, and feel free to post any time!

DunklerPhoenix, 2020-12-16:
Thank you for your hard work. If you need someone for testing, I'm here :D