FormerLurker/ArcWelderPlugin #69: Intercepting of Uploaded Files [Suggestion]

Cutwow, 2020-11-08:
I do not know if this is possible, but it would be nice for the plugin to "intercept" the g-code sent directly to print from programs such as Cura. This would make it so Octoprint would not use the uploaded g-code but instead the ArcWelded g-code. This would make it more streamlined, as you would be able to still use the "print using Octoprint" feature inside of cura and other softwares and use the post processed g-code instead of the original g-code uploaded.

FormerLurker, 2020-11-08:
Working on a cura script now.  Got a sample py file from another user and will see what can be accomplished with the console library.  It is easy to do in slicer, just add the console app as a post processor.  Cura is a bit trickier.

FormerLurker, 2020-11-08:
Oh, and i do have a new 'print after processing' option that may do until the cura plugin is ready.  Let me know if you wouy like to try that out

Cutwow, 2020-11-08:
I would love to try it out! Currently flashing new firmware to my printer to enable the Arc command. So I will hopefully be able to test that out soon.

Cutwow, 2020-11-08:
Ok, got my printer flashed and all is working.

FormerLurker, 2020-11-08:
Here is an install url for the beta:  https://github.com/FormerLurker/ArcWelderPlugin/archive/d92528cdebf158f67b906caf99c50a35c7fcbc4e.zip

You can install via the plugin manager by pasting the link in the 'from url...' box.

A few things are not 100% ready, but it is close.  The firmware checker needs the most work.  Let me know how it goes!

Cutwow, 2020-11-09:
Will do. Currently printing some Flex, so after that I will install and try to print directly from Cura.

Cutwow, 2020-11-09:
OK, looking at your firmware_checker.py file I have a few questions:
My printer dose not have EXTENDED_CAPABILITIES_REPORT enabled, so the M115 command dose not output the EXTENDED_CAPABILITIES_REPORT. Will this effect the capabilities of this build or the functionality of ArcWelder?

FormerLurker, 2020-11-09:
If the firmware checker doesn't see Arcs:0 in the extended capabilities report, it will send a naked G2 command.  Depending on your firmware, it may return a an 'unknown command' response, or an 'invalid parameters' response, or an OK.  I haven't completed the invalid parameters portion yet (I just learned about that), but it would indicate that arcs are enables.  Unknown command means they definitely are not enabled.  An OK response is ambiguous (may be enabled, may not be).  

I then attempt to perform a look up to find the firmware in a list of known firmwares.  If I find it, I try to match the firmware version in my known firmware file.  I can't tell if arcs are enabled that way, but I can say that they are supported or not.  Some firmware versions are simply incompatible, and others have know issues.  I report thos back with the test results.

I am also going to add a help file for some of the versions explaining any issues, mitigations, or resolutions I am aware of.

Before I release, i am going to create a repository to store the firmware lookup data in a versioned for..  That way I can add more firmware versions as I discover them.

I probably explained more than you ever wanted to know.  Hope it makes sense!

Cutwow, 2020-11-09:
No, that was perfect. I guess I just missed the part wehre it sends the G2 command. I was really confused why it wouldn't. My printer returns the invalid paramteres when a naked G2 command is sent. As you said you hadn't finished this will this negativaly effect my experience. 

Also, you might want to add in a override for any edge cases. So the user can manually say that the arc commands work.

FormerLurker, 2020-11-09:
In that case, you can disable the firmware check in the settings.  I've only tried it on my prusa and a virtual printer, so it still needs quite a bit of testing and improvement.

Cutwow, 2020-11-09:
Sounds good! Will report back in the coming days

Cutwow, 2020-11-09:
Ok, so the print after processing works really well, the only thing is, that if I print from cura it still prints the original file. As cura dose not just upload but also starts the print, which I hope will fixed with the cura plugin. But for now, this is a fine solution.

FormerLurker, 2020-11-09:
Yeah.  I recommend disabling the option to start printing from Cura, and enable 'Start Print After Upload' in Arc Welder.  Just remember you enabled it, cause it will start printing after you upload any gcode file, with or without cura!

I'm working on a cura plugin script too.  Got a working script sent to me, and I'm working to add parallelism and to test it.

Cutwow, 2020-11-09:
Ya, mabey you could make it so when parsing the g-code it checks to see if
;Generated with Cura_SteamEngine 4.4.1
Is in it, then only auto starts if the word Cura is in the g-code. that would be a quick and dirty way to autostart with cura and only cura.

FormerLurker, 2020-11-09:
Actually that is a pretty good idea, thanks for the suggestion!  This is feasible and simple.  I may need to add checks for other slicers too, which shouldn't be hard.  Maybe I could add custom triggers for slicers that are uncommon, or some token that can be added to the start gcode.  I'll create a new issue and add this to the milestones for 1.1.

Cutwow, 2020-11-09:
Sweet

FormerLurker, 2020-11-09:
Do you mind if I close this out?  If not, we can continue the discussion [here](https://github.com/FormerLurker/ArcWelderPlugin/issues/70).

Cutwow, 2020-11-09:
Not at all, I will make another issue, if somthing comes up

Cutwow, 2020-11-09:
Glad I could help

closed 2020-11-09 by FormerLurker