FormerLurker/ArcWelderPlugin #72: Suggestion | Folders

Cutwow, 2020-11-09:
Add an option to allow the user to do the following:
Add all ArcWelded files to a Folder labeled "..."
Add all original files to a Folder labeled "..."
Separate original and ArcWelded files into separate folders
Add original and ArcWelded file into Folder with name of original file

This could be an alternative to people that want there files page to be neat, but do not want to delete any of the G-Code.

Cutwow, 2020-11-09:
Also options to move either the original or post-processed file to SD card

FormerLurker, 2020-11-24:
Sorry I haven't responded to this yet, but I've been mulling this over.  I'm not going to add this to the next release (1.1) since it's already long overdue, but I'll think about this for 1.2.  No promises at this point since I'm not sure how I feel about it.  On the one hand it would be a neat feature, and on the other hand it will significantly add to the complexity of creating/overwriting/renaming the source/target file and somewhat increase the growing number of settings.  That being said, maybe there are ways around all of that.  I will think about it, mark this as an enhancement, and watch for any other user reactions.

In any case, thanks for posting!