FormerLurker/ArcWelderPlugin #80: Add Firmware Compensation for Older Arc Interpolation

FormerLurker, 2020-11-23:
Some firmware, especially those based on an older version of Marlin, have only the MM_PER_ARC_SEGMENT setting to control arc interpolation.  The default value is 1mm, which can result in noticeably flat edges on smaller arcs.  This can be somewhat compensated for within the arc generation routine at the cost of less compression.

Add two new settings in a new section called 'Firmware Compensation':

MM Per Arc Segment - The value of the MM_PER_ARC_SEGMENT define within the compiled firmware
Min Arc Segments - This will simulate the MIN_ARC_SEGMENTS firmware parameter

The general idea is that when an arc is created we will check how may segments would be interpolated along a circle of the same radius using the MM_PER_ARC_SEGMENT value.  If it is less than MIN_ARC_SEGMENTS, then we will check the length of the current arc and see if that length divided by the circumference of a circle with the same radius would produce MIN_ARC_SEGMENTS arcs.  If it does, the arc will be created.  If not, the G2/G3 will be left alone.

The effect of this will be that in some cases no arcs will be generated at all for some circles.  In other cases, instead of having an individual arc, it will be split up into multiple arcs.  This obviously isn't a perfect solution, but it does provide many of the benefits Arc Welder was designed for, while allowing for more accurate prints, even without the fancy new arc interpolation firmware settings.  This will be especially useful when a user can't recompile their firmware for various reasons.

Here is an example of two rows of circles, one separated and the other joined.  Both rows start with a radius of 0.1mm and end at 4.7mm.  This is the gcode produced by Prusa Slicer as visualized by NCViewer:

[image]

Here is gcode produced from the standard *Arc Welder* algorithm:

[image]

You can see that arcs are generated even for the smallest circles.  These would not look very good on firmware with MM_PER_ARC_SEGMENT = 1, since many of these circles are less than 1MM in radius!  Here is what happened when I turned firmware compensation on:

[image]

Here I used a value of 12 for *Min Arc Segments* and 1mm for *MM Per Arc Segment*  You can see that as the circles get smaller and smaller, fewer and fewer arcs are generated.  At a radius of about 3.6mm, full arc generation is seen.  At 0.1mm, no arcs are generated at all.

Here is a close up of the original gcode with a high number of segments:

[image]

And here is the same figure using Arc Welder and the new firmware compensation settings I mentioned above:

[image]

FormerLurker, 2020-11-25:
I realize that the 'wipe while retracting' slicer option made the images above somewhat unclear.  here is another image of the arc welded code using firmware compensation, but with wipe while retracting disabled and curved infill:

[image]

And here is a closeup of the transition from full to partial arcs:

[image]

5axes, 2020-12-01:
in my firware MM_PER_ARC_SEGMENT are set to 1 and MIN_ARC_SEGMENTS to 10 but in my tests the better result seams to be 14 and not 12. At 12 we still have some facettes on some radii.
[image]
Sample test :
[CylindreG1.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5625735/CylindreG1.txt)
[image]
[Cylindre_12.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5625736/Cylindre_12.txt)
[Cylindre_14.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5625737/Cylindre_14.txt)

FormerLurker, 2020-12-01:
OK, I will update my defaults before release.  Let me know if we need an even higher value.  I would l like it as low as possible for reasonable results with most prints, but not too low that there are common and obvious problems.  Setting this too high would negate a lot of the benefits of arc-welder, so something like I would use in my firmware settings (say 24) prob isn't going to cut it.  

Also, I'm quite surprised at the results you are showing in the third cylinder in the 12 arcs row..  Going to look into that.  Possible bug, or unexpected artifact???

FormerLurker, 2020-12-02:
Hey, quick question, I see in your gcode files some very elaborate slicer settings documentation from cura 4.8.0.  Is that a standard feature?  If so, how is it enabled?

5axes, 2020-12-02:
Is that a standard feature? If so, how is it enabled?   >> Yes and No  it's a standard feature because I Use a Postprocessing Script  and no because it's my own script ..  A copy can be found Here :[https://github.com/5axes/Cura-Postprocessing-Scripts]( https://github.com/5axes/Cura-Postprocessing-Scripts)
[https://github.com/5axes/Cura-Postprocessing-Scripts/blob/master/GCodeDocumentation.py](https://github.com/5axes/Cura-Postprocessing-Scripts/blob/master/GCodeDocumentation.py)

FormerLurker, 2020-12-02:
Well that is awesome!  I use a script to inject the settings so Octolapse can suck in the required slicer settings, but it isn't nearly as pretty.  Nice column formatting and indents too!

FormerLurker, 2021-01-23:
I think this feature has been vetted enough to close this issue.  I've updated the defaults per your recommendations (14 min_arc_segments).  Thank you SO much for helping with this @5axes!  It's been a pleasure :)

closed 2021-01-23 by FormerLurker