FormerLurker/ArcWelderPlugin #84: arcwelder fails with UnicodeDecodeError: ascii codec can't decode

susisstrolch, 2020-11-27:
OS: Armbian 20.08.05 Buster, running on hummingboard ARM cpu
Octoprint: Version 1.4.2
ArcWelder: https://github.com/FormerLurker/ArcWelderPlugin/archive/5d89642fb879a6e7af34db3a3e2fed37b2f65e40.zip

When trying to weld a gcode file (created with CURA, running on Linux Mint), I get the following error in octoprint.log:
```
2020-11-27 12:40:16,712 - octoprint - ERROR - Exception on /plugin/arc_welder/process [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python2.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python2.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python2.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python2.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1296, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1315, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_arc_welder/__init__.py", line 312, in process_request
    success = self.add_file_to_preprocessor_queue(name, path, origin, True)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_arc_welder/__init__.py", line 1229, in add_file_to_preprocessor_queue
    ArcWelderPlugin.SEARCH_FUNCTION_SETTINGS
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_arc_welder/utilities.py", line 118, in search_gcode_file
    return _search_gcode_file(f, search_functions)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_arc_welder/utilities.py", line 139, in _search_gcode_file
    comment_start_index = line.find(";")+1
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 22: ordinal not in range(128)
2020-11-27 12:40:16,919 - tornado.access - ERROR - 500 POST /plugin/arc_welder/process (::ffff:192.168.254.135) 233.22ms
```

FormerLurker, 2020-11-27:
Ugg.  Getting ascii and unicode to work in python 2 and 3 is pretty difficult.  I mist have goofed somewhere.  Will take a look later today.  There is a tiny chance that I already fixed this, so fingers crossed :)

FormerLurker, 2020-11-27:
So, if you have a second could you upload your source gcode file?  I'll run it through the debugger, but for now I'll just add some unicode to a gcode file and will see what happens.

FormerLurker, 2020-11-28:
Can you try installing this dev version via the plugin manager with this link in the 'From Url...' box:  https://github.com/FormerLurker/ArcWelderPlugin/archive/e7a6879a16310f095e6e683c8fb4fc5feee2f0e6.zip

Hopefully I have fixed this.  Turned out to be quite nasty to fix, so hopefully I got it right.

FormerLurker, 2020-11-28:
Hmm, actually, do NOT install with that URL.  was running some tests and my Pi locked up.  Not sure if it is related, but better safe than sorry.  I will resend another link ASAP once I figure this out.

susisstrolch, 2020-11-28:
Ok, here are two gcode snippets...
```
; Fußbälle sind rund
M117 heating up to {material_print_temperature_layer_0}°C
M107 ;Fan off
```

FormerLurker, 2020-11-28:
Only got one snippet, but that should suffice.  How would we know that soccer balls are round without the ß and the ä?  Lol.

Also, fyi, while working on this I discovered some issues with the logger that may speed up processing.  Fingers crossed!

FormerLurker, 2020-11-28:
OK, this should work: ```https://github.com/FormerLurker/ArcWelderPlugin/archive/05c850e3461a4bcac0877163f7ab6db78b2c9644.zip```

I'm running additional tests.  I was never able to get the logger to correctly write the unicode (it seems to be escaping them), but it appears to work in the output .gcode.  Python's unicode handling is nuts if you have to deal with both Python 2.7 and 3.x, especially if there are any custom c++ extensions involved...

closed 2021-02-06 by FormerLurker