FormerLurker/ArcWelderPlugin #85: Issue with V1.0.0

5axes, 2020-11-30:
I was working on the Cura PlugIn made by @fieldOfView. This weekend I have made some tests with the ArcWelder console. everything was running fine. Yesterday we have switch on your release 1.0.0 binary files. With the same files and the same options the result is different and much more worryingly, the result in NCViewer also appears to be invalid. 

Sample layer in Cura
[image]

Tested File in NCViewer with the fieldOfView binary ArcWelder.exe file
[image]

The new result in NcViewer ...  (Not tested on the machine for the moment )
[image]

5axes, 2020-11-30:
Sample G01 File
[G01.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5616117/G01.txt)

Options in Cura
[image]

Tested file on the machine  
[TestedCode.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5616132/TestedCode.txt)

New Gode issue from the release V1.0.0  ( I'm using the windows release)
[ResultV100.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5616134/ResultV100.txt)

FormerLurker, 2020-11-30:
Will take a look, thanks for the report.

FormerLurker, 2020-11-30:
Might be a false alarm.  It looks like a series of short arcs kills every viewer I've tried, even simplify 3d.  I have a program that simulates what marlin will do (the ArcStraightener). It replaces G2/G3 with G1 as the firmware would do (using the G2/G3 code from the prusa fork, slightly modified).  Here's the output gcode:

[inverse_output.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5616765/inverse_output.txt)

Here's what that looks like in simplify 3d:

[image]

I'm not particularly happy that no viewers can properly handle G2/G3.  I've had more problems with this than the actual conversion algorithm.  This problem stumped me at first because I've NEVER seen simplify fail to properly render a gcode file.  After reviewing the first arc manually (ug....) everything looked fine, so that's why I thought to use the ArcStraightener.  Anyway, this is going to seriously confuse users, especially since it confused me!

5axes, 2020-11-30:
Well, I'm not so optimistic. I looked in another viewer and the result looks weird. The tested Gcode is fine in this viewer but the new one also has some flaws.   And it's strange that on these areas circular interpolations can  be so segmented as well.
[image]

[image]

FormerLurker, 2020-11-30:
I'm not sure what to trust here honestly.  I know that most viewers have issues, and I also know that the firmware interpolations look correct (down to the number of expected segments).  I will take another look and see if I can find any issues.  Your example looks like the top of a benchy, so I may just print one out with your settings.

Also, fyi (not that it's hurting anything in this case), but the maximum arc radius setting you have is way too low.  This should be on the order of at least 100 meters.  Shouldn't hurt anything, I just figured i'd point it out.

FormerLurker, 2020-11-30:
Also, I'm not sure what you are pointing out in the second image.  Are you questioning why the 'arc' there isn't circular?  If so, it's because of the firmware compensation, and it appears to be doing what it was designed to do.  If you turn that off it will be on big circle.  Sorry if I'm misunderstanding you.

FormerLurker, 2020-11-30:
OK, I think I figured out what you were trying to illustrate on that third pic.  I plotted some of these arcs and did notice the endpoints were off a bit (I need to find a better tool for this, btw... if you know of any please let me know!).  I realized that the current_arc was getting aborted due to firmware compensation, but the underlying circle was being overwritten (a circle is calculated before an arc is generated), which resulted in some small deviations.  I think this is fixed.  The inverse processor wasn't showing this clearly (though it didn't look great obviously)because the very last segment is set to go to the endpoints (X and Y) of the G2/G3 statement, masking the underlying issue somewhat.

Here is the welded output now using the settings you have listed (except I bumped the min radius way up):

[image]

That looks better.  Here's what the firmware will do assuming MM_PER_ARC_SEGMENT = 1 as the only option (using ArcStraightener.exe):

[image]

That also looks much better.  Thanks for pointing this out and insisting you found what you did :)  The previous code that included firmware compensation actually handled this correctly, but I moved some code out of the arc generation routine into the segmented_arc class so I could track how many arcs were aborted.  In doing so I was allowing the currently tracked arc to update and wasn't swapping it back to the previous one.  Before the update this wasn't necessary because the arc wasn't created at all.  Very nice catch, thank you!

I just pushed a fix [you can find here](https://github.com/FormerLurker/ArcWelderLib/actions/runs/392124947).  Let me know if that fixes this for you.

FormerLurker, 2020-11-30:
Oh, also, I'd like to drop an image of what would happen without firmware compensation (again, assuming only MM_PER_ARC_SEGMENT is set to 1.0):

[image]

So, at least for this model it looks like the firmware compensation is doing what it supposed to now :)

FormerLurker, 2020-11-30:
I patched the release and updated the artifacts minus the raspberry pi binary, which I will add soon.  Thanks for testing and for your report!

5axes, 2020-11-30:
Ok tests on machine are OK, I can close the Issue.  On my machine (Ender 3 with Marlin) these test shows that options mm_per_arc_segment and min_arc_segment is a must have. The lower part was made without this option and small Arcs doesn't have the right shape at all.
[image]

closed 2020-11-30 by 5axes

FormerLurker, 2020-11-30:
Thank you!  I am v glad, since I don't have this issue and can't test it well (using custom firmware).  Amazing feedback, but the testing is where I really need help.

FormerLurker, 2020-11-30:
Oh, and btw, that pic is GREAT.  Can I steal it and use it as an example image in the docs?

5axes, 2020-12-01:
No problem , feel free  to use it for your documentation

FormerLurker, 2020-12-01:
Oh, I didn't answer your last question:

> But I still don't understand why with the friday console application release , with the same parameter the number of arc were lower than the result we got now ? The new parameter were already used in this case and during my tests the smallest radius was split into small circular interpolation but nor the 2 biggest circle

I will look into this.  My instincts tell me that it it because the new process correctly swaps out the current arc, but I will verify.