FormerLurker/ArcWelderPlugin #86: Update with Software Update in Octopi not possible

bill-orange, 2020-11-30:
In Octopi's software update the Arc Welder Plugin reports as this:

```
Arc Welder: Anti-Stutter: unknown 
Release Channel: 
Devel RCs
 
Installed: ?
Available: ?
Unknown error while checking for update, please check the logs
```

The log reports:

`ConfigurationInvalid: Update configuration for arc_welder of type github_release needs all of user, repo and current set and not None`

I have not tried writing plugins myself so I have no theory where the problem might be found.

FormerLurker, 2020-11-30:
What versions of OctoPrint and ArcWelder are you using?

bill-orange, 2020-11-30:
Octopi 1.5.0, Same behavior with the earlier release candidate.   Arc-Welder (1.0.1rc1.dev1) and on a different machine Arc-Welder (1.0.0+u.bb71e8f)

FormerLurker, 2020-11-30:
That's not surprising.  I spent nearly every evening the past few days debugging issues related to 1.5 on both ArcWelder and Octolapse.  I didn't expect a drop of 1.5 so soon, and haven't had time to update things, especially since I was right in the middle of dropping an RC myself.  This may even be solved already.  Anyway, I'll try to push out a fix ASAP, but sounds like update won't work until it's uninstalled anyway :(

FormerLurker, 2020-11-30:
Try installing from the plugin manager with this URL in the 'from url...' box:  https://github.com/FormerLurker/ArcWelderPlugin/archive/91fdae17ea26232560c0a39cd14255c8243a50e4.zip

bill-orange, 2020-11-30:
Yes, that fixes it!  Thanks.

closed 2020-11-30 by bill-orange

FormerLurker, 2020-11-30:
Thanks for reporting this Bill!  Next time I'll try not to add a feature without testing it, lol!