FormerLurker/ArcWelderPlugin #87: Unable to update ArcWelder

bvrielink, 2020-12-03:
From the OcroPrint Software Update screen:

[image]

Also changing Release Channel to RC doesn't change the error. This is on OctoPrint 1.5.0.

Error in octoprint.log:

```
2020-12-03 15:30:30,115 - octoprint.plugins.softwareupdate - ERROR - Could not check arc_welder for updates
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1496, in _get_current_version
    target, check, online=online, credentials=credentials
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/version_checks/github_release.py", line 305, in get_latest
    target
ConfigurationInvalid: Update configuration for arc_welder of type github_release needs all of user, repo and current set and not None
```

Already uninstalled and reinstalled the plugin without effect.

FormerLurker, 2020-12-03:
Fixed in a patch that is not yet released.  Are you installing 1.0.0 or the
dev rc?

bvrielink, 2020-12-03:
I have 1.0.0+u.bb71e8f installed. I was about to update to the RC but was unable to do it via the normal way. 

Fixing it in a patch means that all users have to uninstall the plugin and reinstall the updated version? That probably will mean that users don't get any notification that there is a new version or not?

FormerLurker, 2020-12-03:
You are right regarding folks using OctoPring 1.5.x :(  I will see if Gina is willing to push a notification about this issue.  

Getting the software update wrong is not good, and it's not easy to test since you have to push an actual release to test it.  This per-release channel is a new feature, and I attempted to program it blind before 1.5 existed, but didn't make the deadline for testing and deploying a fix for the final version.  Most unfortunate.  Also, Octolapse had conflicts with a new Knockout version, so I've been dealing with that for the past week too.  That bug affected at least 18K ppl, so been trying to get that solved before fixing ArcWelder (2.5K users).

But, you don't have to uninstall ArcWelder to get the new version.  If you want to try it out, I'd be most appreciative.  I need some feedback since I'm going to have to rush a release.  It's really close to 1.1.0rc1, which I'll push out for a few days before dropping a master.  Going to skip the devel/rc this time due to these issues.  Here is a link to the latest zip you can install through the plugin manager by pasting it into the the 'from url...' box and clicking install:  ```https://github.com/FormerLurker/ArcWelderPlugin/archive/2a95fab3a7797b3dad3664bece27f08137d0f46c.zip```

Again, no need to uninstall, and in fact I'd like to see how the settings migration works for you.  Also, another benefit for me would be when I push out the live version, you can tell me if the update notification is working as expected.  I've seen cases where it worked for me but not others..

bvrielink, 2020-12-03:
I updated to the latest zip without problem:

[image]

Settings seems to be intact. Got some other remarks but will post them in the RC issue list.

FormerLurker, 2020-12-03:
There isn't an RC issue for this version yet because it hasn't been released :)  Just drop it in here and i'll move it when that happens.

There is 1 known issues and 2 potential issues.  The known issue is that the firmware checker might disconnect your printer.  I know how to fix this, and have in one case, but haven't gotten to it yet.

The other two issues could be firmware, both related to the new 'Firmware Compensation' and '3D Arcs' options.  Minor stuff as far as I can see, but it will take some research to pin the blame down.

bvrielink, 2020-12-03:
I meant the feedback post. But never mind will put it in here (more easy to remember as well :) ).

Installed the RC and don't see help yet for setting "3D Arcs(vase mode)". Also got this error/warning:

[image]

This is on an Ender 5 Pro with my own firmware which was working with the 1.0 version of this plugin. I am quitte handy with the firmware stuff but don't know which setting I have to enable for 'G90/G91 Influences Extruder'. This is the output from M115: [M115.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/5638210/M115.txt)

bvrielink, 2020-12-03:
Message got me confused but it seems it's not a firmware setting. Then the next question would be what setting is preferred/better?  The setting from OctoPrint itself or should I use the one from the plugin?

FormerLurker, 2020-12-03:
Yup, that error is correct.  You need to edit the arc welder settings and change the G90/G91 Influences Extruder setting to true for Marlin 2.x.

If it was working without issues for you you are lucky.  That means your slicer is always outputting a M82/M83 after G90/G91, or that you are only using absolute XYZ and E modes all the time, and that all your start/end gcode in your slicer and OctoPrint does this as well.  If any of that changes, you could have issues in OctoPrint, Octolapse, Arc Welder, and more.  

See this setting in OctoPrint:

[image]

By default, ArcWelder will use that (though it should mention marlin 2, and this setting should be in the printer profile since it is firmware specific.. oh well).

But, you can also change it here if you're concerned about changing working OctoPrint settings:

[image]

This setting COULD have an impact on ArcWelder, but not necessarily.  Depends on your gcode.  Worst case arc welder would role up extrusion with something like a wipe action and give you a nice hole in your print somewhere.

Regarding the help, I guess this is what you mean:

[image]

Yeah, that's something I need to update.  Been waiting for some feedback.  See [this issue](https://github.com/FormerLurker/ArcWelderPlugin/issues/56) for details.

You do NOT need to enable firmware compensation for Marlin 2.0.6+, unless the settings in configuration_adv.h are really bad :)

Thanks for the feedback!  Please let me know how I could have made things more clear in the firmware checker, or any other things that come to mind.  I really need the input :)

FormerLurker, 2020-12-03:
> Message got me confused but it seems it's not a firmware setting.

It's a property of your firmware.

> Then the next question would be what setting is preferred/better? The setting from OctoPrint itself or should I use the one from the plugin?

Either way as far as arc welder is concerned.  This probably explained in my last post.

bvrielink, 2020-12-03:
Thanks. It's clear now. Next step for me is to test some prints but that will take a couple of days. Doing some upgrades to the printer atm. If you need me to test something specific let me know.

FormerLurker, 2020-12-05:
Since this is resolved, I'm going to close this out.  Keep your eyes peeled for the next pre-release, I will definitely need some testers :)

closed 2020-12-05 by FormerLurker